grblHAL/STM32F4xx #105: Feed Hold through ! doesn't work, is only accepted in IDLE-State

giutdn, 2022-12-09:
Hello,

i'm currently programming a test rig and using GrblHal 1.1 for that. I've had some experience on Grbl with an Arduino Uno, where everything worked fine. To test a new controller and be more flexible with GPIO's i bought a STM32F411RE and flashed it with the STM32-Software that comes with it. Everything works fine except trying to stop the jog-motion, that is initiated through $J=G91, with an exclamation mark "!". $39 is equal to 1, so i think Grbl should be able to read the ! when it comes in and instantly stop the motion. I'm not sure where i'm going wrong and i hope to get some input from you. Thank you very much :-)

If you need any more information, feel free to ask.

terjeio, 2022-12-09:
Jog motions should be [cancelled/stopped with 0x85](https://github.com/grblHAL/core/blob/d10804433c2b1060bb8ad6c5fa04218c2cf435a8/grbl.h#L125), not feed hold.

I might have missed some state changes when I untangled the legacy state machine, and that feed hold should be available during jogging. However, I am not able to check that for a while as I am on holiday and do not have access to a 8 bit controller.

giutdn, 2022-12-09:
Thanks for clearing that up, works fine with 0x85 :-)

closed 2022-12-09 by giutdn