grblHAL/STM32F4xx #107: Debug Error  / Couldn't start GDB Server / Blackpill STM32f411CE

SebastianWantia, 2023-01-12:
I have a problem to start debugging with Segger J-Link and also with the ST-Link debugger.
The project compiles without a error, other projects in my workspace have no problems like this. 
I couldn't find any reason for that anyone an idea?

Error mesage is: "Error in final launch sequence: Failed to start GDB Server" 

[image]

dresco, 2023-01-12:
Might be worth checking whether the SWO pin is assigned to anything else in the board map? That has caught me out before..

SebastianWantia, 2023-01-12:
The same board with a different software but same debugger works without any error. Must be something with the software.

terjeio, 2023-01-12:
> Must be something with the software.

You are using STM32CubeIdIDE? If so is the debug configuration set up correctly? From the debug tool dropdown select "Debug configurations..." and then click the configuration you want to use to open the settings.

You may also find some hints [via a search](https://duckduckgo.com/?t=ffab&q=Error+in+final+launch+sequence%3A+Failed+to+start+GDB+Server&ia=web).

dresco, 2023-01-12:
> Must be something with the software.

I'd still suspect the pin mapping. Is spindle sync enabled? If so, the SWO pin (PB3) is in use here..

https://github.com/grblHAL/STM32F4xx/blob/7bb854b29d2dbf11d54f6f34ab938e9cb0ec9b4a/Inc/blackpill_map.h#L124-L125

terjeio, 2023-01-13:
> I'd still suspect the pin mapping.

This would only be possibe if an initial flashing was successful. And this statement suggest that the SWO interface is working: "The same board with a different software but same debugger works without any error."

dresco, 2023-01-13:
> This would only be possible if an initial flashing was successful

Yeah, it may not be the problem in this instance, but mentioned as it's something I've tripped over before. Another project of mine was using the SWO pin in the code. Symptom was that flashing worked just fine, but debug would error out (understandably)..

SebastianWantia, 2023-01-13:
Thanks for your help so far!
I haven't tried your suggestion yet. What I can say is that it worked after I created a new Project and copied all files from the GRBLHal Project in the new one. Like in the picture attached. 
Best regards
[image]

SebastianWantia, 2023-01-13:
> You are using STM32CubeIdIDE? If so is the debug configuration set up correctly? From the debug tool dropdown select "Debug configurations..." and then click the configuration you want to use to open the settings.

--> Yes I would say configuration, in the Debugsetting are correct see attached pictures
[image]
[image]

closed 2023-01-14 by SebastianWantia