grblHAL/STM32F4xx #111: A stupid question, how can I send the uid of the stm32f4 through the hal.stream.write(); function?

hanke-cnc, 2023-02-17:
char uid[17];
    	    	    memset(uid, 0, 17);
    	    	    int i;
    	    	    for (i = 0; i < 16; i++)
    	    	    {
    	    	        uid[i] = *(unsigned int*)(0x1FFF7A10 + i);
    	    	    }
    	    	   hal.stream.write(uid);

I received many "6"
[image]

terjeio, 2023-02-17:
> uid[i] = (unsigned int)(0x1FFF7A10 + i);

This looks wrong to me. You want to read the uid from address 0x1FFF7A10 onwards? If so perhaps this works (if the uid is stored as a string)?

uid[i] = *((char *)(0x1FFF7A10 + i));

hanke-cnc, 2023-02-17:
> > uid[i] = (unsigned int)(0x1FFF7A10 + i);
> 
> This looks wrong to me. You want to read the uid from address 0x1FFF7A10 onwards? If so perhaps this works (if the uid is stored as a string)?
> 
> uid[i] = *((char *)(0x1FFF7A10 + i));

I don't know, I'm a programming noob, I'm confused by the data types, this is the code I asked chatgpt gave me,

I don't know why I still get so many
[image]

hanke-cnc, 2023-02-17:
> 

I didn't know that the hal.stream.write function would just send a "6" character, confusing

closed 2023-02-17 by hanke-cnc

terjeio, 2023-02-17:
> this is the code I asked chatgpt gave me,

chatgpt is a silly fool that assembles from and spits out stuff it has found on the internet, including all the BS fom there... IMO not to be trusted with anything. chatgpt got the location of the device identifier right though.

From 34.1 in the PRM: _The 96-bit unique device identifier can also be read in single bytes/half-words/words in
different ways and then be concatenated using a custom algorithm._

This means that the identifier is binary data and has to be converted to ASCII in whatever format you need. If a standard UUID then convert each byte to hex representation before printing. Calling sprintf is perhaps the easiest way to to do it, similar to how I do it for [MAC addresses](https://github.com/grblHAL/STM32F7xx/blob/27db628d147b009657cdf8c631940fcb3ad5e073/Src/enet.c#L140). The format string I use for that is found [here](https://github.com/grblHAL/Plugin_networking/blob/e76ecfb7415adf552f3b94abd3284a4ebb6b8e36/networking.h#L145).

> I didn't know that the hal.stream.write function would just send a "6" character, confusing

It is a bit odd, but then it prints whatever it is fed up to a terminating null character. And do not put the code inside the main protocol loop - it will repeatedly output the data. Better to add it to the [function that outputs the $I response](https://github.com/grblHAL/core/blob/2779c833fdf59a0a39f64110ca6e580cfaecae7f/report.c#L807)? However, since this is a customization the perfect place would be in board specific code associated with your board map since this can extend the $I report.

hanke-cnc, 2023-02-17:
> chatgpt is a silly fool that assembles from and spits out stuff it has found on the internet, including all the BS fom there... IMO not to be trusted with anything. chatgpt got the location of the [...]

Thanks to terjeio, using the sprintf function to get the correct uid and send it out, cool

uint32_t UIDw[3];
	
	  UIDw[0]              = HAL_GetUIDw0();
	  UIDw[1]              = HAL_GetUIDw1();
	  UIDw[2]              = HAL_GetUIDw2();
	  char str[60];
	  sprintf(str, "%08X\n", UIDw[0]);    //将int转化并保存至str中
	  hal.stream.write(str);
	  sprintf(str, "%08X\n", UIDw[1]);    //将int转化并保存至str中
		  hal.stream.write(str);
		  sprintf(str, "%08X\n", UIDw[2]);    //将int转化并保存至str中
			  hal.stream.write(str);