grblHAL/STM32F4xx #112: Kompilierungsfehler mit Nucleo-F446RE (BOARD_GENERIC_UNO)

TOKENone7472, 2023-02-26:
Kann den Code nicht Kompilieren. Alte Versionen oder Neu geht nicht. Kann mir jemand Helfen? Ich finde das problem nicht.
Ich bedanke mich schon einmal für die Hilfe.

`Processing nucleo_f446re_GENERIC_UNO (board: nucleo_f446re; platform: ststm32; framework: stm32cube)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f446re.html
PLATFORM: ST STM32 (9.0.0) > ST Nucleo F446RE
HARDWARE: STM32F446RET6 180MHz, 128KB RAM, 512KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES:
 - framework-stm32cube @ 2.0.200813
 - framework-stm32cubef4 @ 1.26.2
 - tool-ldscripts-ststm32 @ 0.1.0
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 44 compatible libraries
Scanning dependencies...
Dependency Graph
|-- grbl
|-- Core
|-- Class
|-- App
|-- Target
Building in release mode
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\alarms.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\coolant_control.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\corexy.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\errors.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\gcode.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\grbllib.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\ioports.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\limits.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\maslow.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\motion_control.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\my_plugin.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\ngc_expr.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\ngc_params.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\nuts_bolts.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\nvs_buffer.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\override.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\pid.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\planner.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\protocol.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\regex.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\report.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\settings.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\sleep.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\spindle_control.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\state_machine.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\stepper.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\stream.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\system.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\tool_change.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libc19\grbl\wall_plotter.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\lib46c\Core\usbd_core.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\lib46c\Core\usbd_ctlreq.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\lib46c\Core\usbd_ioreq.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libd4a\Class\CDC\Src\usbd_cdc.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\lib1fe\App\usb_device.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\lib1fe\App\usbd_cdc_if.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\lib1fe\App\usbd_desc.o
Compiling .pio\build\nucleo_f446re_GENERIC_UNO\libd0c\Target\usbd_conf.o
In file included from Middlewares\ST\STM32_USB_Device_Library\Core\Inc/usbd_core.h:29:0,
                 from Middlewares\ST\STM32_USB_Device_Library\Core\src\usbd_core.c:21:
USB_DEVICE\Target/usbd_conf.h:34:10: fatal error: main.h: No such file or directory

**************************************************************
* Looking for Compiling .pio\build\nucleo_f446re_GENERIC_UNO\src\btt_skr_1.1.o
main.h dependency? Check our library registry!
[... 117 lines cut ...]
 #include "main.h"
          ^~~~~~~~
compilation terminated.
*** [.pio\build\nucleo_f446re_GENERIC_UNO\lib1fe\App\usb_device.o] Error 1
In file included from Middlewares\ST\STM32_USB_Device_Library\Core\Inc/usbd_def.h:29:0,
                 from USB_DEVICE\Target\usbd_conf.c:25:
USB_DEVICE\Target/usbd_conf.h:34:10: fatal error: main.h: No such file or directory

**************************************************************
* Looking for main.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:main.h"
* Web  > https://registry.platformio.org/search?q=header:main.h
*
**************************************************************

 #include "main.h"
          ^~~~~~~~
compilation terminated.
*** [.pio\build\nucleo_f446re_GENERIC_UNO\libd0c\Target\usbd_conf.o] Error 1
==================================================================== [FAILED] Took 1.99 seconds ====================================================================
Environment                Status    Duration
-------------------------  --------  ------------
nucleo_f446re_GENERIC_UNO  FAILED    00:00:01.986
============================================================== 1 failed, 0 succeeded in 00:00:01.986 ============================================================== `

terjeio, 2023-02-26:
You are using the supplied platformio.ini? If so there is no section for the uno, you may try adding this:

```
[env:nucleo_f446re_generic_uno]
board = nucleo_f446re
board_build.ldscript = STM32F446RETX_FLASH.ld
build_flags = ${common.build_flags}
  # See Inc/my_machine.h for options
  -D BOARD_GENERIC_UNO=
  -D NUCLEO_F446=
lib_deps = ${common.lib_deps}
  eeprom
lib_extra_dirs = ${common.lib_extra_dirs}
# Alternatively, place the .pio/build/<env name>/firmware.bin on the NODE_F4xxRE drive
upload_protocol = stlink
```

Also comment out [this line](https://github.com/grblHAL/STM32F4xx/blob/68d6ece611723d250308c20f02d01fe9486b007a/Src/driver.c#L126) as it incorrectly reports the error for the Uno.

TOKENone7472, 2023-02-27:
Danke Super. Läuft 
2 Felher waren es.
[env:nucleo_f446re_GENERIC_UNO] Groß geschrieben von dir klein geschrieben [env:nucleo_f446re_generic_uno]   
und die Felher meldung aus Kommentieren.
Danke nochmal.

closed 2023-02-27 by TOKENone7472