grblHAL/STM32F4xx #113: STM32F401 compile use web builder, connection virtual UART Always get notif from uart GrblHAL 1.1f ['$' or '$HELP' for help]

MOMOK30, 2023-02-27:
STM32F401 compile use web builder, connection virtual UART. Always get notif from uart GrblHAL 1.1f ['$' or '$HELP' for help]
[image]

terjeio, 2023-02-27:
Which board?
And do other senders, such as [ioSender](https://github.com/terjeio/ioSender), behave in the same way?

MOMOK30, 2023-02-27:
blackpill stm32f401cc 
i have try lasergbrl, lightburn, ugs, grblguru, they have same notification..io sender not yet try

MOMOK30, 2023-02-27:
ioSendor not reponding

terjeio, 2023-02-27:
Ok, I do not have a F401 Blackpill to test with. Using my F411, which only differs in the clock setup and linker script, it runs ok - both via UART and USB.
However I see that the reset/estop pin (B6) is left floating when it should have had a pull-up resistor enabled.
Is this pin left floating (unconnected) in your setup or is it connected to ground or 3.3V? If left floating can you try with connecting it to ground?

MOMOK30, 2023-02-27:
if B6 pull-up resistor i got this error, like picture

if use connection USB, There isn't any error, but if any relay active,connection will disconnect between stm32 and pc, so must to reset the stm32 to connect again, 
[image]

terjeio, 2023-02-27:
> but if any relay active,connection will disconnect

Ok, this is interesting - the relays are activated by commands to the controller? Or something external to it?
I suspect you are having an interference issue causing the controller to reboot. So if you disconnect the relays does it run as it should?

terjeio, 2023-02-27:
How are the relays connected to the controller? Do you have a schematic and/or photo(s)?

MOMOK30, 2023-02-27:
Maybe the wiring like this 

PB1 (ENABLE SPINDLE) -> HC244 -> 10K -> Base BD139

GND -> Emiter BD139

Relay Coil (-) -> Colector BD139
VCC -> relay coil (+)

[image]

terjeio, 2023-02-27:
No [snubber diode](https://resources.altium.com/p/using-flyback-diodes-relays-prevents-electrical-noise-your-circuits) across the relay coil?

MOMOK30, 2023-02-27:
Add snubber diode, I not try it yet 

Because if i use UART connection on stm32f1 and esp32 there's no interference,

i will try add snubber diode tomorrow..

MOMOK30, 2023-02-28:
I have try add snubber diode
Still same like before