grblHAL/STM32F4xx #115: Auto squaring with more than one axis (SKR Pro v1.2)

matthiasng, 2023-04-10:
In https://github.com/grblHAL/STM32F4xx/commit/7bb854b29d2dbf11d54f6f34ab938e9cb0ec9b4a the skr pro 1.1 map was changed to support auto squaring only on one axis.

My machine uses auto squaring for x & y. I am currently using a grlb version before this commit and it works for me.

Can you explain to me why this was a problem?
If the problem is only related to the specific pin, do you think it is possible to use another one?

terjeio, 2023-04-10:
> Can you explain to me why this was a problem?

From the commit comment: "limited number of auto squared axes to 1 due to IRQ conflict"

The STM32 architecture can only have a single interrupt enabled for a given pin number, e.g. PA1 and PB1 cannot have interrupts enabled at the same time.

> If the problem is only related to the specific pin, do you think it is possible to use another one?

Yes, as long as the pin is free to use as an input and does not clash with another pin that has interrupts enabled. It _is_ possible to use polling - but I do not like to introduce that in the codebase... 

FYI I had to make more changes to the board map due to another IRQ clash - this has not been published yet.

terjeio, 2023-04-16:
FYI if swapping M3 and M4 motor definitions auto squaring for two axes will be possible. Not sure if I should modify the map file for that.,,

closed 2023-08-26 by terjeio