grblHAL/STM32F4xx #117: Machine pause randomly for some time and resume

muhammadelmogy, 2023-04-14:
I just have finished an upgrade to my machine , I was using grbl-mega-5x on Arduino Mega and using bCNC.
I upgraded to grblHAL  on stm32f411 (currently prototype on a bread board) and I am using bCNC also.

The machine is running just fine, Homing, jogging and G-Code from MIDI all works fine.
However, when I start a job, the machine starts and after sometime, randomly, the G-Code execution pause for several seconds maybe up to less than a minute and then resume and I mean by pause that the controller software(I tried this on bCNC and ioSender) stop sending the G-code to the hardware controller and I know this by looking on the code view in ioSender, the lines was just flowing one after the other and suddenly the line flow stops for sometime and the machine at this moment freeze and the motors still enabled.

This behavior (the pause and resume) happens as it appears randomly and there is no definitive time line for it to happen nor any specific duration, some time happen only once in a long time and sometime and sometimes may happen twice within a couple of minutes !!
All my inputs(start-halt-hold-probe) are going through opto-isolators and also all the limit switch are the same
Also all my outputs(Stepper drivers signals also run through opto-isolators)
Any suggestions?

muhammadelmogy, 2023-04-14:
> Add - as I wrote above. I does not matter if before or after.

Man you are quick :D
I deleted my last comment after I realized what did you ment and after I made a quick search about it!
Anyways, I added the mentioned line and it is now working fine on a 100% feed rate override and it took 22:10
I will try it one more time on a 200% just to make sure about it!

muhammadelmogy, 2023-04-14:
I have tested at 200% and it is working fine!
Is changing  to down-counting mode will be the fix for this issue?

terjeio, 2023-04-16:
> Is changing to down-counting mode will be the fix for this issue?

Yes, but I have changed IRQ priorities as well to be sure.

closed 2023-04-16 by terjeio