grblHAL/STM32F4xx #120: Failed to start GDB server (STM32 Nucleo-F446RE)

hemsta, 2023-05-17:
I am using the STM32F446RE. When I build the project, it works with 0 errors and 0 warnings:

15:09:45 **** Incremental Build of configuration Release F446 Nucleo64 for project GRBL Driver STM32F401 ****
make -j4 all 
arm-none-eabi-size   grblHAL\ F446\ Nucleo.elf 
   text	   data	    bss	    dec	    hex	filename
 138316	   1496	  16480	 156292	  26284	grblHAL F446 Nucleo.elf
Finished building: default.size.stdout

15:09:46 Build Finished. 0 errors, 0 warnings. (took 941ms)

However, when I actually run the project, it first builds with 0 errors and 0 warnings and then the IDE gives me this message:

'Launching GRBL Driver STM32F4xx Debug F446 Nucleo 64' has encountered a problem.

Error in final launch sequence:

Failed to start GDB server.

Other projects I wrote do work, so I assume the problem is not the IDE. One guess I have is that the project name is "GRBL Driver STM32F401" and that I did not configure something correctly. Any idea why the GDB server fails?

terjeio, 2023-05-17:
> Failed to start GDB server.

I get this message on my RPi 400 if I forget to power the target board or if the debug connection is not working.

> Other projects I wrote do work

With the same board? Is it a Nucleo-64? Are you downloading with Debug or Run - and are the selected configuration properties ok?

If you are not debugging you may try to flash the binary with the [STM32 STLINK-Utility](https://www.st.com/en/development-tools/stsw-link004.html). I find this useful for checking the debug interface as well.

hemsta, 2023-05-17:
Yes, with the same board, it is a Nucleo-F446RE. Weirdly enough, when I run a different project first, it downloads onto the board without issue. If I then download the Grbl Driver project, get the "Failed to start GDB server" error and afterwards run my first project again, I now also get "Failed to start GDB server" for this project. So I somehow must be changing something about the GDB server.

hemsta, 2023-05-24:
I tried that, the debugger settings are the same. But I noticed all projects say "Program does not exist" except for the GRBL Driver STM32F4xx Debug F446 Nucleo64 (I used the Release before). Building works now, but when I run the main I still get the error Failed to start GDB server. Now I get the warning : Invalid project path: Missing project folder or file: \GRBL Driver STM32F401\encoder for source path.

hemsta, 2023-05-24:
To clarify, when I build and run it still says "0 errors, 0 warnings". Then the window saying "Failed to start GDB server" pops up. And when I look at the "Problems" tab, it says the warning above "Invalid project path: Missing project folder or file: \GRBL Driver STM32F401\encoder for source path."

closed 2023-08-26 by terjeio