grblHAL/STM32F4xx #121: MSG:Warning: Settings write failed!

nickshl, 2023-05-26:
I made controller using WeAct Studio BlackPill F411. I build firmware using web builder with defaults except I enabled Spindle Sync. When I tried to enter command in console I got an error:
```
$14=64
ok
[MSG:Warning: Settings write failed!]
```
Build info:
```
$I
[VER:1.1f.20230519:]
[OPT:VNMSL,35,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,ES,TC,SS,SED]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32F411]
[DRIVER VERSION:230513]
[BOARD:BlackPill]
[AUX IO:1,2,0,0]
ok
```
NVS STORAGE:*FLASH - should be right... wait, let's try Nucleo-64 F401, it used to work. And it works now, with firmware built with web builder. However:
```
$I
[VER:1.1f.20230519:]
[OPT:VNMSL,35,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,ES,TC,SS,SED,RTC]
[FIRMWARE:grblHAL]
[NVS STORAGE:*EEPROM]
[DRIVER:STM32F401CC]
[DRIVER VERSION:230513]
[BOARD:Nucleo-64 CNC Breakout]
[AUX IO:4,7,0,0]
ok
```
NVS STORAGE:*EEPROM - doesn't looks right to me, Nucleo does not have any EEPROM, I didn't connect external EEPROM either... but it works. Something wrong here...

nickshl, 2023-05-26:
Additional info: after I sent $14=64 and got error, I opened ioSender and set all other settings. After that all settings was saved and preserved via power cycle. When I tried to change $14 via PuTTY I got error again.

hanke-cnc, 2023-05-27:
This issue only occurs on the F401 series chips and not on the F446. I have reported this issue to terjeio, but it seems that terjeio does not have an F401 circuit board, so they have not fixed it and only send error messages when writing fails. Unfortunately, I am not a programmer. The specific manifestation of this problem is that when the chip is powered on for the first time, if you only set one parameter, that parameter will not be saved after power-off. You must set another parameter, such as 384=1 or something, for the parameter to be saved.

terjeio, 2023-05-27:
The first write (or rather erase) fails after power up if the MCU is programmed with the STM32 ST-Link Utility, when I program with STM32CudeIDE it does not - I have no idea why.

Retrying the erase if it fails seems to fix the issue, I will commit an update later today with this workaround.

For new board designs I strongly recommend adding at the minimum a footprint for an I2C EEPROM, even better to add one as well - best is FRAM. 2Kb minimum, 4Kb+ recommended as new features may require more capacity. This since there is no wear levelling implemented when storing to flash.

nickshl, 2023-05-27:
Since there no wear leveling, I prefer to have control of writing. Setting change command can be volatile and change value only in RAM, and another command can be issued to save settings($SAVE probably?)

terjeio, 2023-05-27:
> another command can be issued to save settings($SAVE probably?)

No, this will require users to be aware of and remember issuing the $SAVE command. This is a breaking change I am not willing to implement.
And most changest to the storage area is not when changing settings but rather when changing offsets, e.g. when setting or zeroing the current position.

Changes are buffered in RAM and written to permanent storage when the controller is in IDLE or ALARM mode and not in file run mode.
If sending more than one settings in one go put the controller in file run mode while sending the settings. This is done by encapsulating the command block in % characters e.g:
```
%
$0=5
$1=49
%
```

It might be possible to create a plugin that intercepts writeouts and implements a $SAVE command, this since the [calls are indirected via function pointers](http://svn.io-engineering.com/grblHAL/html/nvs_8h.html).

nickshl, 2023-05-27:
> or zeroing the current position

Wait, what? Every time when I press "zero" button in sender, grblHAL write it into NVM? It is unacceptable behavior in case flash used for store settings!

> No, this will require users to be aware of and remember

User doesn't need to remember anything. User uses sender. Sender have to do it. Like change all desired settings, then press "Write" button... wait, that exactly what I have to do if I want save settings using ioSender, isn't it? Beside, it needed only if flash used to store settings. EEPROM and FRAM can save settings immideatly.

terjeio, 2023-05-27:
> Wait, what? Every time when I press "zero" button in sender, grblHAL write it into NVM? It is unacceptable behavior in case flash used for store settings!

That is why I recommend using EEPROM or FRAM. AFAIK most flash tolerate in the region of 10000 - 100000 erase/write cycles before wearing out so it isn't a likely problem for hobby grade use...

> User doesn't need to remember anything. User uses sender. Sender have to do it.

Ok, I may consider adding it when _all_ relevant senders has the $SAVE command implemented for flash storage that does not have wear levelling support. You can write up the spec and distribute to sender developers and come back to me with an estimate for when they are ready for this?

nickshl, 2023-05-28:
> You can write up the spec and distribute to sender developers

Just gave me a list of senders specifically made for grblHAL and know about all additional settings and I will consider to do that 🙂

If you don't want to break anythyng by introducing any new functionality, it may be configurable with additional $ parameter and have existing behavior by default.

nickshl, 2023-05-29:
> For new board designs I strongly recommend adding at the minimum a footprint for an I2C EEPROM, even better to add one as well - best is FRAM.

Decided to give it a try, ordered MB85RC256V module, then found 24C256 module in electronic component pile, opened blackpill_map.h... and realized that there no I2C definitions. Web builder does not have EEPROM option available for this board either.

terjeio, 2023-05-30:
Have you made a board already or are you wiring directly to a Blackpill?
Motor 3 pins GPIOA 6 and 7 could be swapped with feed hold/cycle start pins on GPIOB 6 and 7 to free up pins for I2C in a new board map.

nickshl, 2023-05-30:
I didn't make PCB and I am not sure if I will. For now I use prototyping PCB to make a lathe controller. But my goal to make simple instruction how to recreate it, so it have to be implemented in Web Builder.

At this point I ended with this pinout:
```
/* Default Pin Assignments:
 *
 *                     VB | +3V
 *                    C13 | GND
 *     Coolant Flood  C14 | +5V
 *      Coolant Mist  C15 | B9   I2C SDA
 *                    RST | B8   I2C SCL
 *             Reset   A0 | B7   Cycle Start
 *       Door Safety   A1 | B6   Feed Hold
 *            X Step   A2 | B5   
 *       X Direction   A3 | B4   Encoder Index
 *            Y Step   A4 | B3   Encoder Pulse (B)
 *       Y Direction   A5 | A15  Encoder Pulse (A)
 *            Z Step   A6 | A12  USB D+
 *       Z Direction   A7 | A11  USB D-
 *       Step En/Dis   B0 | A10  USART1_RX
 *    Spindle Enable   B1 | A9   USART1_TX
 * Spindle Direction   B2 | A8   Spindle PWM
 *                    B10 | B15  Probe
 *                    +3V | B14  Z Limit
 *                    GND | B13  Y Limit
 *                    +5V | B12  X Limit
 */
```
- Y2 Step, Dir & Limit removed since lathe uses only X & Z axis. Y axis still there just in case someone will try to upgrade lathe to a mill by adding third axis on cross slide.
- X, Y and Z Step/Dir shifted, Reset and Door Safety moved to PA0 & PA1. PA0 connected to a button on BlackPill.
- Cycle Start moved to PB6.
- PB9 & PB8 used for I2C.
- Encoder Index moved to PB4, this pin can be used as TIM3 input.
- PB3 left for encoder phase B input since TIM2 can use PA15 & PB3 as encoder input. 
- PA10 & PA9 left for UART1 for MPG stream input which I hope will support in future.
- PB5, PB10 and PC13(LED on BlackPill) is not assigned to anything

Is there any issues with such layout? If no, could you add attached blackpill_lathe_map.h to repo and Web Builder?

[blackpill_lathe_map.h](https://github.com/grblHAL/STM32F4xx/files/11605812/blackpill_lathe_map.h.txt)

terjeio, 2023-06-02:
I have added the map as Blackpill Alt. 2 with a number of additional options:
* Aux I/O depending on available pins.
* I2C Keypad option, again depending on available pins.
* EEPROM defaults to 2K in the Web Builder - disable or select as needed.
* Spindle sync option is not forced.
* ...

Note that I have not tested with the map.

If you plan to add a website for your build then a link can be added to the [board _Homepage_ button](https://github.com/grblHAL/STM32F4xx/blob/2d65a02b9b7af1bc59bce0df6af8dc5749575428/driver.json#L89) in the  Web Builder .

closed 2023-08-26 by terjeio

nickshl, 2023-11-17:
Hi Terje Io, I have a request about Blackpill Alt. 2 configuration: is it possible to add the 5th axis? I missed two pins when I did this confuguration: PA13 and PA14. Those pins are used for debug/programming via SWO, however they can be used as GPIO. I would like to use PA13(SWDIO) as B Step and PA14(SWCLK) as B Dir.

I am a trace PCB for the CNC controller and want to order boards this weekend. It is almost(there always a place for improvement 🙂) perfect, but last minute I realize that it would be nice to have the 5th axis... and there are two unused pins suitable for that.

terjeio, 2023-11-18:
> is it possible to add the 5th axis?

Yes, that is possible as it will be a non-breaking change.

nickshl, 2023-12-07:
I finally close to release my version of CNC breakout board(actually controller, since BlackPill will be included).
And because of that I would like to change some things... if possible.

1. I would like to change name of the board(and name of file too) from "BlackPill Alt. 2 (STM32F411)" to "Devtronic breakout board (BlackPill F411)"
2. Since this board have dedicated door input, probably make sense to make "Safety door input" enabled by default in Web Builder
3. This board have MB85RC256 FRAM and I would like to have appropriate setting by default in Web Builder(EEPROM is FRAM and appropriate size)
4. Since this board have encoder input I would like to have "Spindle sync" enabled by default, if it doesn't hurt anything if not used of course.

> Yes, that is possible as it will be a non-breaking change.

5. Since this board have 5th axis connector I would like to include 5 as an option for "Number of axes:".  Also I would like to modify Pin assignment section in .h file to reflect that:

```
/* Pin Assignments:                  
 *
 *                    B Direction -   - B Step
 *                                 | |
 *                               G S D +
 *                               N C I 3
 *                               D K O V
 *                             -----------
 *                         VB |           | +3V
 *           Step En/Dis  C13 |           | GND
 *         Coolant Flood  C14 |          | +5V
 *          Coolant Mist  C15 | *     - * | B9   I2C SDA
 *                        RST |      |K|  | B8   I2C SCL
 *                 Reset   A0 |       -   | B7   Feed Hold
 * Safety door/I2C strobe  A1 |           | B6   Cycle Start
 *                X Step   A2 |           | B5   Probe
 *           X Direction   A3 |    / \    | B4   Encoder Index
 *                Y Step   A4 |   <MCU>   | B3   Encoder Pulse (B)
 *           Y Direction   A5 |    \ /    | A15  Encoder Pulse (A)
 *                Z Step   A6 |           | A12  USB D+
 *           Z Direction   A7 |   -   -   | A11  USB D-
 *                A Step   B0 |  |R| |B|  | A10  USART1_RX
 *           A Direction   B1 |   -   -   | A9   USART1_TX
 *        Spindle Enable   B2 |           | A8   Spindle PWM
 *     Spindle Direction  B10 |           | B15  A Limit
 *                        +3V |   -----   | B14  Z Limit
 *                        GND |  |     |  | B13  Y Limit
 *                        +5V |  | USB |  | B12  X Limit
 *                             -----------
 */
```

Please let me know if you have any questions or concerns.

terjeio, 2023-12-08:
> And because of that I would like to change some things... if possible.

The listed changes cannot be added to the current map file without breaking backwards compatability, so you will have to add a new one.

2. Making the safety door input default in the Web Builder is currently not possible. However you may force it by undefining/defining `SAFETY_DOOR_ENABLE 1` in the map file.

3. Either force [FRAM in the map fil](https://github.com/grblHAL/STM32F4xx/blob/596a3c71f30710705a10a353cc2c818678820d76/Inc/st_morpho_map.h#L43-L48)e or set it as default in the entry for the board in [driver.json](https://github.com/grblHAL/STM32F4xx/blob/master/driver.json).

4. Again, force it the map file by undefining/defining `SPINDLE_SYNC_ENABLE 1`.

5. Add it to the new map file.