grblHAL/STM32F4xx #123: SKR 2 SDCARD_ENABLE

asward, 2023-05-27:
I've got an SKR 2  (SMT32F407) running with TMC2209 drivers. I was hoping to use a telnet forwarder (such as ESP3D running on an ESP8266) and ioSender to align, control, and configure my machine each run. However, I would prefer to not use WiFi to send gCode for running jobs.

The SD plugin seems to be what I want. When I enable it (in my case via platformio.ini -D SDCARD_ENABLE=1) I get a compilation error from btt_skr_2.0_map.h (line 159) stating that 'SD SDIO/MMC interface does not support SPI.'. 

A few questions:

1. I had assumed my drivers were configured using UART and not SPI, but I don't think I had many options with the SKR2 configs to date. Is this the SPI being referenced in the pre-processor error?
2. Is this a limitation of the chip, grblHal (somehow), or just where the development on this controller was left to make a functioning system?
3. Is this something I can assist with accomplishing (assuming item 2 above isn't a physical restriction)? I have some decent c/c++ experience but I'm pretty unfamiliar with the SMT32F. But I can read datasheets (if I have to!).

dresco, 2023-05-28:
I think it's just that the current F4 driver expects the SDCard to be interfaced over SPI, whereas on the SKR2 it's a full SDIO/MMC connection. So not a hardware limitation..

terjeio, 2023-05-29:
The SKR-2 map was provided by a user and he decided not to add SD card support. I do not own one and at the time it was added I did not have other hardware to test with.

3. Yes, easiest would be use the STM32CubeIDE and set up a new project and use that to generate most of the needed code, then integrating it back into this driver. Jons [H7xx](https://github.com/dresco/STM32H7xx) repo has SDIO code in place already so that might be a useful reference. What could be an issue is that this driver already has a single lane SPI interface for SD card available, there might be clashes with that that has to be resolved.

asward, 2023-05-29:
Thanks for the info, I'll give it a shot.

dresco, 2023-06-03:
> I will commit the F7xx changes as soon as I am satisfied with the tests.

Cool, I'll keep an eye out - as I would like to keep the H7 driver as close to upstream F7 as possible.

Most of the aggravation I had with the H7 implementation was related to cache coherency (that wasn't handled properly in the ST generated code).

asward, 2023-06-28:
I haven't given up! But this is more convoluted than I even imagined. 

I had built a bare bones sd/fatfs project with STMCubeIDE for the F4. Pulled the relevant middleware and FATFS folders over and did what I thought was right to mitigate sdio vs spi configuration. This is what I thought would be the tricky thing, but turned out to be pretty straight forward. I compared against @terjeio SDIO changes for the F7 and I think I got pretty close (again, not an embedded guy, so trying to stay inline with project best practices as I find them).

The real stumbling block now is that STMCubeIDE v1.12.1 (current as of June '23) is generating FatFS code and including middleware for v0.12c of FATFS. This seems to be a downgrade from earlier versions of STMCube which were written against at least 0.14. There are a number of ~breaking changes as far as I can see. 

I think this would be all here nor there, but the sdcard plugin also seems to depend on at least v0.13. This is based mostly on the configuration defines the sdcard plugin is looking for (FATFS 0.13 change log: Changed heading character of configuration keywords "_" to "FF_".)

All this to say - I'm wondering how I should proceed here. Would it work to pull in the F7 code directly and the HAL will take care of it? It seems like STM reverted on purpose but I have no idea why or what affects upgrading would have. I could use an older version of the IDE to get real F4 code. Or I could hack it all in some other way?

Certainly out of my element at this point, but very interested to learn how this typically works and hopefully make a contribution here.

dresco, 2023-06-29:
> The real stumbling block now is that STMCubeIDE v1.12.1 (current as of June '23) is generating FatFS code and including middleware for v0.12c of FATFS. This seems to be a downgrade from earlier [...]

Fwiw, my v1.10 CubeIDE install also generates FATFS support using v0.12c (both F4 and H7 projects). I suspect the FATFS code in the F4 repo was added manually? (I know I had to manually update to later files for my H7 driver). 

In your position (assuming default CubeIDE generated file locations), I think I would update the FATFS source under Middlewares to the later upstream version, and then fix up the generated platform specific files under FATFS as needed.

Is worth noting that Terje's file locations for the F7 driver are a bit different to the generated ones, so perhaps that layout is preferred?

terjeio, 2023-06-29:
> Is worth noting that Terje's file locations for the F7 driver are a bit different to the generated ones, so perhaps that layout is preferred?

To support both single lane SPI and four lane SPI (SDIO) I decided to refactor both the original grblHAL code and the STM32 library code allowing both to be used. I changed file locations in order to achieve this without duplicating the core FatFS code and thus have the same FatFs version (R0.14) for both. I guess the F7 layout _and_ code can be copied over to the F4 driver "as-is"...

terjeio, 2023-07-08:
I have  managed to get SDIO working with a Nucleo 144 board, with luck it should work with the BTT SKR2.0.
Polling mode is used, DMA mode could be added for speed/performance?
Note that I had to slow down the fast clock to avoid recieve FIFO overflows.

closed 2023-08-26 by terjeio