grblHAL/STM32F4xx #126: Hi, may I ask if F407 requires specific pins for the two input pins for spindle synchronization?

hanke-cnc, 2023-06-10:
I am remaking a GRBLHAL control board and have led out the two pins for spindle synchronization. Since I do not have a lathe for testing, I am not sure if there are any special requirements for the two spindle synchronization pins. I have defined them as B11 and B12, and the compilation did not show any errors.

terjeio, 2023-06-10:
Spindle pulse: currently D2 or A15, I may add more pins later.
Spindle index: any free pin that can have interrupt enabled (not clashing with other interrupt enabled pins).

hanke-cnc, 2023-06-11:
> Spindle pulse: currently D2 or A15, I may add more pins later. Spindle index: any free pin that can have interrupt enabled (not clashing with other interrupt enabled pins).

Thank you so much!! Thank you!!

closed 2023-06-11 by hanke-cnc