grblHAL/STM32F4xx #127: HardFault when stepper segment buffer is empty

dagerlandsson, 2023-07-04:
Running a Blackpill STM32F411CE. Driving a galvo with a adc using code from st_morpo_dac.c.
In this setup I can have very high feedrate and while testing how far I can go i run into a hardfault when stepper segment buffer become empty when there is lot of short segments i gcode file.
Running in laser_mode.

The crash is in this code, grbl/stepper.c
>ISR_CODE void ISR_FUNC(stepper_driver_interrupt_handler)(void)
{
....
// Segment buffer empty. Shutdown.
st_go_idle();
// Ensure pwm is set properly upon completion of rate-controlled motion.
if (st.exec_block->dynamic_rpm && st.exec_block->spindle->cap.laser)
  st.exec_block->spindle->update_pwm(st.exec_block->spindle->pwm_off_value);

I put a breakpoint at the st_go_idle() row.
Second time breakpoint hits st.exec_block is null.
The st struct looks like this when it crashes
>B::Var.View__%Recursive_%open_st
  st = (
    counter_x = 52520,
    counter_y = 107472,
    counter_z = 107472,
    counter_a = 107472,
    counter_b = 107472,
    new_block = FALSE,
    dir_change = FALSE,
    step_outbits = (mask = 2, value = 2, x = 0, y = 1, z = 0, a = 0, b = 0, c = 0, u = 0, v = 0),
    dir_outbits = (mask = 1, value = 1, x = 1, y = 0, z = 0, a = 0, b = 0, c = 0, u = 0, v = 0),
    steps = (8, 214944, 0, 0, 0),
    amass_level = 0,
    step_count = 0,
    step_event_count = 214944,
    exec_block = 0x0 -> NULL,
    exec_segment = 0x0 -> NULL)

First time breakpoint hit (when it not crashes) st looks like this
>B::Var.View__%Recursive_%open_st
  st = (
    counter_x = 52520,
    counter_y = 107472,
    counter_z = 107472,
    counter_a = 107472,
    counter_b = 107472,
    new_block = FALSE,
    dir_change = FALSE,
    step_outbits = (mask = 2, value = 2, x = 0, y = 1, z = 0, a = 0, b = 0, c = 0, u = 0, v = 0),
    dir_outbits = (mask = 1, value = 1, x = 1, y = 0, z = 0, a = 0, b = 0, c = 0, u = 0, v = 0),
    steps = (8, 214944, 0, 0, 0),
    amass_level = 0,
    step_count = 0,
    step_event_count = 214944,
    exec_block = 0x2000115C -> (id = 2, next = 0x2000119C, steps = (8, 214944, 0, 0, 0), step_event_count = 214944, direction_bits = (mask = 1, value = 1, x = 1, y = 0, z = 0, a = 0, b = 0, c = 0, u = 0, v = 0), overrides = (value = 0, feed_rate_disable = 0, feed_hold_disable = 0, spindle_rpm_disable = 0, parking_disable = 0, reserved = 0, sync = 0), steps_per_mm = 600.0, millimeters = 44.779999, programmed_rate = 50000.0, message = 0x0, output_commands = 0x0, backlash_motion = FALSE, dynamic_rpm = FALSE, spindle = 0x20000DA8),
    exec_segment = 0x0 -> NULL)

terjeio, 2023-07-04:
You are pushing the MCU too hard?
Try changing [these lines](https://github.com/grblHAL/STM32F4xx/blob/c0969fa26c570f825bb5ff21faf5613a0108cb47/Src/driver.c#L2530-L2538) to
```
// Main stepper driver
void STEPPER_TIMER_IRQHandler (void)
{
    if ((STEPPER_TIMER->SR & TIM_SR_UIF) != 0)                  // check interrupt source
    {
        hal.stepper.interrupt_callback();
        STEPPER_TIMER->SR = ~TIM_SR_UIF; // clear UIF flag
    }
}
```
This should ensure that the stepper interrupt is not fired after the timer is disabled if it timed out before the interrupt callback was completed. If this helps then you are at the edge of or exceeding the max step rate possible - not a good place to be.

dagerlandsson, 2023-07-04:
Yes, I'm pushing it (to?) hard. Changed the suggested code and now it don't crash anymore.

Trying to understand how far I can go.
I have 600 steps/mm, changed "Step pulse time" to 2us.
If then make assumption it is possible to send out a pulse every 2.5uS it will give
600*2.5 = 1500 us/mm that is 40000mm/min (60*1000000/1500)

It crashed out when trying a feedrate above around 25000mm/min.
25000mm/min give a step freq of 250kHz (25000*600/60)

Maybe it is to optimistic that it is possible to send out pulses this fast.
[https://www.grbl.org/single-post/how-fast-can-it-go](url) is indicating that it is possible to run up to 400kHz. But that is with a faster cpu. So maybe I hit the limit at 250kHz.
Not a big deal for me, I think it will be enough for me to run 25000mm/min.

Thanks for your help and to a great project.

hanke-cnc, 2023-07-05:
> Yes, I'm pushing it (to?) hard. Changed the suggested code and now it don't crash anymore.
>
> Trying to understand how far I can go. I have 600 steps/mm, changed "Step pulse time" to 2us. If then make [...]

F446 can run to 400KHZ, H743/H750 can run to 700KHZ, I think this is meaningless for CNC machine tools, because you need to pay more money to buy a processor

dagerlandsson, 2023-07-17:
Thanks for all you help. It was overrun of the stepper ISR. Closing this issue.

closed 2023-07-17 by dagerlandsson