grblHAL/STM32F4xx #129: Errors building GRBLHAL for STM32F4XX using directions from the Readme file

jbhome, 2023-08-02:
I am new to GRBLHAL. I have been reading up on it and decided to start by building the default system using the instrctions in "STM32F4xx grblHAL driver" readme.md file.
I installed platformIO core and the repo without a problem.
Running pio, yielded the list of targets. Great. It works.
Building with "pio run --environment nucleo_f411re_protoneer" failed though reporting a missing header file.
I found this file in the grblhal/core repo that was not installed using the "git clone https://github.com/grblHAL/STM32F4xx.git" command. Note the directory is STM32F4XX, not STM32F4. This should be changed as well or maybe mention to add the appropriate name extension "XX". Who says I'm picky. In my defence, it's these details that reduce the number of trouble calls/complaints once something is released.
I ended up downloading and copying the contents of the core folder to the grbl sub-folder of the STM32F4xx project folder and this allowed me to comple and link the project without error.
I am wondering if this is the right aproach, or if there is another set of instructions i missed and need to follow.
Either way, it would seem these instructions should be included in the readme file to support a complete error free first build and allow one to have a trouble free foundation to learn and build from.
Sorry if these sounds like a complaint. The work done here and in all the open source projects is truly amazing and I'm sure very much appreciated if not acknowledged by the silent majority.
Regards,
Jim Brooks

andrewmarles, 2023-08-02:
Probably would point you at the web builder which should skip all of the things you mention here as long as your board is supported.

http://svn.io-engineering.com:8080/

jbhome, 2023-08-02:
Thanks but I would prefer the cli version. I know what a dufus. Just trying
to keep the process noise down.

I looked at the other build options and I guess they are just not
maintained and going to the web based solution as well.

Perhaps change readies to point to the web guide or a common generic vuild
instruction page.
Sorry to be a pain but having all the different directions makes it very
confusing and.... leads to more questions from ppl like me, something that
gets in the way of the real work on the project

Regards,
Jim Brooks

dresco, 2023-08-02:
> I am wondering if this is the right approach, or if there is another set of instructions i missed and need to follow.

You're just missing the various common submodules that are linked from each driver repo. (If you look at the contents on GitHub, you'll see all the submodule folders listed there).

Something like `git clone --recurse-submodules https://github.com/grblHAL/STM32F4xx.git` should do it..

jbhome, 2023-08-03:
Thanks the heads up Jon,

It did the trick. Should be in the instructions.
I know the web builder is easier.
I just have mixed feelings about the web builder, vscode etc. it hides so
much from you and is another layer of detail to learn.

Regards,
Jim Brooks

terjeio, 2023-08-06:
Hmm, I wonder if providing the _platformio.ini_ file generated by the Web Builder for download would be better than trying to keep the ones in the repo up to date. Or are someone willing to take on the job of mantaining and testing platformio build configurations?

terjeio, 2025-11-26:
The .ini file used by the Web Builder can now be [added to the download](https://github.com/grblHAL/core/wiki/Compiling-grblHAL#platformio).