grblHAL/STM32F4xx #131: FYSETC S6 with TMC2209

terjeio, 2023-08-05:
### Discussed in https://github.com/grblHAL/STM32F4xx/discussions/130

<div type='discussions-op-text'>

<sup>Originally posted by **hartmutgw2** August  2, 2023</sup>
Hello,

since weeks I try to get my MPCNC with Fysetc S6 V2.1 and Fysetc TMC2209 V3.1 drivers working.
Without TRINAMIC_ENABLE=2209 it works, however I cannot define a hold current for Z stepper and due to weight of the spindle this configuration doesn't work for me.

When TMC2209 is enabled the board does not show up with lsusb so no communication to controller.
Jumper for UART communication to stepper drivers is set on the board and the controller worked with klipper and tmc2209 support before so I assume it is not hardware related.

If I flash the non-Trinamic version it works again and I'm wondering what I've missed.

Anybody here with a similar configuration or any hint what I'm doing wrong?

Thanks,
Hartmut
</div>

hartmutgw2, 2023-08-05:
Many thanks for your help @terjeio !!!
Merged with my pin mapping I'm able to communicate.
note that PROBE and RESET use the same PORT/PIN in fysetc_s6_map.h

M122
```
[TRINAMIC]
                      X       Y       Z
Driver          TMC2209 TMC2209 TMC2209
Set current           0       0       0
RMS current        1767    1767    1767
Peak current       2498    2498    2498
Run current       31/31   31/31   31/31
Hold current      15/31   15/31   15/31
CS actual         15/31   15/31   15/31
PWM scale            15      14      16
--- snip ---
```

I plan to test the new version in the next week.

Hartmut

terjeio, 2023-08-06:
Great, I'll assign _Reset_ to THERM1 in the next commit. Thanks for reporting.

hartmutgw2, 2023-08-06:
Do you think THERM1 is a good idea as this is an analog port?

I use 5 steppers (dual ganged X&Y) and 6 endstops (X-,Y-,Z-,X+,Y+,Z+)
so as I already use all of the endstop headers I will  use one expansion header for Probe, Reset, Hold and Start

terjeio, 2023-08-08:
> Do you think THERM1 is a good idea as this is an analog port?

It is a digital port until reconfigured. There is a large capacitor to ground that might cause issues, mainly "slow" restart if connected to a NC switch. It may also cause the switch to wear out a tad faster from the discharge current...

closed 2023-08-26 by terjeio