grblHAL/STM32F4xx #134: Compiling error with FatFS  '_VOLUMES' undeclared here.

Am0k-GIT, 2023-08-17:
Good afternoon. I'm trying to compile the firmware of the current version 20230816 with SD support. I encountered the following error:

```
In file included from FatFs\STM\ff_gen_drv.c:19:0:
FatFs\STM\ff_gen_drv.h:59:42: error: '_VOLUMES' undeclared here (not in a function); 
did you mean 'FF_VOLUMES'?
   uint8_t                 is_initialized[_VOLUMES];
```

Here is my configuration for PlatformIO:

```
[env:Halcyon_v1]
board                = genericSTM32F401RC
board_build.ldscript = STM32F401RC_FLASH.ld
build_flags          = ${common.build_flags}
  # See Inc/my_machine.h for options
                     -D BOARD_HALCYON_V1=
                     -D HSE_VALUE=8000000
                     -D USB_SERIAL_CDC=1
                     -D ODOMETER_ENABLE=1
                     -D SDCARD_ENABLE=1
lib_deps             = ${common.lib_deps}
          eeprom
          trinamic
          FatFs
          sdcard
lib_extra_dirs       = ${common.lib_extra_dirs}

[env:Halcyon_v1_dfu]
extends              = env:Halcyon_v1
upload_protocol      = dfu

[env:Halcyon_v1_stlink]
extends              = env:Halcyon_v1
upload_protocol      = stlink
```

If I modify the file FatFs\STM\ff_gen_drv.h (add #define _VOLUMES 1) everything compiles.

Where should I set the _VOLUMES parameter so as not to modify the library ?

terjeio, 2023-08-18:
Add
`-D _VOLUMES=1`
to _build_flags_ in platformio.ini.

Am0k-GIT, 2023-08-18:
Thank you very much.

closed 2023-08-18 by Am0k-GIT