grblHAL/STM32F4xx #138: STM32F411 Plugin_SD_card How to enable

auxp, 2023-09-02:
I found a piece of stm32f411 today, compiled with cnc3040. I don't know how to enable SD card

terjeio, 2023-09-02:
You cannot, pins are not assigned for it. A new board map has to be made that defines a SPI port and a pin for SD card chip select (CS). [An example](https://github.com/grblHAL/STM32F4xx/blob/d5c03e63ba41ac054ccfe9a29abcb6d422e53922/Inc/fysetc_s6_map.h#L153-L157), if the same pins are assigned those already mapped to these has to be remapped.

auxp, 2023-09-03:
I added it on CNC 3040
#if SDCARD_ENABLE
#define SD_CS_PORT                  GPIOA
#define SD_CS_PIN                   11
#define SPI_PORT                    1 // 1  GPIOA, SCK_PIN = 5, MISO_PIN = 6, MOSI_PIN = 7 (EXP2 Header)
#endif
my_machine.h change #define SDCARD_ENABLE        2
Connect to the corresponding interface Not working？
transmitter ioSender XL 2.0.42
May I ask what is the reason for this? Where else do I need to make changes?

terjeio, 2023-09-03:
> May I ask what is the reason for this?

Because you have not remapped the pins you want to use for SPI? E.g. `GPIOA 5` is originally mapped as `Z_DIRECTION_PIN`, this has to be changed to different unused pin.

closed 2023-10-12 by auxp