grblHAL/STM32F4xx #139: BTT_SKR_Pro v1.2 onboard sd_card

Melkiyby, 2023-09-06:
Hello. I'm trying to set up sd card support. I figured out the problems during compilation, but still an error 60 mounting the SD card.
I commented the exp2 part and uncommented it from the onboard
`//#if SDCARD_ENABLE
// Connect to the  EXP2 Header
//#define SD_CS_PORT                  GPIOB
//#define SD_CS_PIN                   12
//#define SPI_PORT                    2 // GPIOB, SCK_PIN = 13, MISO_PIN = 14, MOSI_PIN = 15
//#endif

//*
#if SDCARD_ENABLE
// On board socket- flaky?
#define SD_CS_PORT                  GPIOA
#define SD_CS_PIN                   4
#define SPI_PORT                    11 // (MCU port 1) GPIOA: SCK_PIN = 5, MISO_PIN = 5, GBIOB: MOSI_PIN = 5
// PB11 - TF_SW - card detect
#endif
//*/`

asulonn, 2023-09-13:
I had the same issue, with the same board (SKR PRO V1.1). 
Unfortunately at the time i misplaced my issue under the core repo instead of here. 
I did some testing, but at some point i got busy doing other stuff and i did not have the time to further investigate.
Meaby this info can help you and/or you can finish where i stopped:
#https://github.com/grblHAL/core/issues/244

My plan was to one day pick this back up and go further with testing...

terjeio, 2023-09-13:
There is a typo in [spi.c](https://github.com/grblHAL/STM32F4xx/blob/51e5e4fb73734a74e2cfc87fde1a02849b451e48/Src/spi.c#L293)... `5` should be `GPIO_PIN_5`.

konikminavto, 2025-10-08:
Hi. I checked. Now the onboard card works, but there's no "OK" response when I type $FM.
Previously, I used the card through a display with a card reader, and it was very difficult to initialize the card; I had to reboot the board 15-20 times.

terjeio, 2025-10-08:
@konikminavto There is this comment in the board map: _// On board socket- flaky?_ I do not have a board to test with so cannot dig into this further. What I know is that the SD card code is working with other STM32F4xxx boards so I suspect a hardware issue.
BTW is your card <= 32Gb?. 32Gb is the max supported size.

closed 2025-10-08 by terjeio