grblHAL/STM32F4xx #142: Synchronous spindle related

auxp, 2023-10-12:
Hello!stm32f411ceu6 How to configure spindle synchronization? Which motors are available for selection

terjeio, 2023-10-13:
> How to configure spindle synchronization?

You are defining your own pin mapping?
A15 or D2 can be used as spindle pulse input, for spindle index use the pin of your choice (must be interrupt capable). Enable spindle sync in [my_machine.h](https://github.com/grblHAL/STM32F4xx/blob/d65df3051cedda74ba2140f189d334dfc3f9d88e/Inc/my_machine.h#L89C1-L89C1), configuration info is [found here](https://github.com/grblHAL/core/wiki/Spindle-sync).

> Which motors are available for selection

I do not understand this. You can define as many motors (up to 6) as you have free pins for. The spindle motor can be either PWM controlled or a Modbus controlled VFD. Using a stepper motor for the spindle is not yet possible but might be added later.

FYI there are some board definitions that has the necessary pin definitions. The Web Builder can be used to build the firmware with spindle sync enabled.

auxp, 2023-10-13:
> 主轴电机可以是 PWM 控制的，也可以是 Modbus 控制的 VFD。主轴使用步进电机尚不可能，但稍后可能会添加。
> 
> 仅供参考，有些板定义具有必要的引脚定义。Web Builder 可用于构建启用主轴同步的固件。

Synchronous motor for lathe? PWM is 0-10V？How to do it specifically? Do you have a template

terjeio, 2023-10-13:
> Synchronous motor for lathe?

You do not need that, RPM and angular position is derived from encoder inputs.

> PWM is 0-10V？

No, default is a 5KHz pulse-width modulated (PWM) signal. STM32 devices are 3.3V. You need a low-pass filter and a gain stage to transform it to a 0-10V analog signal. PWM to analog converters are, AFAIK, [readily available](https://duckduckgo.com/?t=ffab&q=pwm+to+analog&ia=web).

> Do you have a template

For how to connect everything?

MCULJ, 2025-05-28:
Regarding G33, is it normal that the spindle has reached speed during the spindle synchronization process while the stepper motor is still accelerating and decelerating?