grblHAL/STM32F4xx #144: Octopus Pro F429 - will it work?

nameisCruzCruz, 2023-11-16:
Hi guys! very new to the grblHAL world, and all the CNC world for that matter. Building a Mill machine currently and I would like to use an octopus board I have laying around, the octopus pro F429, although I don't know where to start, the page says F4XX but then mentions "A grblHAL driver for the STM32F401xC, STM32F407xx, STM32F411xE and STM32F446xx ARM processors." so I'm confused...

nameisCruzCruz, 2023-11-16:
> It seems so, @yoft has it running. See [this discussion](https://github.com/grblHAL/STM32F4xx/discussions/141#discussioncomment-7296884).

Do you know if the firmware.bin file is available somewhere? I'm very lost and wanted to give a shot at grblHAL

nameisCruzCruz, 2023-11-16:
https://github.com/yoft/STM32F4xx/tree/new-octo This seems to be his repository but I don't see the .pio folder, am I missing something?

yoft, 2023-11-16:
Hi @nameisCruzCruz,
I haven't updated the repo on gihub, however I have the F446 variant, which will be quite different.

I can attempt an F429 build using the same process I went through with the F446. I might have some time late next week to have a crack at it for you.

nameisCruzCruz, 2023-11-16:
That would be amazing if you have some spare time, can I have your email? So we can talk further?

nameisCruzCruz, 2023-11-19:
> Hi @nameisCruzCruz, I haven't updated the repo on gihub, however I have the F446 variant, which will be quite different.
> 
> I can attempt an F429 build using the same process I went through with the F446. I might have some time late next week to have a crack at it for you.

Did you manage to do it?

yoft, 2023-11-20:
> That would be amazing if you have some spare time, can I have your email? So we can talk further?

I'd prefer to keep discussion here for everyone to see.

> > Hi @nameisCruzCruz, I haven't updated the repo on gihub, however I have the F446 variant, which will be quite different.
> > 
> > I can attempt an F429 build using the same process I went through with the F446. I might have some time late next week to have a crack at it for you.
> 
> Did you manage to do it?

I haven't had time yet, however I have found most of the settings required.
When I get a chance I'll send through a firmware.bin and you can give it a test?
Some of the clock settings are missing which are required for usb connection, I think I can work out what they are, but do you have a usb-uart adapter for the first build? Then we can work on usb after that.

nameisCruzCruz, 2023-11-21:
that would be amazing, I'll be waiting and let you know if it works once you have the file. Thank you for your time

ashokmanker, 2024-05-08:
I have BTT octopus pro 429 board. Please help in compiling grblhal for this board.

yoft, 2024-05-08:
Have a look [https://github.com/grblHAL/STM32F4xx/discussions/141#discussioncomment-7398013](here) for progress.
Thanks,