grblHAL/STM32F4xx #145: GRBL for STM32F407 ，I want to use

mr-huang-mou, 2023-11-20:


mr-huang-mou, 2023-11-20:
I cannot see about Serial Init in main.c

mr-huang-mou, 2023-11-20:
And how to startup IDLE Status in Candle2

terjeio, 2023-11-20:
> I cannot see about Serial Init in main.c

Serial port(s) are registered in `Src/driver.c`, if the board is not a Nucleo board then USB is the default comms channel. You can disable USB in by commenting out [this line my_machine.h](https://github.com/grblHAL/STM32F4xx/blob/e05ca77100338d5ea6ecb4de432a721f5ecd8e82/Inc/my_machine.h#L60).

> And how to startup IDLE Status in Candle2

Ensure that the [controller is not starting in Alarm state](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings)?

mr-huang-mou, 2023-11-20:
> Serial port(s) are registered in `Src/driver.c`, if the board is not a Nucleo board then USB is the default comms channel. You can disable USB in by commenting out [this line [...]

([image])
Although disable this line ,I also didn't make STM32F407 chip to communicate with my host computer（Candle2） 

The status is Port opened in Candle2, not Alarm state 
[image]

terjeio, 2023-11-20:
What kind of board do you have?
Your own design?
It is important that the build you select matches the properties of the board, most importantly the MCU xtal frequency.

mr-huang-mou, 2023-11-20:
> What kind of board do you have? Your own design? It is important that the build you select matches the properties of the board, most importantly the MCU xtal frequency.
Currently,I want to know how to resolve error:79, and did the MCU must be connected the motor drive ? I didn't connect except serial port and power .I followed your First Run Grbl Settings but now the state still is alarm and send $X it show me error:79. I am a beginner and thank you for your help.

mr-huang-mou, 2023-11-21:
> > the state still is alarm and send $X it show me error:79
> 
> This is likely due to the reset or estop input beeing in the asserted state. Have you tried with setting `$14=70` and resetting the controller? If this does not help what is the response when you send `?`

send `?` and receive is <Alarm|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:P> .

terjeio, 2023-11-21:
> send ? and receive is <Alarm|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:P> .

Looks good, if you clear the alarm by sending `<CTRL>+<X>` followed by `$X` then what is the state?

mr-huang-mou, 2023-11-21:
I don't know whether sending success,send `Ctrl +X` and back package was `error:9`,and `$X` was `error:79`

terjeio, 2023-11-21:
Set `$10=1535` and reboot, then post a `?` response so I can see the reason for the alarm.

terjeio, 2023-11-21:
> but how to set $50~$55 ? I couldn't set ,and didn't find in code .

By adding the [keypad plugin](https://github.com/grblHAL/Plugin_keypad) to the build.

mr-huang-mou, 2023-11-22:
> 
> By adding the [keypad plugin](https://github.com/grblHAL/Plugin_keypad) to the build.

Now,I have a new problem. I sent `G1 X10 F100` and receive `ok` 
but it didn't move and I also sent `?` receive `<Idle|MPos:0.524,1.372,0.000|Bf:34,1023|FS:0,0|Pn:PXYZ>`
but it doesn't change `<Idle|MPos:0.524,1.372,0.000|Bf:35,1023|FS:0,0|Pn:PXYZ|Ov:100,100,100>` 
 I am not sure whether something I didn't set.Oh,I also sent `$J=G53 Y1.0 F10` and `?` ,it show me `<Jog|MPos:0.608,1.372,0.000|Bf:32,1023|FS:78,0|Pn:PXYZ>` ,it moved,but it didn't act as expected.

terjeio, 2023-11-22:
Have you [calibrated your motors](https://duckduckgo.com/?t=ffab&q=grbl+calibrate+motors&ia=web)? The step/mm settings are important for correct movement.
Stepper driver and motor wiring is ok?

mr-huang-mou, 2023-11-22:
> Have you [calibrated your motors](https://duckduckgo.com/?t=ffab&q=grbl+calibrate+motors&ia=web)? The step/mm settings are important for correct movement. Stepper driver and motor wiring is ok?

I haven't wire motor and Stepper driver.Temporarily only MCU is used for testing.If needs,I can wire.Stepper driver is TB6600 and motor is 42BYGH48-401A.Other than that, do I need to modify in the code?

terjeio, 2023-11-22:
> Other than that, do I need to modify in the code?

No, code modifications are seldom required.

mr-huang-mou, 2023-11-22:
> > Other than that, do I need to modify in the code?
> 
> No, code modifications are seldom required.

do I need wire motor,Stepper driver and MCU ?

mr-huang-mou, 2023-11-22:
I sent `G1 X10 F100` and `?`,but received 
`ok`
`<Idle|MPos:0.000,0.000,0.000|Bf:34,1023|FS:0,0|Pn:PXYZ>`
it didn't run.
I have no idea to fix it.

mr-huang-mou, 2023-11-22:
this is my set
$0=10.0
$1=255
$2=0
$3=0
$4=7
$5=0
$6=0
$7=0
$10=511
$11=0.010
$12=0.002
$13=0
$14=7
$15=0
$16=0
$17=0
$18=0
$19=0
$20=0
$21=0
$22=0
$23=0
$24=25.0
$25=500.0
$26=250
$27=1.000
$28=0.100
$29=0.0
$30=1000.000
$31=0.000
$32=0
$33=5000.0
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=0
$43=1
$44=4
$45=3
$46=0
$50=100.0
$51=600.0
$52=3000.0
$53=0.250
$54=500.0
$55=3000.0
$62=0
$63=2
$64=0
$65=0
$100=250.000
$101=250.000
$102=250.000
$110=500.000
$111=500.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=100.0
$384=0

terjeio, 2023-11-22:
> I have no idea to fix it.

And I have no idea about what is going on.
Are you using Windows? If so can you [download ioSender](https://github.com/terjeio/ioSender/releases) and use that to communicate with the board?

mr-huang-mou, 2023-11-22:
Why do I need to cycle the `$J=G53 X100 F500` command lots of times to finish jog ?
And how do I make `G1 X10 F100` command work?

terjeio, 2023-11-22:
There is something seriously wrong with your build, but I do not know what.
How do you build the firmware? Using the Web Builder or STM32CubeIDE or something else?
If the STM32CubeIDE have you selected a build for the correct MCU? There are many combinations available from the _Hammer_ tool i the toolbar.
How do you flash the controller?
And post the output from the `$I` command.

mr-huang-mou, 2023-11-22:
> There is something seriously wrong with your build, but I do not know what. How do you build the firmware? Using the Web Builder or STM32CubeIDE or something else? If the STM32CubeIDE have you [...]

Using STM32CubeIDE and selecting `Debug F407 8Mhz`
`$I` command show that
[VER:1.1f.20220625:]
[OPT:VNMSL,35,1024,3,0]
[NEWOPT:ENUMS,RT+,TC,SED]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32F407]
[DRIVER VERSION:130522]
[PLUGIN:KEYPAD v1.34]
ok
I thought that maybe `DRIVER VERSION` was old.

mr-huang-mou, 2023-11-22:
I have new problem in using the lastest edition, I use to the new version to test ,but it show me `alarm: 10` I am not sure what trigger Estop.

terjeio, 2023-11-23:
> I am not sure what trigger Estop.

Short the estop (reset) pin to ground (PB5 in the generic map) or invert the input with $14.

mr-huang-mou, 2023-11-23:
> > I am not sure what trigger Estop.
> 
> Short the estop (reset) pin to ground (PB5 in the generic map) or invert the input with $14.

Are both estop pin and reset pin in PB5 ?

mr-huang-mou, 2023-11-23:
> > I am not sure what trigger Estop.
> 
> Short the estop (reset) pin to ground (PB5 in the generic map) or invert the input with $14.

Thank you for help, I tried to short the estop pin and it can start ,but invert the input was still alarm.

terjeio, 2023-11-23:
Did you invert using the correct bitmap? Send `$$=14` to get which values to use, or use ioSender to change the setting.

mr-huang-mou, 2023-11-24:
> Did you invert using the correct bitmap? Send `$$=14` to get which values to use, or use ioSender to change the setting.

I set `$14=71` and view the settings ,it's true of `$14=71` .I also tried to use ConfigApp to change the settings.

mr-huang-mou, 2023-11-24:
I have new problem .how can I do to release the hold state When I send `M0` command .

terjeio, 2023-11-24:
> I have new problem .how can I do to release the hold state When I send M0 command .

Send cycle start: `~`

mr-huang-mou, 2023-11-24:
> > I have new problem .how can I do to release the hold state When I send M0 command .
> 
> Send cycle start: `~`

Thank you ，this is very helpful to me .

mr-huang-mou, 2023-11-27:
Hello,terjeio
I had a new problem ,how should I set the origin coordinates of the mechanical coordinate system?
`G92 X0 Y0 Z0` can't set the origin coordinate.

terjeio, 2023-11-28:
> how should I set the origin coordinates of the mechanical coordinate system?

By homing the machine? By default the origin is in negative space (from max travel settings) but can be set to 0 by a homing option flag ($22):

[image]
`$$=22` outputs:
```
$22: Homing cycle as bitfield where setting bit 0 enables the rest:
    0 - Enable (1)
    1 - Enable single axis commands (2)
    2 - Homing on startup required (4)
    3 - Set machine origin to 0 (8)
    4 - Two switches shares one input pin (16)
    5 - Allow manual (32)
    6 - Override locks (64)
    7 - Keep homed status on reset (128)

Enables homing cycle. Requires limit switches on axes to be automatically homed.

When `Enable single axis commands` is checked, single axis homing can be performed by $H<axis letter> commands.

When `Allow manual` is checked, axes not homed automatically may be homed manually by $H or $H<axis letter> commands.

`Override locks` is for allowing a soft reset to disable `Homing on startup required`.
```

Use G54 - G59.3 to set offsets, not G92 unless you understand how G92 works.