grblHAL/STM32F4xx #146: webbuilder and autosquaring skr pro 1.2

zurdoid123, 2023-11-20:
Hello, So i'm new in this sphere of grblhal, have been trying to read up but cant really get my head around some things, mainly how to set up auto-squaring.

I'm trying to use the webbuilder to compile the firmware.bin but if I select auto-squaring on both Z and Y, which is where I need it, it says "AUTO SQUARE: available 1, required 2." and if I only choose Z I'm allowed to build but it fails and gives the error 

"Command failed: pio run -etest -cjob.ini
In file included from Src/driver.c:34:0:
./grbl/motor_pins.h:469:4: error: #error "Auto squared Z-axis requires second limit pin input"
#error "Auto squared Z-axis requires second limit pin input"
^~~~~
*** [.pio/build/test/src/driver.o] Error 1"

On Y axis the build completes if I set auto-squaring.

Also i'm planning to use external drivers TB6600, my idea is to simply use the pins from the standard sockets and not the uart pins, shouldnt be a problem? Or can I simply use the uart pins without changing the code? As I'm obviously isn't a very good swimmer in the world of coding, even if I'm not completely behind, I'm trying to keep it as simple as possible haha

Thank you beforehand for any pointers!

Philip

terjeio, 2023-11-20:
> which is where I need it, it says "AUTO SQUARE: available 1, required 2.

grblHAL cannot use the limit input for the 5th motor (M4/E1) since [the pin conflicts with the X limit pin](https://github.com/grblHAL/STM32F4xx/blob/e05ca77100338d5ea6ecb4de432a721f5ecd8e82/Inc/btt_skr_pro_v1_1_map.h#L105-L106). This is a recurring problem with some 3D printer boards since the 3D printer firmware polls limit switches and Grbl/grblHAL uses interrupts for monitoring them. The limit switch for the 6th motor (M5/E2) could be used instead to solve this problem.

> ... if I only choose Z I'm allowed to build but it fails and gives the error

I am not able to replicate this, can you try again?

> Also i'm planning to use external drivers TB6600, my idea is to simply use the pins from the standard sockets and not the uart pins, shouldnt be a problem? Or can I simply use the uart pins without changing the code?

I am sorry, I am not sure I understand this. You are talking about step/dir/enable signals? If so they can be wired from the stepper sockets. The UART pins has nothing to do with these.

zurdoid123, 2023-11-20:
Okey, I see.  How would I make the change to use e1 pin if you dont mind me asking? Would build it in the web-builder and then simply change the file? Tried to open it before quickly but didnt succeed, do I need to use platformio?

I tried it multiple times and every time the same error. And now again the same, x-single, y-ganged and z auto-squared give error I wrote. 

Yeah, sorry. I should've been more clear. But yeah thats what I wanted to confirm, thanks.

terjeio, 2023-11-21:
I have now switched to use the E2 limit input for the E1 motor (M4). I've put a note in the board map about the change, and you should now be able to build with the Web Builder.

zurdoid123, 2023-11-21:
Ah wow, It indeed worked to build now! Still haven't had time to try and load it but I assume it will work out just fine :) 

Thanks alot!

zurdoid123, 2023-11-23:
Hey again.

So i finally got around to trying to flash the board and moved the firmware.bin to a sd card and started up the controller after moving over the sd card. No light is flashing and the file doesn't change name as i've read it should. i tried both the bootloader version and the other version. 

Any ideas?

terjeio, 2023-11-23:
The bootloader has been overwritten?
If you try to flash 3D printer firmware does that change the filename?

zurdoid123, 2023-11-28:
Im not sure, I haven't managed to write anything as far as I know.  I tried the firmwaee thats available in the btt github aswell, also didn't work.

terjeio, 2023-11-29:
Sounds like there is no bootloader on the MCU. You may have to get yourself a ST-Link for programming.

A few things to check:
Can you connect to the controller?
Does it respond to gcode commands?

zurdoid123, 2023-11-30:
Yeah thats what I suspected and have got myself an st-link and are about to give it a go.

For now it doesn't connect to any software. It does show up in device manager though but just like "Usb Serial Device" a bit wierd is that it doesn't matter what port I put it in it always says COM 6..

zurdoid123, 2023-11-30:
So I managed to flash it with the st-link using the reset file from the BTT github, the computer detects it and all is good. Tried a few times to flash the bootloader version from the web-builder but it's not recognized by the computer after doing so, no connection sound and doesnt show in device managed either.

I then flashed it again with the reset firmware and tried the SD-card method. Now the green light flashed for a while and the file changed format as i've read it should do. But the board is again not being detected by the computer.

closed 2023-12-15 by terjeio