grblHAL/STM32F4xx #149: Devtronic CNC Controller support

nickshl, 2023-12-13:
Hi @terjeio, I would like to add support for this board: https://github.com/grblHAL/core/discussions/406

> The listed changes cannot be added to the current map file without breaking backwards compatibility, so you will have to add a new one.

The question is: Do we even need old configuration? Is anyone but me used it?

> Making the safety door input default in the Web Builder is currently not possible. However you may force it by undefining/defining SAFETY_DOOR_ENABLE 1 in the map file.

> Again, force it the map file by undefining/defining SPINDLE_SYNC_ENABLE 1.

I don't like this idea. Defining SAFETY_DOOR_ENABLE or SPINDLE_SYNC_ENABLE  in map file will make those settings in Web Builder useless, which will be confusing. If it isn't possible right not - that is fine.

> Either force FRAM in the map file or set it as default in the entry for the board in [driver.json]

I would like to do it via json file.

> Add it to the new map file.

I created map file: [Devtronic_CNC_Controller_map.zip](https://github.com/grblHAL/STM32F4xx/files/13664636/Devtronic_CNC_Controller_map.zip)
It also contains this:
```
#if MPG_MODE == 1
#define MPG_MODE_PORT           GPIOA
#define MPG_MODE_PIN            10
#define MPG_POLARITY            1 // High - need to switch to MPG mode
#define MPG_MODE_POOLING        // Pooling mode - no iterrupt used to switch MPG mode
#define MPG_MODE_NOINIT         // Do not initialize hardware(shared with UART RX pin)
#endif
```
Which should be used in conjunction with changes I described in this issue: https://github.com/grblHAL/core/issues/327#issuecomment-1854519560
I don't see any option for MPG_MODE in Web Builder. For this configuration it should be set 1.

terjeio, 2023-12-13:
> The question is: Do we even need old configuration? Is anyone but me used it?

I do not know, so better not removing it.

>> Either force FRAM in the map file or set it as default in the entry for the board in [driver.json]

> I don't like this idea. Defining SAFETY_DOOR_ENABLE or SPINDLE_SYNC_ENABLE in map file will make those settings in Web Builder useless, which will be confusing. If it isn't possible right not - that is fine.

For now it is not possible - Web Builder code changes and new parameters in driver.json are required.

> I would like to do it via json file.

Defining the [size as negative](https://github.com/grblHAL/STM32F4xx/blob/5f601d9fe16087ce65b62b2f2dcc791dadebeeea/driver.json#L266-L267) locks the dropdown in the _Plugins_ tab, [positive](https://github.com/grblHAL/STM32F4xx/blob/5f601d9fe16087ce65b62b2f2dcc791dadebeeea/driver.json#L266-L267) makes it overridable.

> I don't see any option for MPG_MODE in Web Builder.

Again Web Builder code changes and new parameters in driver.json are required.  To be added...

nickshl, 2023-12-15:
@terjeio, I did Pull Request with configuration for this board. It does not contain MPG changes. Please check and merge it if all looks good to you.

nickshl, 2024-01-23:
Hi @terjeio, I have a question about ModBus spindle: does it need only UART? It looks like BlackPill has two available UARTs: one already used for MPG and another can be used for ModBus... if spindle need only ModBus. The idea to change pinout a little bit to allow Spindle En & Spindle Dir be UART capable. Then there will be possible to use it as Spindle En, Spindle Dir and Spindle PWM, or as UART TX, UART RX and UART DIR to connect RS-485 transceiver.

nickshl, 2024-02-21:
Hi @terjeio , could you answer my question above? 

I didn't sold a single board so far, may be people really want to see optocouplers, or may be this form-factor isn't interesting to anyone. Anyway, I am considering(_don't want to spend time & money to boards that probably no one need - first revision works good for me so far_) to spin another board revision with optocouplers. But someone on reddit ask me about ModBus spindle and I realized that it technically can be added too. PA2 & PA3 can be used as UART2 on F411(_UART1 is used for MPG_). I can move Step & Dir pins and move Spindle Enable & Spindle Direction to PA2 & PA3 - in this case will be possible to configure those as Spindle  En/Dir or TX/RX for RS-485. And PA8(_Spindle PWM, looks like F411 doesn't have dedicated hardware DE pin like STM32G0_) can be used as direction for RS-485 transceiver. I don't see any problems with this solution. What do you think?

So, basically this is layout I considering right now:

```
/* Pin Assignments(v2 Support ModBus spindle):
 *
 *                          B Direction -   - B Step
 *                                       | |
 *                                     G S D +
 *                                     N C I 3
 *                                     D K O V
 *                                   -----------
 *                               VB |           | +3V
 *                 Step En/Dis  C13 |           | GND
 *               Coolant Flood  C14 |           | +5V
 *                Coolant Mist  C15 | *     - * | B9   I2C SDA
 *                              RST |      |K|  | B8   I2C SCL
 *                       Reset   A0 |       -   | B7   Feed Hold
 *      Safety door/I2C strobe   A1 |           | B6   Cycle Start
 *    Spindle Enable/RS-485 TX   A2 |           | B5   Probe
 * Spindle Direction/RS-485 RX   A3 |    / \    | B4   Encoder Index
 *                      X Step   A4 |   <MCU>   | B3   Encoder Pulse (B)
 *                 X Direction   A5 |    \ /    | A15  Encoder Pulse (A)
 *                      Y Step   A6 |           | A12  USB D+
 *                 Y Direction   A7 |   -   -   | A11  USB D-
 *                      Z Step   B0 |  |R| |B|  | A10  USART1_RX
 *                 Z Direction   B1 |   -   -   | A9   USART1_TX
 *                      A Step   B2 |           | A8   Spindle PWM/RS-485 DE
 *                 A Direction  B10 |           | B15  A Limit
 *                              +3V |   -----   | B14  Z Limit
 *                              GND |  |     |  | B13  Y Limit
 *                              +5V |  | USB |  | B12  X Limit
 *                                   -----------
 */
```

terjeio, 2024-02-21:
If using a [auto-direction](https://duckduckgo.com/?t=ffab&q=rs485+auto+direction&ia=web) tranceiver only a UART is needed. If not an additional pin for direction switching is required.

The only issue with sharing the spindle dir/enable pins with the UART is that it only allows PWM or Modbus to be selected at compile time. Some wants both - having a Modbus spindle and a laser available and switching between them without reflashing...

Tip: with [board specific code](https://github.com/grblHAL/ESP32/blob/master/main/boards/BlackBoxX32.c) you can get away with two axes sharing a limit switch input.

nickshl, 2024-02-21:
> The only issue with sharing the spindle dir/enable pins with the UART is that it only allows PWM or Modbus to be selected at compile time. Some wants both - having a Modbus spindle and a laser available and switching between them without reflashing...

When there no enough pin, some compromises have to be made. And most important: this issue can be fixed later in software without hardware changes. For example if laser module need En and PWM, and grblHAL can produce software PWM, B-axis can be reconfigured to control laser module. Or coolant outputs and Stepper En/Dis(_absolutely useless signal in my opinion_) can be reconfigured to control laser module. Whichever suited best for particular application.

Maybe someday something similar to Betaflight Resource Remapping will be added into grblHAL: https://betaflight.com/docs/wiki/guides/current/Resource-remapping

terjeio, 2024-02-22:
> Maybe someday something similar to Betaflight Resource Remapping will be added into grblHAL

Some bits already in the next release, more later. But only for what I regard as non-critical pins. E.g. I want step/dir pin output to be as close to bare metal as possible for speed - I do not want to add indirection for those.
I am using extended [crossbar](http://svn.io-engineering.com/grblHAL/html/crossbar_8h.html) and [ioports](http://svn.io-engineering.com/grblHAL/html/ioports_8h.html) APIs for it, which also allows plugin code to claim pins.

nickshl, 2025-02-28:
Hi @terjeio, is there a way to use A motor as second Y motor and A limit switch as second Y limit switch to use auto square feature?

terjeio, 2025-02-28:
Uncomment [this line](https://github.com/grblHAL/STM32F4xx/blob/dac05996ecec443d851b0d138dffcd64d1022106/Inc/my_machine.h#L145) and recompile.

nickshl, 2025-03-01:
> Uncomment [this line](https://github.com/grblHAL/STM32F4xx/blob/dac05996ecec443d851b0d138dffcd64d1022106/Inc/my_machine.h#L145) and recompile.

Will it work if in Web Builder for "Devtronic CNC Controller V2" select "Y-axis: Auto-squaired"? It is not for me, someone else ask me how to configure it. Using Web Builder would be way more convenient...

terjeio, 2025-03-01:
What is not working? The $pins command list the pins as it should.

nickshl, 2025-03-03:
Just tried it - it works! Thank you!

nickshl, 2025-04-04:
Hi @terjeio, spindle sync doesn't work anymore on my lathe. FS in status response show only two numbers, Commands $SD and $SR doesn't work. I wondering if I missed something:
```
GrblHAL 1.1f ['$' or '$HELP' for help]
<Idle|MPos:0.000,0.000,0.000|Bf:35,1024|FS:0,0|Pn:Y|Ov:100,100,100|TLR:0>
<Idle|MPos:0.000,0.000,0.000|Bf:35,1024|FS:0,0|Pn:Y|WCO:-13.000,0.000,0.000|WCS:G54|A:|Sc:|MPG:0|H:0|D:0|T:0|TLR:0|FW:grblHAL>
[VER:1.1f.20250329:]
[OPT:VNMSL,35,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,ES,MPG,LATHE,EXPR,SS,SED,PID]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FRAM 32K]
[FREE MEMORY:81K]
[DRIVER:STM32F411]
[DRIVER VERSION:250328]
[BOARD:Devtronic CNC Controller with Spindle Sync]
[AUX IO:2,2,0,0]
[PLUGIN:Bootloader Entry v0.03]
ok
$0=10.0
$1=25
$2=0
$3=0
$4=7
$5=5
$6=1
$9=1
$10=4607
$11=0.010
$12=0.002
$13=0
$14=70
$15=0
$16=0
$17=0
$18=0
$19=0
$20=0
$21=0
$22=0
$23=0
$24=25.0
$25=500.0
$26=250
$27=1.000
$28=0.100
$29=0.0
$30=2000.000
$31=0.000
$32=2
$33=5000.0
$34=0.0
$35=0.0
$36=100.0
$37=0
$38=2000
$39=1
$40=0
$43=1
$44=4
$45=3
$46=0
$62=0
$63=3
$64=0
$65=0
$90=5.000
$91=0.005
$92=0.000
$95=0.000
$100=1000.00000
$101=250.00000
$102=200.00000
$110=600.000
$111=500.000
$112=1000.000
$120=200.000
$121=10.000
$122=2000.000
$130=200.000
$131=200.000
$132=200.000
$340=90.0
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=200.0
$346=1
$370=4
$372=0
$384=0
$394=0.0
$398=35
$481=0
$484=1
$486=0
$534=0
$539=0.0
$673=1.0
$676=3
ok
[GC:G0 G54 G8 G17 G21 G90 G94 G97 G49 G98 G50 M5 M9 T0 F0 S0.]
ok
[G54:-13.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
ok
<Idle|MPos:0.000,0.000,0.000|Bf:35,1024|FS:0,0|Pn:Y|WCO:-13.000,0.000,0.000>
$pins
[PIN:PB7,Feed hold]
[PIN:PB6,Cycle start]
[PIN:PA0,Emergency stop]
[PIN:PB5,Probe]
[PIN:PB12,X limit min]
[PIN:PB13,Y limit min]
[PIN:PB14,Z limit min]
[PIN:PB4,Spindle index]
[PIN:PA15,Spindle pulse]
[PIN:PA1,Aux in 0,P0]
[PIN:PB15,Aux in 1,P1]
[PIN:PA2,X step]
[PIN:PA4,Y step]
[PIN:PA6,Z step]
[PIN:PA3,X dir]
[PIN:PA5,Y dir]
[PIN:PA7,Z dir]
[PIN:PC13,Steppers enable]
[PIN:PB2,Spindle on]
[PIN:PB10,Spindle direction]
[PIN:PA8,Spindle PWM]
[PIN:PC14,Mist]
[PIN:PC15,Flood]
[PIN:PB0,Aux out 0,P0]
[PIN:PB1,Aux out 1,P1]
[PIN:PB8,I2C SCK]
[PIN:PB9,I2C SDA]
[PIN:PA10,RX,MPG]
[PIN:PA9,TX,MPG]
ok
$SD
error:3 - '$' system command was not recognized or supported.
$SR
ok
error:3 - '$' system command was not recognized or supported.
```