grblHAL/STM32F4xx #152: BTT SKR2.0 Spindle Sync and HSM PostPro for Lathe

JoachimF, 2023-12-19:
Hello,

I just get a BTT SKR2, is there a reason that spindle sync is not available on this board? (tried in webbuilder)

2nd

The target is a lathe, what is the PP used for HSM, LinuxCNC Turn or other, GRBL Turn doesn't support G33

Regards

terjeio, 2023-12-19:
> is there a reason that spindle sync is not available on this board?

The encoder pins are not defined in the map file plus that the currently verified pins are used for other purposes on this board.
The spindle pulse pin has to be one that can be routed as clock input to a counter that is free (or can be swapped), such a pin obviously has to be determined from checking the board schematic against the MCU datasheet, Then at least the map has to be changed and then the change tested. Since I do not have a board for testing I have not done this.

> The target is a lathe, what is the PP used for HSM, LinuxCNC Turn or other, GRBL Turn doesn't support G33

Do you have links to these CAM programs? I can't find them when doing a quick search.

JoachimF, 2023-12-21:
The PP are here : https://cam.autodesk.com/hsmposts

You can perhaps code a PP for GRBLHAL

terjeio, 2023-12-28:
I do not have/use Fusion and have never written or modified a PP for it. It is better to ask somebody (via a forum?) that knows the product?