grblHAL/STM32F4xx #158: crashing/freezing

zurdoid123, 2024-02-05:
I have a reoccuring problem. Iosender keeps freezing and if i clock something, like trying to jog ir pressing stop, a messages shows up saying that the port is closed or that basestream isn't available. It seems to happen randomly and the program keeps running until a certain point. 

Sometimes i can finish a job of one hour, sometimes it stops after 10min. 

I have tried 3 different usb cables, one of them was working without problem on my old machine. 

I have turned of usb turn of in energy settings. 

Any ideas?

zurdoid123, 2024-02-05:
Ah whoops, now realisdd that im in the grblhal part. Sorry

terjeio, 2024-02-05:
EMI? Which board do you have and which settings?
Select _Help > About > To clipboard_ in ioSender and paste it in a comment.

zurdoid123, 2024-02-06:
I was also thinking that it's maybe EMI but I'm not sure what I can do to find prevent it anymore.

[FIRMWARE:grblHAL]
[NVS STORAGE:*EEPROM 2K]
[FREE MEMORY:91K]
[DRIVER:STM32F407]
[DRIVER VERSION:231228]
[BOARD:BTT SKR PRO v1.2]
[AUX IO:1,2,0,0]

terjeio, 2024-02-06:
Ok, that board is using native USB for the connection. I ran a 10 hour test on my bench yesterday on a F756 dev board that uses nearly the same codebase, without any issues.
Have you tried to dry run the gcode with the spindle disconnected? Still failing?

zurdoid123, 2024-02-06:
I'll try and get back to you. It won't be until next week though. I'm just using a makita router and the power cable goes straight up to the ceiling with the dust extraction so shouldn't be a big problem from what I've read. I will also try to plug it at the opposite side of the machine from the control box though.

zurdoid123, 2024-02-21:
Now I have done a longer time of testing and all issues have stopped since I rerouted the power cable for my router from the control box to the opposite corner of the table.

closed 2024-02-22 by terjeio