grblHAL/STM32F4xx #161: Ethernet only connect when usb is connected and used by sender

semos2k, 2024-02-20:
Hi, i using a board stm32f407vet6 and compiled version of the grblhal activating ethernet, that works correctly when connect first the usb cable and iosender for check boot.

But when not connect the usb cable and power the board from PSU try to connect iosender using the ip not work, testing using ping the ip reponds.

GrblHAL 1.1f ['$' or '$HELP' for help]
[VER:1.1f.20240204:]
[OPT:VNMSL,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,ES,SED,ETH]
[FIRMWARE:grblHAL]
[SIGNALS:HSE]
[NVS STORAGE:*FLASH]
[FREE MEMORY:30K]
[DRIVER:STM32F407]
[DRIVER VERSION:240205]
[AUX IO:1,0,0,0]
[WIZCHIP:W5500]
[MAC:56:34:12:dc:08:00]
[IP:192.168.10.135]
[NETCON:Websocket]
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]

thks.

terjeio, 2024-02-21:
How do you compile and do you have your own board map?

The reason I ask is that [this code](https://github.com/grblHAL/STM32F4xx/blob/019dc048242a669da8866ff7c01e21e60bfecac7/USB_DEVICE/App/usbd_cdc_if.c#L233-L234) was added by me to the ST framework to be able to detect if a USB client is connected or not. If it is missing it may explain why.

semos2k, 2024-02-21:
my platformio config is

> [env:black407ve]
> board = genericSTM32F407VET6
> board_build.ldscript = STM32F407VETX_FLASH.ld
> build_flags = ${common.build_flags}
>   -Wl,-u,_printf_float
>   -I lwip/src/include
>   -I networking/wiznet
>   -D HSE_VALUE=8000000
>   -D USB_SERIAL_CDC=1
>   -D ETHERNET_ENABLE
>   -D BOARD_BLACK_407
>   -D _WIZCHIP_=5500
>   -D TELNET_ENABLE=1
>   -D WEBSOCKET_ENABLE=1
> lib_deps = ${common.lib_deps}
>   eeprom
>   lwip
>   networking
> lib_extra_dirs = ${common.lib_extra_dirs}
> upload_protocol = stlink

and my custom map is 

> /*
>   generic_map.h - driver code for STM32F4xx ARM processors
> 
>   Part of grblHAL
> 
>   Copyright (c) 2020-2022 Terje Io
> 
>   Grbl is free software: you can redistribute it and/or modify
>   it under the terms of the GNU General Public License as published by
>   the Free Software Foundation, either version 3 of the License, or
>   (at your option) any later version.
> 
>   Grbl is distributed in the hope that it will be useful,
>   but WITHOUT ANY WARRANTY; without even the implied warranty of
>   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
>   GNU General Public License for more details.
> 
>   You should have received a copy of the GNU General Public License
>   along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
> */
> 
> #if N_ABC_MOTORS > 1
> #error "Axis configuration is not supported!"
> #endif
> 
> #if TRINAMIC_ENABLE
> #error "Trinamic plugin not supported!"
> #endif
> 
> #define SERIAL_PORT             1 // GPIOA: TX = 9, RX = 10
> 
> // Define step pulse output pins.
> #define STEP_PORT               GPIOA
> #define X_STEP_PIN              0
> #define Y_STEP_PIN              1
> #define Z_STEP_PIN              2
> #define STEP_OUTMODE            GPIO_MAP
> //#define STEP_PINMODE            PINMODE_OD // Uncomment for open drain outputs
> 
> // Define step direction output pins.
> #define DIRECTION_PORT          GPIOA
> #define X_DIRECTION_PIN         4
> #define Y_DIRECTION_PIN         5
> #define Z_DIRECTION_PIN         6
> #define DIRECTION_OUTMODE       GPIO_MAP
> //#define DIRECTION_PINMODE       PINMODE_OD // Uncomment for open drain outputs
> 
> // Define stepper driver enable/disable output pin.
> #define STEPPERS_ENABLE_PORT    GPIOA
> #define STEPPERS_ENABLE_PIN     15
> //#define STEPPERS_ENABLE_PINMODE PINMODE_OD // Uncomment for open drain outputs
> 
> // Define homing/hard limit switch input pins.
> #define LIMIT_PORT              GPIOB
> #define X_LIMIT_PIN             12
> #define Y_LIMIT_PIN             13
> #define Z_LIMIT_PIN             14
> #define LIMIT_INMODE            GPIO_SHIFT12
> 
> // Define ganged axis or A axis step pulse and step direction output pins.
> #if N_ABC_MOTORS == 1
> #define M3_AVAILABLE
> #define M3_STEP_PORT            STEP_PORT
> #define M3_STEP_PIN             3
> #define M3_DIRECTION_PORT       DIRECTION_PORT
> #define M3_DIRECTION_PIN        7
> #if N_AUTO_SQUARED
> #define M3_LIMIT_PORT           LIMIT_PORT
> #define M3_LIMIT_PIN            15
> #endif
> #endif
> 
> // Define driver spindle pins
> 
> #if DRIVER_SPINDLE_PWM_ENABLE
> #define SPINDLE_PWM_PORT_BASE   GPIOA_BASE
> #define SPINDLE_PWM_PIN         8
> #else
> #define AUXOUTPUT0_PORT         GPIOA
> #define AUXOUTPUT0_PIN          8
> #endif
> 
> #if DRIVER_SPINDLE_DIR_ENABLE
> #define SPINDLE_DIRECTION_PORT  GPIOB
> #define SPINDLE_DIRECTION_PIN   0
> #else
> #define AUXOUTPUT1_PORT         GPIOB
> #define AUXOUTPUT1_PIN          0
> #endif
> 
> #if DRIVER_SPINDLE_ENABLE
> #define SPINDLE_ENABLE_PORT     GPIOB
> #define SPINDLE_ENABLE_PIN      1
> #else
> #define AUXOUTPUT2_PORT         GPIOB
> #define AUXOUTPUT2_PIN          1
> #endif
> 
> // Define flood and mist coolant enable output pins.
> #define COOLANT_FLOOD_PORT      GPIOE
> #define COOLANT_FLOOD_PIN       7
> #define COOLANT_MIST_PORT       GPIOE
> #define COOLANT_MIST_PIN        8
> 
> // Define user-control controls (cycle start, reset, feed hold) input pins.
> #define CONTROL_PORT            GPIOE
> #define RESET_PIN               9
> #define FEED_HOLD_PIN           10
> #define CYCLE_START_PIN         11
> #define CONTROL_INMODE GPIO_SHIFT5
> 
> #define AUXINPUT0_PORT          GPIOE
> #define AUXINPUT0_PIN           12
> 
> #if SAFETY_DOOR_ENABLE
> #define SAFETY_DOOR_PORT        AUXINPUT0_PORT
> #define SAFETY_DOOR_PIN         AUXINPUT0_PIN
> #endif
> 
> #if MOTOR_FAULT_ENABLE
> #define MOTOR_FAULT_PORT        AUXINPUT0_PORT
> #define MOTOR_FAULT_PIN         AUXINPUT0_PIN
> #endif
> 
> // Define probe switch input pin.
> #define PROBE_PORT              GPIOA
> #define PROBE_PIN               7
> 
> #if I2C_STROBE_ENABLE
> #define I2C_STROBE_PORT         GPIOB
> #define I2C_STROBE_PIN          7
> #endif
> 
> 
> #if ETHERNET_ENABLE
> #undef SPI_ENABLE
> #define SPI_ENABLE              1
> #define SPI_PORT                12
> #define SPI_CS_PORT             GPIOB
> #define SPI_CS_PIN              7
> #define SPI_IRQ_PORT            GPIOB
> #define SPI_IRQ_PIN             8
> #define SPI_RST_PORT            GPIOB
> #define SPI_RST_PIN             6
> #endif

terjeio, 2024-02-22:
I have wired up my F446 dev board and that works as it should. Perhaps PlatformIO is using its own framework version without the code modification? Have you checked that?

semos2k, 2024-02-22:
No, I have not verified that, how can I verify it?

terjeio, 2024-02-22:
By locating the file and checking if it has the added code linked to above. If not replace the file with the one from the repository - there are a few #include files added at the top of it too.
In the Web Builder linux server the framework code is located in `~/.platformio/packages/framework-stm32cubef4`.

Tip: one way to find out which file the compiler uses is to deliberately add an error in the file.

semos2k, 2024-02-22:
I have tried the following: I have removed these two directories, I have compiled and it still does not work.

[image]

I have commented USB_SERIAL_CDC support and it works correctly.

terjeio, 2024-02-22:
Then check the file in the download. Does it have the changes?
It could also be that the (virtual) DTR pin is left in an incorrect state by the OS leading the controller to believe there is a client connected. If so the USB TX buffer will fill up and block all output.

semos2k, 2024-02-22:
I do the test by disconnecting the USB connector, and connecting the board to a 5v source directly.

terjeio, 2024-02-22:
> I do the test by disconnecting the USB connector, and connecting the board to a 5v source directly.

Oops - forgot to test that, seems that I have to figure out how to check for the USB port beeing powered or not.

semos2k, 2024-02-22:
Yes it worked correctly, thank you

datdadev, 2024-07-19:
How can you port from VGT6 to VET6? Is it easy by simply modifying the VGT6 FLASH.ld file? Or I have to modify any other file to get it worked?