grblHAL/STM32F4xx #163: Unable to compile. ioports error

plantez, 2024-02-24:
Hi
I'm unable to compile any targets. I get this error

`Src\driver.c: In function 'driver_init':
Src\driver.c:2858:18: error: 'input_signal_t {aka struct <anonymous>}' has no member named 'user_port'
             input->user_port = aux_digital_in.n_pins++;
                  ^~
Src\driver.c:2859:60: error: 'input_signal_t {aka struct <anonymous>}' has no member named 'user_port'
             input->id = (pin_function_t)(Input_Aux0 + input->user_port);
...
`

I am using platformio and have tried a couple of the targets, all give the same error.
I see that there have been changes to ioports in recent build, but there are no branches to pull for previous builds like [20240202] that compiles in the web tool.

terjeio, 2024-02-25:
My bad, the updated _Inc/driver.h_ was not included in the last commit - fixed now.
Maintaining branches looks like a lot of work so I have not chosen to do so. What I could do is to tag new releases, I'll try to add that to the commit script I use.

plantez, 2024-02-25:
All good. Everything is working now, thanks :)
And tagged releases would be great.

closed 2024-02-25 by plantez