grblHAL/STM32F4xx #167: stm32f411 black pill

maximeh2, 2024-04-06:
[VER:1.1f.20240404:]
[OPT:VNMSL,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,ES,SED]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:95K]
[DRIVER:STM32F411]
[DRIVER VERSION:240404]
[BOARD:BlackPill]
[AUX IO:1,2,0,0]
[PLUGIN:Bootloader Entry v0.02]
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
$X
error:79 - Not allowed while critical event is active.

please help mi

terjeio, 2024-04-06:
[First run](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings).

maximeh2, 2024-04-06:
thanks for your response
no effect
for this moment i use only the blackpill without stepper, limit switch... juste for a test after i build a shield 
that's for a co2 or diode laser, only X and y   excuse me i speak french

maximeh2, 2024-04-06:
has anyone managed to get a stm32 F411 black pill CEU to work?

it's to give me the solution but just to save me wasting my time

thanks

layerswing, 2024-04-06:
Yes, it's working for me.
$i
[VER:1.1f.20231005:]
[OPT:VNMSL,35,1024,4,0]
[AXS:4:XYZA]
[NEWOPT:ENUMS,RT+,ES,SED]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[FREE MEMORY:102K]
[DRIVER:STM32F411]
[DRIVER VERSION:230922]
[BOARD:BlackPill]
ok
$G
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
ok

$pins
[PIN:PB6,Emergency stop]
[PIN:PB7,Feed hold]
[PIN:PB8,Cycle start]
[PIN:PB15,Probe]
[PIN:PB12,X limit min]
[PIN:PB13,Y limit min]
[PIN:PB14,Z limit min]
[PIN:PA0,X step]
[PIN:PA2,Y step]
[PIN:PA4,Z step]
[PIN:PA6,A step]
[PIN:PA1,X dir]
[PIN:PA3,Y dir]
[PIN:PA5,Z dir]
[PIN:PA7,A dir]
[PIN:PB0,Steppers enable]
[PIN:PB1,Spindle on]
[PIN:PB2,Spindle direction]
[PIN:PC15,Flood]
[PIN:PC14,Mist]
[PIN:PA8,Spindle PWM]

maximeh2, 2024-04-06:
lightburn? ok?

layerswing, 2024-04-06:
I'm using bcnc with Stepcraft 420D for milling and 3dPrinting

layerswing, 2024-04-06:
My settings for this machine
$$
$0=10.0
$1=20
$2=0
$3=0
$4=14
$5=0
$6=0
$9=1
$10=511
$11=0.010
$12=0.002
$13=0
$14=0
$15=0
$16=0
$17=0
$18=0
$19=0
$20=0
$21=0
$22=0
$23=0
$24=25.0
$25=250.0
$26=50
$27=2.000
$28=0.100
$29=0.0
$30=1000.000
$31=0.000
$32=0
$33=5000.0
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=0
$43=1
$44=4
$45=1
$46=2
$47=0
$62=0
$63=2
$64=0
$65=0
$100=133.333
$101=133.333
$102=133.333
$103=213.330
$110=2800.000
$111=2800.000
$112=2800.000
$113=3400.000
$120=50.000
$121=50.000
$122=50.000
$123=50.000
$130=290.000
$131=410.000
$132=130.000
$133=200.000
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=100.0
$346=1
$376=0
$384=0
$398=35
$481=0
$484=1
$486=0
ok

maximeh2, 2024-04-06:
programming from which interface to compile?

maximeh2, 2024-04-06:
USB speed   : Full Speed (12MBit/s)
Manuf. ID   : STMicroelectronics
Product ID  : STM32  BOOTLOADER
SN          : 379233563032
DFU protocol: 1.1
Board       : --
Device ID   : 0x0431
Device name : STM32F411xC/E
Flash size  : 512 KBytes (default)
Device type : MCU
Revision ID : --
Device CPU  : Cortex-M4

layerswing, 2024-04-06:
git clone https://github.com/grblHAL/core
git clone https://github.com/grblHAL/STM32F4xx.git
cd STM32F4xx
(copy core/ folder content to STM32F4xx/grbl
pio run --list-targets
pio run --environment blackpill_f411ce

I've uploaded to F411 with CubeProgrammer and ST-LINK V2

layerswing, 2024-04-06:
I've modified platformio.ini line 266 added -D N_AXIS=4 as I have 3d Printer option too

maximeh2, 2024-04-06:
In file included from Src\MCP3221.c:24:0:
Inc/driver.h:49:10: fatal error: grbl/driver_opts.h: No such file or directory
 #include "grbl/driver_opts.h"

terjeio, 2024-04-06:
Error 18 is from reset/estop pin (PB6) asserted - either invert as described in the [first run page](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings#input-controls) or short it to ground.

> Inc/driver.h:49:10: fatal error: grbl/driver_opts.h: No such file or directory

This is due to the core repo not being fetched correctly, use [--recurse-submodules when fetching](https://github.com/grblHAL/core/wiki/Compiling-grblHAL#using-git-command-line) or use the [Web Builder](http://svn.io-engineering.com:8080).

maximeh2, 2024-04-06:
web builder with blackpill f411cue not compatible error 79

grbl\spindle_control.c:54:22: warning: 'spindle_cap' defined but not used [-Wunused-variable]
 static spindle_cap_t spindle_cap;

after compilation gcode  error 79

terjeio, 2024-04-06:
> grbl\spindle_control.c:54:22: warning: 'spindle_cap' defined but not used [-Wunused-variable]

This does not matter.

> after compilation gcode error 79

Did you short the reset/estop pin to ground or invert the input?

maximeh2, 2024-04-06:
WOW WOW PERFECT
Many thanks