grblHAL/STM32F4xx #169: Can you implement the kinematics for the scara arms?

maximeh2, 2024-04-12:
obsolete, useless?
thanks in advance

terjeio, 2024-04-12:
I, or somebody else that has access to a machine (I do not), can.

> obsolete, useless?

If there are valid use cases then not.

maximeh2, 2024-04-14:
If I'm talking about obsolescence or uselessness, it's because there's no kinetix in this github.

I speak French and I didn't quite understand your message.

thank you

terjeio, 2024-04-16:
Since I do not have access to a scara arm machine it is not easy for me to implement the scara kinematcs. So perhaps somebody that has (you?) can do it?

terjeio, 2024-04-16:
I found [this thread](https://github.com/grblHAL/core/issues/348) while doing some housekeeping, perhaps the author has something to share?

maximeh2, 2024-04-19:
Thank you very much for your help, I can't find how to define the arm size.

maximeh2, 2024-04-19:
very thanks for your help
when I compile everything is fine except for a small error that I have commented on but which has no particular impact then I upload without a hitch but when I want to connect I get a usb error

terjeio, 2024-04-19:
Incorrect clock tree setup? The USB peripheral clock must be 48 MHz.

maximeh2, 2024-04-19:
I don't know

terjeio, 2024-04-19:
I don't know either - when I compile and flash with Eclipse my F411 Blackpill works as it should.
Does it work again if you disable the Scara kinematics? If so then there is a bug in the kinematics code?

maximeh2, 2024-04-19:
without  kinematic, no problem

terjeio, 2024-04-19:
Then the platformio.ini file might be incorrect? Both works for me:
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[FREE MEMORY:94K]
[DRIVER:STM32F411]
[DRIVER VERSION:240418]
[BOARD:BlackPill]
[AUX IO:1,2,0,0]
[PLUGIN:Bootloader Entry v0.02]
[KINEMATICS:Polar v0.02]
```

maximeh2, 2024-04-19:
[env:blackpill_f411ce]
board = blackpill_f411ce
board_build.ldscript = STM32F411CEUX_FLASH.ld
build_flags = ${common.build_flags}
  # See Inc/my_machine.h for options
  -D BOARD_BLACKPILL=
  -D USB_SERIAL_CDC=1 
  # Uncomment to enable Spindle PWM output on the SpinDir pin
  #-D PROTONEER_SPINDLE_PWM=
lib_deps = ${common.lib_deps}
lib_extra_dirs = ${common.lib_extra_dirs}
# Alternatively, place the .pio/build/<env name>/firmware.bin on the NODE_F4xxRE drive
; change MCU frequency
upload_protocol = dfu

maximeh2, 2024-04-20:
another discovery that fried me 2 stm32, I downloaded from web builder the bin for stm32black pil wall plotter, when I connect with my programs I configure the limits to invert them and then I want to test one of the engines and no reaction, is it possible that the bin generated by web builder are unusable?

maximeh2, 2024-04-22:
Finaly with webbuilder polar and wall bin ok but a smale question for polar, servo pin in stm32f411 ?

terjeio, 2024-04-23:
The real pity is that I do not make enough money from this project to hire people that could add useful stuff when I do not have enough time myself?

The spindle PWM output can be used to control a servo when the spindle is configured for it via settings. See [footnote 13](https://github.com/grblHAL/drivers) for details.

BTW this week I will be busy so do not have much time to spare for support.