grblHAL/STM32F4xx #173: Devtronic CNC Controller V2 Web Builder build failed

nickshl, 2024-04-26:
By some reason build failed when I tried to build it via Web Builder(Spindle Sync & Log for tuning options enabled)
```
Command failed: pio run -etest -cjob.ini
In file included from Src/driver.c:38:0:
./grbl/pin_bits_masks.h:560:2: error: #error "Spindle encoder requires at least SPINDLE_PULSE_PIN defined in the board map!"
#error "Spindle encoder requires at least SPINDLE_PULSE_PIN defined in the board map!"
^~~~~
./grbl/pin_bits_masks.h:578:2: error: #error "Spindle sync requires SPINDLE_PULSE_PIN and SPINDLE_INDEX_PIN defined in the board map!"
#error "Spindle sync requires SPINDLE_PULSE_PIN and SPINDLE_INDEX_PIN defined in the board map!"
^~~~~
Src/driver.c: In function 'driver_setup':
Src/driver.c:2420:19: error: 'SPINDLE_PULSE_PORT' undeclared (first use in this function); did you mean 'SPINDLE_PULSE_BIT'?
HAL_GPIO_Init(SPINDLE_PULSE_PORT, &GPIO_Init);
^~~~~~~~~~~~~~~~~~
SPINDLE_PULSE_BIT
Src/driver.c:2420:19: note: each undeclared identifier is reported only once for each function it appears in
Src/driver.c:2426:16: error: 'SPINDLE_PULSE_PIN' undeclared (first use in this function); did you mean 'SPINDLE_PULSE_BIT'?
.pin = SPINDLE_PULSE_PIN,
^~~~~~~~~~~~~~~~~Command failed: pio run -etest -cjob.ini
In file included from Src/driver.c:38:0:
./grbl/pin_bits_masks.h:560:2: error: #error "Spindle encoder requires at least SPINDLE_PULSE_PIN defined in the board map!"
#error "Spindle encoder requires at least SPINDLE_PULSE_PIN defined in the board map!"
^~~~~
./grbl/pin_bits_masks.h:578:2: error: #error "Spindle sync requires SPINDLE_PULSE_PIN and SPINDLE_INDEX_PIN defined in the board map!"
#error "Spindle sync requires SPINDLE_PULSE_PIN and SPINDLE_INDEX_PIN defined in the board map!"
^~~~~
Src/driver.c: In function 'driver_setup':
Src/driver.c:2420:19: error: 'SPINDLE_PULSE_PORT' undeclared (first use in this function); did you mean 'SPINDLE_PULSE_BIT'?
HAL_GPIO_Init(SPINDLE_PULSE_PORT, &GPIO_Init);
^~~~~~~~~~~~~~~~~~
SPINDLE_PULSE_BIT
Src/driver.c:2420:19: note: each undeclared identifier is reported only once for each function it appears in
Src/driver.c:2426:16: error: 'SPINDLE_PULSE_PIN' undeclared (first use in this function); did you mean 'SPINDLE_PULSE_BIT'?
.pin = SPINDLE_PULSE_PIN,
^~~~~~~~~~~~~~~~~
SPINDLE_PULSE_BIT
In file included from Src/driver_spindles.c:30:0:
./grbl/pin_bits_masks.h:560:2: error: #error "Spindle encoder requires at least SPINDLE_PULSE_PIN defined in the board map!"
#error "Spindle encoder requires at least SPINDLE_PULSE_PIN defined in the board map!"
^~~~~
./grbl/pin_bits_masks.h:578:2: error: #error "Spindle sync requires SPINDLE_PULSE_PIN and SPINDLE_INDEX_PIN defined in the board map!"
#error "Spindle sync requires SPINDLE_PULSE_PIN and SPINDLE_INDEX_PIN defined in the board map!"
^~~~~
*** [.pio/build/test/src/driver_spindles.o] Error 1
*** [.pio/build/test/src/driver.o] Error 1
========================= [FAILED] Took 23.73 seconds =========================
SPINDLE_PULSE_BIT
In file included from Src/driver_spindles.c:30:0:
./grbl/pin_bits_masks.h:560:2: error: #error "Spindle encoder requires at least SPINDLE_PULSE_PIN defined in the board map!"
#error "Spindle encoder requires at least SPINDLE_PULSE_PIN defined in the board map!"
^~~~~
./grbl/pin_bits_masks.h:578:2: error: #error "Spindle sync requires SPINDLE_PULSE_PIN and SPINDLE_INDEX_PIN defined in the board map!"
#error "Spindle sync requires SPINDLE_PULSE_PIN and SPINDLE_INDEX_PIN defined in the board map!"
^~~~~
*** [.pio/build/test/src/driver_spindles.o] Error 1
*** [.pio/build/test/src/driver.o] Error 1
========================= [FAILED] Took 23.73 seconds =========================
```
I able to build it on my local PC with SPINDLE_SYNC_ENABLE without any errors...

terjeio, 2024-04-26:
My bad, I did not update the Web Builder server after merging.

closed 2024-04-26 by terjeio

nickshl, 2024-04-26:
By the way, any plands to rename "MPG & DRO mode:"? Test 1 and Test 2 really unintuitive.

nickshl, 2024-04-27:
Couple more issues I found:

**1)** I have StepperOnline CL57T  V4.1 closed loop stepper drivers. Those drivers has Alarm output. In case driver can't move stepper motor(travel limit reached) it pulls Alarm output to ground. I wired those outputs to Reset(EStop) button. On my previous board it works just fine. But on new one every time when I turn my milling machine on I got Alarm 10. I could reset alarm and after that all works as it should - no more alarms, as soon axis reached end of travel, but I still will push it, it will throw Alarm 10.
It may happens because before I powered controller from different power supply, now I powered everything from one PSU. May be before controller started a little bit late and missed pulse on Reset button line. I wish there configurable via settings delay at startup before driver will start monitoring those lines(_if sate is IDLE of course_)...

**2)** Looks like when Alarm happens on startup(Issue 1), controller ignores commands asking to gain control by MPG. And since there no PC connected, there no status reported to MPG and MPG show status Unknown instead Alarm.

For now I just disconnected stepper drivers Alarm output from Reset button input.

nickshl, 2024-04-27:
@terjeio one last thing: did I configured RS-485 correctly? How can I test it? I don't have Modbus spindle but I do have USB - RS-485 converter that I can use with PuTTY to check communication.

terjeio, 2024-04-27:
> By the way, any plands to rename "MPG & DRO mode:"? Test 1 and Test 2 really unintuitive.

Yes that is my plan, but it has not reached top of my todo list...

1. Try first with changing [this line](https://github.com/grblHAL/core/blob/3a7a65ea7c3312987cd7b4e7b9552e63e4639fb9/system.c#L73) to `if(signals.value && !sys.cold_start) {`
If this does not help a small plugin that hooks into `hal.driver_setup` and delays a bit before calling the original function is the next step to try?

2. This might be fixed by ignoring system interrupts during startup (by changing the code as shown above).

> did I configured RS-485 correctly? How can I test it?

The most complete test would be to connect a supported VFD or a VFD simulator (I added one in my PSOC 5 based CNC machine simulator). Alternatively set up the port as the main communication channel and check if that is working.

nickshl, 2024-04-27:
> 2. This might be fixed by ignoring system interrupts during startup (by changing the code as shown above).

No, it will not help in all cases. For example fresh flashed controller with wiped settings. In future I plan to add read/write settings to the SmartPendant, and Alarm at startup(_because of inverted EStop input_) makes it impossible if controller won't allow to control from MPG UART in Alarm state. Same(_probably_) will happens if EStop button pressed - pendant become useless, I have to power up my laptop, connect it to USB and clear the Alarm.

terjeio, 2024-04-27:
I see now, the keypad plugin blocks switching when in estop state. Other MPG modes should not.

nickshl, 2024-04-27:
I use "Test 2" - I assume the keypad plugin shouldn't be active in this case and the realtime command should be handled by the grblHAL core?

terjeio, 2024-04-27:
If stream switching is by the `0x8B` character then it is accepted when in[ idle, alarm and estop states and not currently streaming a file](https://github.com/grblHAL/core/blob/3a7a65ea7c3312987cd7b4e7b9552e63e4639fb9/stream.c#L506). Are you sure you are sending the character? Or are you using the MPG handshake signal?

nickshl, 2024-04-27:
I am using a command. I will double check what is going on since I could turn on streaming all data(received and send) to USB CDC in SmartPendant. 

But if the Keypad plugin ignores the command in Alarm state... could you double check that Web Builder uses actual MPG & DRO value? It may be binary, in this case 2 becomes 1...

By the way, how can I enable backlash compensation during the build? I didn't find any define related to backlash in my_machine.h

terjeio, 2024-04-27:
> By the way, how can I enable backlash compensation during the build? 

It is a [core config](https://github.com/grblHAL/core/blob/3a7a65ea7c3312987cd7b4e7b9552e63e4639fb9/config.h#L501).

nickshl, 2024-04-28:
Firmware built with Web Builder, option MPG & DRO Test 2 and no PC connected:

1) When grblHAL started, it never report MPG:0 to MPG UART and respond to ? status request command. Not a big deal(MPG powered from PC USB all the time and it is part of the problem), but better to include it.

2) When Alarm 10 happens and MPG not in control: ALARM 10 string sent to MPG. Controller ignores any attempts to gain control via real time command - no any  respond.

3) When Alarm 10 happens and MPG is in control: ALARM 10 string sent to MPG. The controller ignores any attempts to release control via real time command. It doesn't respond to real time command with status with MPG:0 and it continues to respond to ? status request command.

terjeio, 2024-04-28:
1. This is how it is [programmed](https://github.com/grblHAL/core/blob/3a7a65ea7c3312987cd7b4e7b9552e63e4639fb9/grbllib.c#L366-L367). The reasoning behind this was not to cause sender issues.

2. Hmm, I only tested for alarm10 at startup. If it happens later then all real-time commands except reset is ignored since alarm 10 (and a few others) are ["blocking"](https://github.com/grblHAL/core/blob/3a7a65ea7c3312987cd7b4e7b9552e63e4639fb9/protocol.c#L477-L524). Only soft reset, real-time report requests and a subset of $-commands are responded to while in this state. So you have to send a soft reset (ctrl-X) to exit the "blocking" state before you can switch MPG mode. I am not sure I want to change this behaviour other than perhaps figuring out why switching to MPG mode is allowed when alarm 10 is raised on controller startup and fix that.

3. See above.

nickshl, 2024-04-28:
1. I probably should reset MPG control state if grblHAL startup string is received. Or request full status report. Don't know if it good idea - grblHAL will send this report both to MPG and sender.

2. Soft reset works fine. So, I added to MPG screen in SmartPendant that in Alarm or Unknown(_additional state when pendant have no Idea what happens with controller_) states Stop button changed to Reset. I should figure out better way for soft buttons on SmartPendant instead copy-paste... by the way, you can always grab latest hex for it [there](https://github.com/nickshl/SmartPendant/tree/main/Release).

3. ioSender locks itself completely if it not in control. As result, if alarm happens when MPG in control, I couldn't release control on MPG(_until now when I added Reset command_) and I couldn't even close the ioSender. I have to open Task Manager to kill it. I don't think it should completely lock up itself. Reset, speed/feed override and close button should work. Also I don't understand why ioSender show message at startup if MPG is in control instead show locked main window.

terjeio, 2024-04-29:
1. Some senders may object to getting a full status report, but if so it may be time to switch to one that does not?
2. Ok.
3.
> ioSender locks itself completely if it not in control.

I'll look into this.

> Also I don't understand why ioSender show message at startup if MPG is in control instead show locked main window.

The MPG in control dialog is shown because ioSender needs to read configuration data from the controller on startup - which it cannot do when a MPG is in control. I have it on my todo list to allow the sender to start with the main screen visible even if no controller is connected and have connect/disconnect menus like most others - it is a rather large change that I have to find time and motivate myself for.