grblHAL/STM32F4xx #175: Spindle PWM duty cycle max value

auxp, 2024-04-27:
In lathe mode Spindle PWM duty cycle max value only10%？
CPU STM32F411

terjeio, 2024-04-27:
No, not with default spindle settings.

auxp, 2024-04-28:
> 不，不适用于默认主轴设置。

So how should the spindle be configured

auxp, 2024-04-28:
> > 不，不适用于默认主轴设置。
> 
> 那么主轴应该如何配置

How to modify to CNC mode  PWM

terjeio, 2024-04-28:
$30-$36 are the basic PWM spindle settings:

`$help spindle` outputs this:
```
$16: Invert spindle signals as bitfield:
    0 - Spindle enable (1)
    1 - Spindle direction (2)
    2 - PWM (4)

Reboot required.

Inverts the spindle on, counterclockwise and PWM signals (active low).

NOTE: A hard reset of the controller is required after changing this setting.

$30: Maximum spindle speed in RPM

Maximum spindle speed, can be overridden by spindle plugins.

$31: Minimum spindle speed in RPM

When set > 0 $35 (PWM min value) may have to be set to get the configured RPM.

$33: Spindle PWM frequency in Hz

Spindle PWM frequency.

$34: Spindle PWM off value in percent, max: 100

Spindle PWM off value in percent (duty cycle).

$35: Spindle PWM min value in percent, max: 100

Spindle PWM min value in percent (duty cycle).

$36: Spindle PWM max value in percent, max: 100

Spindle PWM max value in percent (duty cycle).
```

auxp, 2024-04-28:
> $help spindle

$help spindle
---- Spindle:
    0 - Enable (1)
    1 - RPM controls spindle enable signal (2)
    2 - Disable laser mode capability (4)
Enable controls PWM output availability.
When `RPM controls spindle enable signal` is checked and M3 or M4 is active S0 switches it off and S > 0 switches it on.
    0 - Spindle enable (1)
    1 - Spindle direction (2)
    2 - PWM (4)
Reboot required.
Inverts the spindle on, counterclockwise and PWM signals (active low).
NOTE: A hard reset of the controller is required after changing this setting.
Maximum spindle speed, can be overridden by spindle plugins.
Minimum spindle speed, can be overridden by spindle plugins.
When set > 0 $35 (PWM min value) may have to be set to get the configured RPM.
Spindle PWM frequency.
Spindle PWM off value in percent (duty cycle).
Spindle PWM min value in percent (duty cycle).
Spindle PWM max value in percent (duty cycle).
Spindle encoder pulses per revolution.
NOTE: A hard reset of the controller is required after changing this setting.
Comma separated list of values: RPM_MIN, RPM_LINE_A1, RPM_LINE_B1, set to blank to disable.
Comma separated list of values: RPM_POINT12, RPM_LINE_A2, RPM_LINE_B2, set to blank to disable.
Comma separated list of values: RPM_POINT23, RPM_LINE_A3, RPM_LINE_B3, set to blank to disable.
Comma separated list of values: RPM_POINT34, RPM_LINE_A4, RPM_LINE_B4, set to blank to disable.
Spindle at speed tolerance as percentage deviation from programmed speed, set to 0 to disable.
If not within tolerance when checked after spindle on delay ($392) alarm 14 is raised.

auxp, 2024-04-28:
> $help主轴 ---- 主轴：9 美元：PWM 主轴作为位场，其中设置位 0 启用其余部分：0 - 使能 （1） 1 - RPM 控制主轴使能信号 （2） 2 - 禁用激光模式功能 （4） 使能控制 PWM 输出可用性。选中且 M3 或 M4 处于活动状态时，S0 将其关闭，S > 0 将其打开。 $16：将主轴信号反转为位场： 0 - 主轴使能 （1） 1 - 主轴方向 （2） 2 - PWM （4） 需要重新启动。反转主轴开启、逆时针和 PWM 信号（低电平有效）。注意： 更改此设置后需要硬重置控制器。 30 美元：最大主轴转速（RPM） 最大主轴转速，可由主轴插件覆盖。 31 美元：最小主轴转速（以 RPM 为单位） 最小主轴转速，可被主轴插件覆盖。当设置为 0 时>可能必须设置 35 美元（PWM 最小值）才能获得配置的 RPM。 33 美元：主轴 PWM 频率，单位为 Hz 主轴 PWM 频率。 34 美元：主轴 PWM 关断值（百分比），最大：100 主轴 PWM 关断值（占空比）。 35 美元：主轴 PWM 最小值（百分比），最大：100 主轴 PWM 最小值（占空比）。 36 美元：主轴 PWM 最大值（百分比），最大值：100 主轴 PWM 最大值（百分比）（占空比）。 38 美元：主轴每转脉冲数 （PPR），需要重新启动 主轴编码器每转脉冲数。注意： 更改此设置后需要硬重置控制器。 66 美元：主轴线性化，第 1 点，最大值：39 逗号分隔的值列表：RPM_MIN、RPM_LINE_A1、RPM_LINE_B1，设置为空白以禁用。 67 美元：主轴线性化，第 2 点，最大值：39 逗号分隔的值列表：RPM_POINT12、RPM_LINE_A2、RPM_LINE_B2，设置为空白以禁用。 68 美元：主轴线性化，第 3 点，最大值：39 逗号分隔的值列表：RPM_POINT23、RPM_LINE_A3、RPM_LINE_B3，设置为空白以禁用。 69 美元：主轴线性化，第 4 点，最大值：39 逗号分隔的值列表：RPM_POINT34、RPM_LINE_A4、RPM_LINE_B4，设置为空白以禁用。 340 美元：主轴速度公差百分比 主轴速度公差与编程速度的百分比偏差，设置为 0 以禁用。如果在主轴延迟（392 美元）后检查时不在公差范围内，则会发出警报 14。`RPM controls spindle enable signal`

These should all be correct, but when sending m3s1000, the PWM is only 1%
[image]

auxp, 2024-04-28:
Hi terjeio have you seen it? It has been bothering me for a long time, with a maximum spindle of 100%. However, the actual measured PWM pin PA8 only has a 1% positive proportion. I found that as long as I write to the lathe or the spindle to start synchronously, PWM will behave like this. The PWM of CNC mode is normal, and the PWM generated by Web is also the same. Where is the problem? Sorry, I only know simple programs. I hope you can give me some guidance@terjeio

auxp, 2024-04-28:
Hi terjeio have you seen it? It has been bothering me for a long time, with a maximum spindle of 100%. However, the actual measured PWM pin PA8 only has a 1% positive proportion. I found that as long as I write to the lathe or the spindle to start synchronously, PWM will behave like this. The PWM of CNC mode is normal, and the PWM generated by Web is also the same. Where is the problem? Sorry, I only know simple programs. I hope you can give me some guidance @terjeio

auxp, 2024-04-28:
Hi terjeio, have you seen it? It has been bothering me for a long time, with a maximum spindle of 100%. However, the actual measured PWM pin PA8 only has a 1% positive proportion. I found that as long as I write to the lathe or the spindle to start synchronously, PWM will behave like this. The PWM of CNC mode is normal, and the PWM generated by Web is also the same. Where is the problem? Sorry, I only know simple programs. I hope you can give me some guidance  @terjeio 

> 30-36 美元是基本的 PWM 主轴设置：
> 
> `$help spindle`输出以下内容：
> 
> ```
> $16: Invert spindle signals as bitfield:
>     0 - Spindle enable (1)
>     1 - Spindle direction (2)
>     2 - PWM (4)
> 
> Reboot required.
> 
> Inverts the spindle on, counterclockwise and PWM signals (active low).
> 
> NOTE: A hard reset of the controller is required after changing this setting.
> 
> $30: Maximum spindle speed in RPM
> 
> Maximum spindle speed, can be overridden by spindle plugins.
> 
> $31: Minimum spindle speed in RPM
> 
> 
> 
> When set > 0 $35 (PWM min value) may have to be set to get the configured RPM.
> 
> $33: Spindle PWM frequency in Hz
> 
> Spindle PWM frequency.
> 
> $34: Spindle PWM off value in percent, max: 100
> 
> Spindle PWM off value in percent (duty cycle).
> 
> $35: Spindle PWM min value in percent, max: 100
> 
> Spindle PWM min value in percent (duty cycle).
> 
> $36: Spindle PWM max value in percent, max: 100
> 
> Spindle PWM max value in percent (duty cycle).
> ```

Hi terjeio, have you seen it? It has been bothering me for a long time, with a maximum spindle of 100%. However, the actual measured PWM pin PA8 only has a 1% positive proportion. I found that as long as I write to the lathe or the spindle to start synchronously, PWM will behave like this. The PWM of CNC mode is normal, and the PWM generated by Web is also the same. Where is the problem? Sorry, I only know simple programs. I hope you can give me some guidance  @terjeio

terjeio, 2024-04-29:
Which board map?
And can you share a gcode fragment that shows the issue?

auxp, 2024-04-29:
> 哪个板图？你能分享一个显示问题的 gcode 片段吗？

 I use blackpill_map.h  Directly through Web Builder Other is the default value, and the content on the image has been selected When I turn on the maximum spindle speed。PWM positive proportion is only 1%
[image]

[image]

terjeio, 2024-04-29:
Does it work if you do not eanble spindle linearization in the Web Builder?
It looks like there might be some spurious data in settings $66-$68.

auxp, 2024-04-29:
> 如果您不在 Web 构建器中执行主轴线性化，它是否有效？看起来在 66 美元至 68 美元的设置中可能存在一些虚假数据。

Thank you very much. Turning off spindle linearization has solved the problem,It was my negligence that caused it. I never thought about removing the spindle linearization before Thank you.

terjeio, 2024-04-29:
Not your fault - it seems the Web Builder does not link to float capable libraries for print and scan functions. I am trying to fix that now.

auxp, 2024-04-29:
> 这不是你的错 - 似乎 Web Builder 没有链接到用于打印和扫描功能的浮动库。我现在正试图解决这个问题。

You are too modest. It was you who reminded me. Thank you