grblHAL/STM32F4xx #176: Homing action is no problem. But Y_hard limit swich is not working

yoshi515rei, 2024-04-29:
Homing action is no problem. X and Z_hard limit swich works well.  only the hard Y_limit is not work. iosenter "signals" shows Y Limit input. but CNC continues work. 
Could you let me know the reason for Y_hard limit swich is no working?

Hard wear
 STM32 Nucleo F411RE with uno shield 
 
uno_map.h
  Added 
  #define VARIABLE_SPINDLE 1
 Swaped io pins
 #define AUXOUTPUT1_PORT         GPIOB // D13 Spindle direction  A5(D13)--> B7
 #define AUXOUTPUT1_PIN          7
 #if DRIVER_SPINDLE_PWM_ENABLE
 #define AUXOUTPUT2_PORT         GPIOA // Spindle enable, on morpho header B7--> A5(D13)
 #define AUXOUTPUT2_PIN          5
 No changes without the above.
Stm32 cube ide

terjeio, 2024-04-29:
It is due to IRQ number conflict with Z- likely from a regression.
Since you are compiling yourself you may add this snipped in front of [this](https://github.com/grblHAL/STM32F4xx/blob/555ebd03991a5213a71fc09d324201b9ea68b03c/Src/driver.c#L1122) and [this](https://github.com/grblHAL/STM32F4xx/blob/555ebd03991a5213a71fc09d324201b9ea68b03c/Src/driver.c#L1881) line for a quick fix.
```
#ifdef Z_LIMIT_POLL
        if(limit->id != Input_LimitZ)
#endif
```
Change` limit->id` to` input->id` in the second instance.

yoshi515rei, 2024-04-30:
Hi terjeio

Thank you giving good advice. Last night I have uploaded firmware which has revised src/driver.c to STM32 Nucleo. Y Hard Limit switch works normally. And then I could not find out any issues.
By the way, I think that BSMCE04U sorce code has has IRQ number conflict issues. E_stop, feed hold and cycle start swtches are nonfunctional, althogh iosender "signals" shows these switches input.

STM32 grbl 4 axis firmware using the 5-17-2023 grblHAL works well. But Web and Stm32CubeIde compiled firmwares  have above issues.
Coul you show me the solutions for above issues?

terjeio, 2024-04-30:
My BSMCE04U works as it should, what is your `$pins` response?

yoshi515rei, 2024-05-01:
I have built the firmware for BSMCE04U with using the latest Web Builder. BSMCE04U with the latest firmware works well. 
Anyway, thanks solving my issues for the firmware.

closed 2024-05-01 by yoshi515rei