grblHAL/STM32F4xx #178: 4 axis with btt octopus 1.1 issue

Flowbey, 2024-05-19:
The Firmware works fine for btt octopus 1.1 with 3 axis and x y z ganged. But as soon as i try 4 axis, something is wrong.
no motor is turning and the command M122 Shows only:

>>> m122
[TRINAMIC]

and the board hangs up.

after compiling the setting in grbl $338  shows $338 = 15.  X Y Z A is on. Thats correct with the doc from plugin:
$338 - Driver enable
Parameter is a axismask where value is sum of X=1, Y=2, Z=4 etc.

when i change the setting $338=7 X Y Z is on everything works fine except i dont have A Axis.

M122 shows:
>>> m122
[TRINAMIC]
                      X       Y       Z
Driver          TMC2209 TMC2209 TMC2209
Set current         500     500     500
RMS current         489     489     489
Peak current        691     691     691
Run current       15/31   15/31   15/31
Hold current       7/31    7/31    7/31
CS actual          7/31    7/31    7/31

and more....... 

so now when i try change back the setting $338=15 it says:
>>> $338=15
An error was detected while sending '$338=15': (error:52)

do i miss something to change before compiling for 4 axis?
could it be that something is missing in eeprom for 4 axis?

Thanks a lot for your effort

terjeio, 2024-05-20:
> when i try change back the setting $338=15 it says:
$338=15
An error was detected while sending '$338=15': (error:52)

You cannot set $338 to 15 unless the fourth axis is added so that is as expected.

Since the board hangs on a M122 command when 4 axes are configured something is going wrong in the code. Sadly I am not able to check that out until I am back home early next month since I do not have a board with a F4 processor available for debugging where I am now.

Flowbey, 2024-05-20:
Thank you
 I will try to find whats going on, but my skills in coding are limited.

terjeio, 2024-05-21:
It could be bad mapping for the M3 UART pin or incorrect jumper settings.
Which source are you using for the build?

Flowbey, 2024-05-21:
> It could be bad mapping for the M3 UART pin or incorrect jumper settings. Which source are you using for the build?

i used a clone of https://github.com/ErdbeermarmeladebrotmitHonig/STM32F4xx and build it with platformio.
the pin map should work because it works with ganged motors.

terjeio, 2024-05-21:
> the pin map should work because it works with ganged motors.

Can you check with the $pins command that the UART port for the fourth (M3) motor is the correct one when enabling 4 axes (and no ganging)?

Flowbey, 2024-05-21:
the command $$ shows:

...
$44 = 4   
$45 = 3   
$46 = 0   
$62 = 0   
$63 = 3   
$64 = 0   
$100 = 250.00000    (X-axis travel resolution, step/mm)
$101 = 250.00000    (Y-axis travel resolution, step/mm)
$102 = 250.00000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
$140 = 500   
$141 = 500   
$142 = 500   
...

should't there settings for A axis ?

terjeio, 2024-05-22:
> should't there settings for A axis ?

Yes, enabled [here](https://github.com/grblHAL/core/blob/1de8d7b7f026d1ed9ce41667aec20cce65c26378/config.h#L42) by changing the value to 4. Or by adding the symbol to platformio.ini (`-D N_AXIS=4`).

Flowbey, 2024-05-22:
> > should't there settings for A axis ?
> 
> Yes, enabled [here](https://github.com/grblHAL/core/blob/1de8d7b7f026d1ed9ce41667aec20cce65c26378/config.h#L42) by changing the value to 4. Or by adding the symbol to platformio.ini (`-D N_AXIS=4`).

I changed the value in config.h

I found something …. But i have to go working now. I can test more in 12 hours

Flowbey, 2024-05-22:
> > should't there settings for A axis ?
> 
> Yes, enabled [here](https://github.com/grblHAL/core/blob/1de8d7b7f026d1ed9ce41667aec20cce65c26378/config.h#L42) by changing the value to 4. Or by adding the symbol to platformio.ini (`-D N_AXIS=4`).

now it works. the thing is what you said with the build_flag in platformio.ini
  -D N_AXIS=4

i mean i had already the define 4 Axes in config.h:
#ifndef N_AXIS
#define N_AXIS 4 // Number of axes
#endif

But it looks like that does not compile correctly.

terjeio, 2024-05-22:
If N_AXIS is defined in the pio file then that takes precedence, maybe that was the reason?

Flowbey, 2024-05-23:
yes, that looks like. 

Thank you for your input with the pio file

closed 2024-05-23 by Flowbey