grblHAL/STM32F4xx #181: Error compiling SKR 2 envs

goeland86, 2024-06-10:
I'm a newbie with grblHAL, so apologies if I missed something in the documentation I should've seen.

I set up a new environment for the SKR 2 with TMC2209s as follows:
```
[env:btt_skr_2_tmc2209]
# Untested and might not boot.  Please report issues at:
# https://github.com/grblHAL/STM32F4xx/issues
board = genericSTM32F407VGT6
board_build.ldscript = STM32F407VGTX_FLASH.ld
build_flags = ${common.build_flags}
  # See Inc/my_machine.h for options
  -D BOARD_BTT_SKR_20=
  -D HSE_VALUE=8000000
  -D USB_SERIAL_CDC=1
  # Boot loader offset (32K)
  -D VECT_TAB_OFFSET=0x8000
  # TMC2209 stepper drivers
  -D HAS_BOOTLOADER
  -D TRINAMIC_ENABLE=2209
lib_deps = ${common.lib_deps}
  eeprom
  trinamic
lib_extra_dirs = ${common.lib_extra_dirs}
# Upload is not supported for this board since BOOT0 is tied to GND.
# With the default boot loader, you must deploy new firmware by copying
# .pio/build/<env name>/firmware.bin (produced by `pio run`) to the SD card.
```

When I build (using PlatformIO in VSCode), I get the following error during compilation (full log truncated for clarity).
```
[...]
Compiling .pio/build/btt_skr_2_tmc2209/FrameworkCMSISDevice/system_stm32f4xx.o
Archiving .pio/build/btt_skr_2_tmc2209/libFrameworkCMSISDevice.a
Indexing .pio/build/btt_skr_2_tmc2209/libFrameworkCMSISDevice.a
Linking .pio/build/btt_skr_2_tmc2209/firmware.elf
.pio/build/btt_skr_2_tmc2209/src/driver.o: In function `driver_init':
driver.c:(.text.driver_init+0x2a0): undefined reference to `_FLASH_VectorTable'
driver.c:(.text.driver_init+0x532): undefined reference to `board_init'
collect2: error: ld returned 1 exit status
*** [.pio/build/btt_skr_2_tmc2209/firmware.elf] Error 1
================================================================================================================================================================== [FAILED] Took 2.46 seconds ==================================================================================================================================================================

Environment        Status    Duration
-----------------  --------  ------------
btt_skr_2_tmc2209  FAILED    00:00:02.458
============================================================================================================================================================= 1 failed, 0 succeeded in 00:00:02.458 =============================================================================================================================================================
```

Out of curiosity, I tried the TMC2130 version, just to see if that builds. It seems not.

```
[...]
Compiling .pio/build/btt_skr_2_tmc2130/FrameworkCMSISDevice/system_stm32f4xx.o
Archiving .pio/build/btt_skr_2_tmc2130/libFrameworkCMSISDevice.a
Indexing .pio/build/btt_skr_2_tmc2130/libFrameworkCMSISDevice.a
Linking .pio/build/btt_skr_2_tmc2130/firmware.elf
.pio/build/btt_skr_2_tmc2130/src/driver.o: In function `driver_init':
driver.c:(.text.driver_init+0x51a): undefined reference to `board_init'
collect2: error: ld returned 1 exit status
*** [.pio/build/btt_skr_2_tmc2130/firmware.elf] Error 1
================================================================================================================================================================== [FAILED] Took 2.52 seconds ==================================================================================================================================================================

Environment        Status    Duration
-----------------  --------  ------------
btt_skr_2_tmc2130  FAILED    00:00:02.518
============================================================================================================================================================= 1 failed, 0 succeeded in 00:00:02.518 =============================================================================================================================================================
 ```

I would appreciate any input on how to resolve this. The WebBuilder also disables any EEPROM option for the SKR 2 board, which has proven frustrating when trying to set up my router.

Thanks for any pointers to resolve the issue.

goeland86, 2024-06-17:
Your provided platformio.ini test file works - and I was using the default from the project for SKR2.

May I kindly suggest that the SKR2 definitions in the provided `platformio.ini` file from the project re-use the provided values listed above?

I'm happy to create a PR to do so if it would be helpful.

paukstelis, 2024-10-15:
I'm glad a found this issue. I was working on a custom board and basing it around the SKR2 and was stumped why none of those would link properly.