grblHAL/STM32F4xx #185: Can you guide me how to change the USART default serial port to different USART pins?

terjeio, 2024-07-19:
Hi @terjeio, can you guide me how to change the USART default serial port to different USART pins?
I am using STM32F407VET6, in GRBLHAL, there is STM32F407VGT6 using BTT SKR 2.0 board with default USART1, I have changed to USART2 in the code, but it did not work when I connect that 2 pins between the laptop and the MCU
I use STM32CubeIDE, to build the selected project which issues STM32F4 8MHz

This is what I have modified in the code:
```c
#define SERIAL_PORT     2   // GPIOA: TX = 2, RX = 3
```

_Originally posted by @datdadev in https://github.com/grblHAL/STM32F4xx/issues/77#issuecomment-2238571477_

terjeio, 2024-07-19:
@datadev Since USB is the default comm channel it has to be commented out in [my_machine.h.](https://github.com/grblHAL/STM32F4xx/blob/f52ea93f96c87c4db52c809962bd7b46b8a50677/Inc/my_machine.h#L64) Have you done that?

datdadev, 2024-07-19:
Yes I have done that

terjeio, 2024-07-19:
Components for thermistor conditioning on the inputs are interfering? Or are you testing with your own board?

FYI my Nucleo-64 board works with the port, from the `$pins` output:
```
[PIN:PA3,RX,Primary UART]
[PIN:PA2,TX,Primary UART]
```

datdadev, 2024-07-19:
Sure, I am using a custom-built board with an STM32F407VET6 microcontroller. My goal is to connect the desired USART port (RS232) between my laptop and the MCU to drive the stepper motors without any additional external sensors. I believe I may have made some errors in the configuration process.

Could you please guide me on how to properly configure the STM32F407VET6 to use the desired serial port? Specifically, I would like to know which files and settings I need to customize in STM32CubeIDE, given that I am using my own custom board.

datdadev, 2024-07-19:
`driver.json`:
```json
   {
      "name": "MY BOARD MAP",
      "symbol": "MY_BOARD",
      "MAP": "boards/my_board_map.h",
      "caps": {
        "uart": 1,
        "axes": 3,
        "serial_ports": 1
      },
      "symbols": {
        "STM32F407xx": "",
        "HSE_VALUE": 8000000
      }
    },
```

`platformio.ini`:
```
[env:my_board]
# Untested and might not boot.  Please report issues at:
# https://github.com/grblHAL/STM32F4xx/issues
board = genericSTM32F407VGT6
board_build.ldscript = STM32F407VGTX_FLASH.ld
build_flags = ${common.build_flags}
  # See Inc/my_machine.h for options
  -D MY_BOARD=
  -D HSE_VALUE=8000000
  -D USB_SERIAL_CDC=1
  -D VECT_TAB_OFFSET=0x8000
lib_deps = ${common.lib_deps}
  eeprom
lib_extra_dirs = ${common.lib_extra_dirs}
```

terjeio, 2024-07-19:
> I use STM32CubeIDE

Then driver.json (only for the Web Builder) and  platformio.ini is not used by the build process.
Modify _my_machine.ini_ instead. Uncomment [this line](https://github.com/grblHAL/STM32F4xx/blob/f52ea93f96c87c4db52c809962bd7b46b8a50677/Inc/my_machine.h#L45) to make the compiler use your board definition.

> Idk if I have to define all the other pins to adapt to my current board (sorry for not mentioning this before) to make whole things work?

No - but some may be required, the compiler will complain if they are. BTW when you edit a comment I am not notified, when doing so add a new comment telling me a bout the edit or just add a new one instead.

datdadev, 2024-07-19:
This is `my_machine_map.h` file:
```c
// file_name: "my_board_map.h"

#define BOARD_NAME "MY BOARD"
#define HAS_BOARD_INIT

#define SERIAL_PORT     2   // GPIOA: TX = 2, RX = 3

// Define step pulse output pins.
#define X_STEP_PORT                 GPIOD
#define X_STEP_PIN                  13                  // X
#define Y_STEP_PORT                 GPIOD
#define Y_STEP_PIN                  11                  // Y
#define Z_STEP_PORT                 GPIOE
#define Z_STEP_PIN                  2                   // Z
#define STEP_OUTMODE                GPIO_BITBAND
//#define STEP_PINMODE                PINMODE_OD // Uncomment for open drain outputs

// Define step direction output pins.
#define X_DIRECTION_PORT            GPIOD
#define X_DIRECTION_PIN             12
#define Y_DIRECTION_PORT            GPIOB
#define Y_DIRECTION_PIN             15
#define Z_DIRECTION_PORT            GPIOE
#define Z_DIRECTION_PIN             3
#define DIRECTION_OUTMODE           GPIO_BITBAND
//#define DIRECTION_PINMODE           PINMODE_OD // Uncomment for open drain outputs

// Define homing/hard limit switch input pins.
#define X_LIMIT_PORT                GPIOA
#define X_LIMIT_PIN                 11                           // X- Limit
#define Y_LIMIT_PORT                GPIOA
#define Y_LIMIT_PIN                 12                           // Y- Limit
#define Z_LIMIT_PORT                GPIOB
#define Z_LIMIT_PIN                 12                           // Z- Limit
#define LIMIT_INMODE                GPIO_BITBAND

// EOF

```

datdadev, 2024-07-19:
When I build it with "17 Release F407 8MHz" option, it gives many warnings and errors that come from those features that my board does not use/support. How can I disable all of them during the build though?

datdadev, 2024-07-19:
<details>
<summary>Error logs from STM32CubeIDE</summary>

```bash
  183 | #warning "Selected spindle is not fully supported - no direction output!"
      |  ^~~~~~~
../Inc/driver.h:189:2: warning: #warning "Selected spindle is not supported!" [-Wcpp]
  189 | #warning "Selected spindle is not supported!"
      |  ^~~~~~~
arm-none-eabi-gcc "../keypad/display/i2c_leds.c" -mcpu=cortex-m4 -std=gnu11 -DUSE_HAL_DRIVER -DSTM32F407xx -DHSE_VALUE=8000000 -D_VOLUMES=1 -D_USE_IOCTL=1 -D_USE_WRITE=1 -c -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/FatFs" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../USB_DEVICE/App -I../USB_DEVICE/Target -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/lwip/src/include" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/networking/wiznet" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/FatFs/STM" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/Drivers/FATFS/Target" -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"keypad/display/i2c_leds.d" -MT"keypad/display/i2c_leds.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "keypad/display/i2c_leds.o"
arm-none-eabi-gcc "../keypad/keypad.c" -mcpu=cortex-m4 -std=gnu11 -DUSE_HAL_DRIVER -DSTM32F407xx -DHSE_VALUE=8000000 -D_VOLUMES=1 -D_USE_IOCTL=1 -D_USE_WRITE=1 -c -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/FatFs" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../USB_DEVICE/App -I../USB_DEVICE/Target -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/lwip/src/include" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/networking/wiznet" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/FatFs/STM" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/Drivers/FATFS/Target" -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"keypad/keypad.d" -MT"keypad/keypad.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "keypad/keypad.o"
arm-none-eabi-gcc "../keypad/macros.c" -mcpu=cortex-m4 -std=gnu11 -DUSE_HAL_DRIVER -DSTM32F407xx -DHSE_VALUE=8000000 -D_VOLUMES=1 -D_USE_IOCTL=1 -D_USE_WRITE=1 -c -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/FatFs" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../USB_DEVICE/App -I../USB_DEVICE/Target -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/lwip/src/include" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/networking/wiznet" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/FatFs/STM" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/Drivers/FATFS/Target" -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"keypad/macros.d" -MT"keypad/macros.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "keypad/macros.o"
In file included from ../laser/coolant.c:23:
../Inc/driver.h:177:2: warning: #warning "Selected spindle is not supported!" [-Wcpp]
  177 | #warning "Selected spindle is not supported!"
      |  ^~~~~~~
../Inc/driver.h:183:2: warning: #warning "Selected spindle is not fully supported - no direction output!" [-Wcpp]
  183 | #warning "Selected spindle is not fully supported - no direction output!"
      |  ^~~~~~~
../Inc/driver.h:189:2: warning: #warning "Selected spindle is not supported!" [-Wcpp]
  189 | #warning "Selected spindle is not supported!"
      |  ^~~~~~~
arm-none-eabi-gcc "../fans/fans.c" -mcpu=cortex-m4 -std=gnu11 -DUSE_HAL_DRIVER -DSTM32F407xx -DHSE_VALUE=8000000 -D_VOLUMES=1 -D_USE_IOCTL=1 -D_USE_WRITE=1 -c -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/FatFs" -I../Inc -I../Drivers/CMSIS/Include -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../USB_DEVICE/App -I../USB_DEVICE/Target -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/lwip/src/include" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/networking/wiznet" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/FatFs/STM" -I"C:/Users/ltd/Documents/STM32/GRBL Driver STM32F401/Drivers/FATFS/Target" -Os -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"fans/fans.d" -MT"fans/fans.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "fans/fans.o"
In file included from ../laser/lb_clusters.c:26:
../Inc/driver.h:177:2: warning: #warning "Selected spindle is not supported!" [-Wcpp]
[... cut ...]
  177 | #warning "Selected spindle is not supported!"
      |  ^~~~~~~
../Inc/driver.h:183:2: warning: #warning "Selected spindle is not fully supported - no direction output!" [-Wcpp]
  183 | #warning "Selected spindle is not fully supported - no direction output!"
      |  ^~~~~~~
../Inc/driver.h:189:2: warning: #warning "Selected spindle is not supported!" [-Wcpp]
  189 | #warning "Selected spindle is not supported!"
      |  ^~~~~~~
"make -j8 all" terminated with exit code 2. Build might be incomplete.
```
</details>

terjeio, 2024-07-19:
For the spindle error unccoment [this line](https://github.com/grblHAL/STM32F4xx/blob/f52ea93f96c87c4db52c809962bd7b46b8a50677/Inc/my_machine.h#L73) and change it to `#define SPINDLE0_ENABLE SPINDLE_NONE`

For the other errors either add the missing pin definitions or remove code in [driver.c](https://github.com/grblHAL/STM32F4xx/blob/f52ea93f96c87c4db52c809962bd7b46b8a50677/Src/driver.c) until compilation succeds.

datdadev, 2024-07-19:
Alright, instead of removing the code, I have commented out all out the pin definitions, and now I just left with this error that I don't know how to fix:
```bash
18:53:57 **** Incremental Build of configuration Release F407 8MHz for project GRBL Driver STM32F401 ****
make -j8 all 
arm-none-eabi-gcc -o "grblHAL F407 8MHz.elf" @"objects.list"   -mcpu=cortex-m4 -T"C:\Users\ltd\Documents\STM32\GRBL Driver STM32F401\STM32F407VGTX_FLASH.ld" --specs=nosys.specs -Wl,-Map="grblHAL F407 8MHz.map" -Wl,--gc-sections -static --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -u _printf_float -u _scanf_float -Wl,--start-group -lc -lm -Wl,--end-group
C:/ST/STM32CubeIDE_1.15.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.100.202403111256/tools/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: warning: grblHAL F407 8MHz.elf has a LOAD segment with RWX permissions
C:/ST/STM32CubeIDE_1.15.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.100.202403111256/tools/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: ./Src/driver.o: in function `driver_init':
driver.c:(.text.driver_init+0x3a4): undefined reference to `board_init'
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:110: grblHAL F407 8MHz.elf] Error 1
"make -j8 all" terminated with exit code 2. Build might be incomplete.

18:53:59 Build Failed. 2 errors, 1 warnings. (took 1s.710ms)
```

terjeio, 2024-07-19:
Uncomment this line in your map file

`#define HAS_BOARD_INIT`

datdadev, 2024-07-19:
> Uncomment this line in your map file

You meant comment the line, I did and built successfully.
However, the board still can't connect with the laptop via USART:

[image]

datdadev, 2024-07-19:
Yes, I have checked them all:

[image]

[image]

When I scope the TX pin, it stays HIGH all the time (from the beginning after I reset the board).

I am unsure if the issue arises from using a different chip, specifically the STM32F407VET6 instead of the STM32F407VGT6. After building and compiling the code, it does not work as expected. Could this discrepancy be the source of the problem?

datdadev, 2024-07-19:
For the debugging, I toggled the breakpoints here as you said:

[image]

[image]

And it did not meet any breakpoints and ended up in this line:

[image]

terjeio, 2024-07-19:
Ok, then something goes badly wrong during startup.
Set breakpoints in _driver.c_ to figure out where/what. Start with driver_init(), is it entered and does it complete? If so then check driver_setup()...

datdadev, 2024-07-19:
Alright, found this in `grbllib.c`, when I step into this function, it brings me to the `HardFault_Handler` function:

[image]

terjeio, 2024-07-19:
Then either is hal.coolant.set_state is null or coolantSetState() (which hal.coolant.set_state points to)  in driver.c fails.

datdadev, 2024-07-19:
I believe I just need to comment out that section since I am not using the coolant feature. However, I encountered another issue. The execution gets stuck at the following line and cannot proceed past this statement:

[image]

Additionally, I would like to ask for more guidance. If I continue debugging like this, I am unsure where to start and stop. Specifically, where does the code loop in the GRBL command send and receive streaming mode?

terjeio, 2024-07-19:
> Specifically, where does the code loop in the GRBL command send and receive streaming mode?

Inside protocol_main_loop(), this function only returns on a soft reset (CTRL-X received or the grblHAL reset input asserted).

datdadev, 2024-07-19:
After redefining the coolant pins and uncommenting the related code, I was able to successfully run and connect the STM32F407VGT6 (test board). However, I am unable to achieve the same result with the STM32F407VET6 (current project-running board).

The primary difference between these two chips is the flash memory size: the "G" variant has 1024 KBytes, while the "E" variant has 512 KBytes. How can I easily port the project from the "G" variant to the "E" variant?

datdadev, 2024-07-19:
In the STM32F407VETX_FLASH.ld file auto-generated by the IDE, the memory configuration includes:
- 512KBytes FLASH
- 64KBytes CCMRAM
- 128KBytes RAM

I modified the memory section of the VGT6 ld file as follows:
```c
MEMORY
{
  CCMRAM    (xrw)    : ORIGIN = 0x10000000,   LENGTH = 64K
  RAM       (xrw)    : ORIGIN = 0x20000000,   LENGTH = 128K
  BOOT_FLASH(rx)     : ORIGIN = 0x08000000,   LENGTH = 16K
  EEPROM_EMUL(xrw)   : ORIGIN = 0x08004000,   LENGTH = 16K
  FLASH     (rx)     : ORIGIN = 0x08008000,   LENGTH = 512K - 16K - 16K
}
```

Additionally, I added the following section for CCMRAM:
```c
_siccmram = LOADADDR(.ccmram);

/* CCM-RAM section
 *
 * IMPORTANT NOTE!
 * If initialized variables are placed in this section,
 * the startup code needs to be modified to copy the init-values.
 */
.ccmram :
{
  . = ALIGN(4);
  _sccmram = .;       /* create a global symbol at ccmram start */
  *(.ccmram)
  *(.ccmram*)

  . = ALIGN(4);
  _eccmram = .;       /* create a global symbol at ccmram end */
} >CCMRAM AT> FLASH
```

After building and uploading it, seemed like not working. Am I do it correctly?

datdadev, 2024-07-20:
Although I can connect to the board and it works pretty well, however, I don't know if I should re-map the memory?

terjeio, 2024-07-21:
> I don't know if I should re-map the memory?

If it is working now then you do not need to. This since nothing is stored at the higher end of the flash.

datdadev, 2024-07-21:
I think this is a good startup for the setup process.
Thank you for helping me @terjeio. Guess that we can close this issue now.

closed 2024-07-24 by terjeio