grblHAL/STM32F4xx #187: CPU STM32F412RET6 Compilation error

auxp, 2024-08-02:
open#define _WIZCHIP_         5500  #define WEBUI_ENABLE         3 .board.  longboard32_map.h
error
make: *** No rule to make target 'D:\Projects\STM\GRBL Driver STM32F401\STM32F412VGTX_FLASH.ld', needed by 'firmware.elf'. Stop.
make: *** Waiting for unfinished jobs....

terjeio, 2024-08-02:
Both debug and release builds compiles for me. Are you using the STM32CubeIDE?

auxp, 2024-08-02:
> 调试和发布版本都为我编译。您在使用STM32CubeIDE吗？

yes.STM32CubeIDE 1.12.0

terjeio, 2024-08-02:
Other boards compiles?

My my_machine.h:
[my_machine.zip](https://github.com/user-attachments/files/16466157/my_machine.zip)

auxp, 2024-08-02:
same
make: *** No rule to make target 'D:\Projects\STM\GRBL Driver STM32F401\STM32F412VGTX_FLASH.ld', needed by 'firmware.elf'.  Stop.
make: *** Waiting for unfinished jobs....

auxp, 2024-08-02:
My my_machine.h:
[my_machine.h.txt](https://github.com/user-attachments/files/16466334/my_machine.h.txt)

auxp, 2024-08-02:
My longboard32_map.h:default
[longboard32_map.h.txt](https://github.com/user-attachments/files/16466356/longboard32_map.h.txt)

terjeio, 2024-08-02:
Is STM32F412VGTX_FLASH.ld missing?

auxp, 2024-08-02:
[image]

auxp, 2024-08-02:
[STM32F412VGTX_FLASH.ld.txt](https://github.com/user-attachments/files/16466444/STM32F412VGTX_FLASH.ld.txt)

terjeio, 2024-08-02:
[This](https://github.com/grblHAL/STM32F4xx/blob/master/STM32F412VGTX_FLASH.ld) is the correct file content.

auxp, 2024-08-02:
Missing one mistake
make: *** No rule to make target 'D:\Projects\STM\GRBL Driver STM32F401\STM32F412VGTX_FLASH.ld', needed by 'firmware.elf'.  Stop.

auxp, 2024-08-02:
i 'm sorry! Just now STM32F412VGTX_FLASH.ld Copy wrong
The current mistake:
arm-none-eabi-gcc -o "firmware.elf" @"objects.list"   -mcpu=cortex-m4 -T"D:\Projects\STM\GRBL Driver STM32F401\STM32F412VGTX_FLASH.ld" --specs=nosys.specs -Wl,-Map="firmware.map" -Wl,--gc-sections -static --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -u _printf_float -u _scanf_float -Wl,--start-group -lc -lm -Wl,--end-group
d:\stm32cubeide_1.12.0\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.200.202301161003\tools\arm-none-eabi\bin\ld.exe: ./boards/longboard32.o: in function `board_init':
D:/CNCGRBL/STM32Fxx/STM32F4xx-master/Debug F412/../boards/longboard32.c:31: undefined reference to `tmc_spi_init'
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:102: firmware.elf] Error 1

auxp, 2024-08-02:
thank you. I have already solved the problem

closed 2024-08-02 by auxp