grblHAL/STM32F4xx #188: STM32F412 WebUI 404error

auxp, 2024-08-02:
DRIVER VERSION:240404]
[BOARD:SuperLongBoard]
[AUX IO:11,7,1,0]
[PLUGIN:Bootloader Entry v0.02]
[WIZCHIP:W5500]
[MAC:56:34:12:dc:08:00]
[IP:192.168.1.10]
[PLUGIN:WebUI v0.21]
[PLUGIN:SDCARD v1.13]
[PLUGIN:FS macro plugin v0.07]

Browser input 192.168.1.10
404: The requested file cannot be found.

terjeio, 2024-08-02:
SD card is mounted successfully? And it has a _www_ folder with `index.html.gz` present? See the [WebUI readme](https://github.com/grblHAL/Plugin_WebUI/blob/main/README.md) - the STM32 drivers does not support flashfs/littlefs.

auxp, 2024-08-03:
Yes, I have resolved it,Without connecting to a computer, entering wbeui solely through a browser and unable to move? Coordinate display?

terjeio, 2024-08-03:
Have you enabled real-time report requests, either in the WebUI itself or by setting `$481` (Auto report interval) to a non-zero value?

auxp, 2024-08-03:
good afternoon. Thank you for your answer

auxp, 2024-08-03:
thank Your answer

siganberg, 2024-10-14:
Any tips compiling STM32F412 for SuperLongBoard?

terjeio, 2024-10-15:
> Any tips compiling STM32F412 for SuperLongBoard?

For now install the STM32CubeIDE and import this repo or use VSCode/PlatformIO with the [Sienci fork](https://github.com/Sienci-Labs/sienciHAL). Web Builder support will hopefully be available soon.