grblHAL/STM32F4xx #189: Error: board does not answered properly 400

auxp, 2024-08-06:
What is the reason for this?
[image]

terjeio, 2024-08-06:
How did you manage to display this page?

auxp, 2024-08-24:
Sorry for taking so long to reply to you due to network connectivity issues. The problem has been resolved. There is an issue with the webui configuration file,

auxp, 2024-08-24:
A new issue has been discovered. Clicking on the soft reset button in the webui immediately causes the CPU to stop running

Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
Core  0 register dump:
PC      : 0x4000c52f  PS      : 0x00060230  A0      : 0x800f1a0b  A1      : 0x3ffcc150  
A2      : 0x3f410299  A3      : 0x3f41036d  A4      : 0x0000002e  A5      : 0x00000003  
A6      : 0x3f41036d  A7      : 0x00000002  A8      : 0x00000000  A9      : 0x3f4102a8  
A10     : 0x0000002e  A11     : 0x3f4102a8  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000003  
EXCVADDR: 0x3f4102a8  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
Backtrace:0x4000c52c:0x3ffcc150 0x400f1a08:0x3ffcc170 0x400ee4fa:0x3ffcc1c0 0x401152da:0x3ffcc210 0x40119bd2:0x3ffcc280 0x4011d9fa:0x3ffcc2b0 0x4011077d:0x3ffcc2d0
ELF file SHA256: 93db4b22818da159
CPU halted.

terjeio, 2024-08-24:
> A new issue has been discovered. 

With a STM32F4xx driver? Looks like an ESP32 stack trace to me. Which board?

auxp, 2024-08-25:
Yes, it's ESP32  board BOARD_BLACKBOX_X32.Sorry, I should have asked under ESP32

terjeio, 2024-08-25:
Which version of grblHAL an the WebUI are you running? Click on the _grblHAL_ logo to find out.
On my test board soft reset works fine:

[image]

auxp, 2024-08-25:
[image]

auxp, 2024-08-25:
WebUl version: 3.0.0-a70.GH2 The same issue is also present

auxp, 2024-08-25:
I also downloaded the driver again from GitHub

terjeio, 2024-08-25:
$I output please. Do you build using the Web Builder?

auxp, 2024-08-25:
$I
[VER:1.1f.20240817:]
[OPT:VNSL,100,1024,4,0]
[AXS:4:XYZA]
[NEWOPT:ENUMS,RT+,REBOOT,SED,RTC,WIFI,FTP,SD,YM]
[FIRMWARE:grblHAL]
[SIGNALS:RHSP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:154K]
[DRIVER:ESP32]
[DRIVER VERSION:240817]
[DRIVER OPTIONS:-128-NOTFOUND]
[BOARD:HXP-CNC-4-AXIS-NO-LIMIT]
[WIFI MAC:cc:db:a7:36:78:ad]
[IP:192.168.1.8]
[PLUGIN:WebUI v0.21]
[PLUGIN:SDCARD v1.13]
[PLUGIN:FS macro plugin v0.08]
[MSG:WEBSOCKET STREAM ACTIVE]

use ESP-IDF 4.3

auxp, 2024-08-25:
$I
[VER:1.1f.20240817:]
[OPT:VNSL,100,1024,4,0]
[AXS:4:XYZA]
[NEWOPT:ENUMS,RT+,REBOOT,SED,RTC,WIFI,FTP,SD,YM]
[FIRMWARE:grblHAL]
[SIGNALS:RHSP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:154K]
[DRIVER:ESP32]
[DRIVER VERSION:240817]
[DRIVER OPTIONS:-128-NOTFOUND]
[BOARD:BlackBoxX32]
[WIFI MAC:cc:db:a7:36:78:ad]
[IP:192.168.1.8]
[PLUGIN:WebUI v0.21]
[PLUGIN:SDCARD v1.13]
[PLUGIN:FS macro plugin v0.08]

use ESO-IDF 4.3

terjeio, 2024-08-25:
Enabling the A-axis causes the crash, will have to check if this is true for other drivers as well. A quick workaround is to delay the soft reset, you may try this version of _webui/server.c_ if you want:

[server.zip](https://github.com/user-attachments/files/16739337/server.zip)

auxp, 2024-08-27:
Thank you, your file has resolved the issue

closed 2024-08-27 by auxp