grblHAL/STM32F4xx #190: SKR 2 not connecting to usb

dangerousderek, 2024-08-27:
Hi!
I used the Web builder to compile for the BTT SKR2. I'm not good enough with code to compile grblhal myself, and no combination of settings on the web builder seems to work. The SKR 2 board works fine with marlin, but as soon as I flash the .bin from the web builder it fails to show up as a usb com port. After a bunch of googling, I found a blog that pointed me to this conversation:
https://community.platformio.org/t/difficulty-with-getting-usb-serial-usb-cdc-working/7501/9
They are discussing the usb not working with this chipset, but once again I don't know enough about coding to compile grblhal and fix the usb issue myself. Is there any way to fix this issue with the web builder?
Thanks!

terjeio, 2024-08-27:
It could be the USB clock beeing off. Which processor is on the board, a 407 or a 429?

dangerousderek, 2024-08-27:
Thanks for the quick reply! It's the 407!

terjeio, 2024-09-04:
It could be due to the basic version (no bootloader) using an incorrect loader file. If this was the one you tried you may try again now. If not I can compile a bootloader version with Eclipse for you to try.

FYI I do not have a board to verify the code with.

cincir-binance, 2024-12-17:
Hi everyone. 
I have the same problem. I have BTT SKR2 with 407. I don't know how to distinguish between basic and bootloader version. But from webbuilder I tried both in combination with native and virtual usb. Any progress on the topic?

terjeio, 2024-12-26:
If you program by using the SD card you need the bootloader version. If via a programmer it depends on the bootloader beeing present in the chip or not.
If you select (Virtual) USB mode then you need to have a USB <> UART converter wired to the UART port.

terjeio, 2024-12-27:
Try this binary built with the IDE:

[firmware.zip](https://github.com/user-attachments/files/18258504/firmware.zip)

Working?

cincir01, 2024-12-27:
Works like a charm. After spending days on variations, the usb connection appears. If I could do it 5 days ago I would not order esp32 controller(it is on the way).Is the only problem PlatformIO itself? How I can compile it myself to tune the settings?

and my last question is, I managed to connect with ugs. but the problem is I am getting all the time error when sending commands. this is the error code 

_"An error was detected while sending '$X': (error:79) An unknown error has occurred. Streaming has been paused."_

Do you know why? Or is it a topic of another thread?

Thank you so much after all my failed affords.

cincir01, 2024-12-27:
I found the error 79 is [Status_NotAllowedCriticalEvent](http://svn.io-engineering.com/grblHAL/html/errors_8h.html#a8c1dfc1ccf00a08192611433ee7f17b4a2fa997e5b46cc5fc9578db7395b581b0)

in map file of SKR2 both reset pin and probe pin is defenied as GPIOE4. That is why I need to be able to compile it myself I think.

cincir01, 2024-12-29:
Installed the IDE , made the configurations compiled and bang. Everything is working now. Thanks for the solution @terjeio

closed 2025-02-17 by terjeio