grblHAL/STM32F4xx #194: STM32F407 feed rate halved after timer updates?

MountaineerGit, 2024-10-15:
Hi,
first of all, thanks for your great work on grbHal - fabulous job!

I have just updated to recent master (d8ffbe3450d4bbe9dfcb18648c71efff999f4744) and saw that my machine was running slower afterwards.
Previously I  was working on the commit from Sep 21, 2024 (a2b04ffbb8c42bc50b6e952fe3a92a094cf22be0), here everything is still fine. I have checked the PWM step-pulse generation on the scope.
For a travel-resolution of 250 step/mm I got a step-pulse distance 230us on the old commit for a 1000 mm/min feed rate, which is fine. Now the step-pulse distance is ~460us -> the step pulse halved.

I will try to figure out which commit caused the changes, but I see multiple commits around the timer-updated. Yet unclear, if this is the root-cause.
I have a STM32F07 discovery board with a custom BOB.
Any ideas ?

terjeio, 2024-10-15:
> Any ideas ?

It was due to a fix that unmasked a bug. Should be ok now.

closed 2024-10-24 by terjeio

MountaineerGit, 2024-10-24:
Verified ok!
Thanks for the quick fix 👍