grblHAL/STM32F4xx #196: The NucleoF446 cannot connect to ioSender.

CHTUZKI, 2024-10-23:
Hi.

I encountered a strange problem. I am using the STM32 Nucleo F446 prototype board and compiled using the `[env:nucleo_f446re_protoneer]` environment, and everything went well. However, when I opened iOSender and connected it to my Nucleo, I found that iOSender could not connect. I used a serial port tool to connect to my Nucleo and restarted it, hoping that the serial port tool would display some information to prove that GRBLHAL is working properly, but no information was displayed. Every time I restart my Nucleo, the serial port tool only shows a 0.

I wrote a simple hello world program using PlatformIO and ran it on my Nucleo F446. It works fine, and the serial tool prints hello world.

I re-uploaded the grblhal firmware and started debugging my Nucleo. I found that the MCU can receive normally in the `void UART0_IRQHandler (void)` function. I set a breakpoint at `if((UART0->SR & USART_SR_TXE) && (UART0->CR1 & USART_CR1_TXEIE))` and opened iOSender. I discovered that my Nucleo returns some information, such as coordinates, but it seems that this content is not being transmitted out.

I don't know why this is happening, can someone help me? Thank you very much.

--Best regards.

terjeio, 2024-10-24:
The platformio.ini file is outdated/incorrect? I do not keep it up to date since I do not use platformio to build. Anyway this is the generated .ini file used by the Web Builder, perhaps that works?
[job.zip](https://github.com/user-attachments/files/17501290/job.zip)
 Or even better try a [Web Builder](http://svn.io-engineering.com:8080) build if you have not modified the map file?

CHTUZKI, 2024-10-24:
I feel that there should be an issue with the serial port configuration of the Nucleo F446. After uploading the grblHAL firmware to the Nucleo F446 and rebooting it, the board should print out the version information of grblHAL and some help messages, but it is not doing so. During debugging, I found that the program is writing data to UART0->DR, but this data is not being sent out through the serial port of the Nucleo F446.

I am unfamiliar with the peripherals of STM32, but I will try to troubleshoot the issue.

CHTUZKI, 2024-10-24:
I used the Web Builder and downloaded the firmware for nucleoF446re, but the serial port only prints \0. When I restart my nucleoF446, the serial port will print a \0 and will not print the grblHAL version information or startup help.

ioSender still cannot connect.

dresco, 2024-10-24:
Just had a look at the morpho map on a board here.

It's a conflict between AUXOUTPUT0 (used for spindle control) and the serial port. Both are attempting to use PA2..

`env:nucleo_f446re_morpho_cnc` works, but you will need to add the `boards` folder to the common build flags in your platformio.ini.

CHTUZKI, 2024-10-24:
Thank you for your reply, the problem is now resolved.

closed 2024-10-24 by CHTUZKI