grblHAL/STM32F4xx #199: sound when move motors using tmc2209 after update

semos2k, 2024-11-14:
I have updated the version of GRBLHal that I was using since January 15 and when updating to the current version my motors have started to make a noise, I am using the same amperage configuration for the motors, the same step and speed configuration.

Should I do something additional to remove this noise?

terjeio, 2024-11-16:
When rechecking the code I found one parameter that was changed when I refactored some parts of the code earlier. I have restored that in the latest commit.

I have also an option for adding a number of low-level settings that can be used to tune the motors, now available in the Web Builder. If compiling yourself you can enable this by adding `#define TRINAMIC_EXTENDED_SETTINGS 1` in _my_machine.h_.
Note that adding (or removing) this option will reset all plugin setting values to default.

[image]

semos2k, 2024-11-17:
Thx, works perfectly.

closed 2024-11-17 by semos2k