grblHAL/STM32F4xx #200: Automatic square, limit switch always active

semos2k, 2024-11-17:
Hi, I am trying to activate the auto square feature, my board is a skr pro v1.2 and it has a limit switch available for each motor, when checking with ioSender it always shows the limit switch active and when activating them by hand they do not change state.

I have seen some videos on youtube and a blog but I have not found a solution. Should I only activate one limit switch for this feature or can I continue using the two that the card has available?

semos2k, 2024-11-19:
The switch works when use only one.

semos2k, 2024-11-19:
[image]
this is diagram for the wiring.

terjeio, 2024-11-21:
Is the corresponding LED changing state when you activate a switch?

semos2k, 2024-11-21:
Hi Terjeio, you were absolutely right. I'm using a TMC driver and one of them was active with the sensorless home pins. Thank you very much for your time.

closed 2024-11-21 by semos2k