grblHAL/STM32F4xx #202: Probe pin issues

paukstelis, 2024-11-27:
I've worked out most of getting a FLY-CDY-V3 (https://github.com/Mellow-3D/Fly-CDYv3) board working with GrblHal. However, I'm having a real problem getting the probe to work. I've tried a variety of different pins and configurations, with no real luck getting it to work. It seems as though the probe pins are never pulled up. If I look through the limit pins, a couple of them are pulled up (GPIOB 10 and 11), however, when I try to use those I can't compile because I get:
 Src/driver.c:75:2: error: #error Interrupt enabled input pins must have unique pin numbers!
 #error Interrupt enabled input pins must have unique pin numbers!`
This is despite the fact those pins aren't used elsewhere.
Any insights on this?

terjeio, 2024-11-28:
The interrupt circutry in STM32 processors only allow assigning a single interrupt per pin number regardless of which port it belongs to. E.g. GPIOA 10 and GPIOB 10 cannot both be interrupt enabled. Ref. the chapter about the Nested Vectored Interrupt Controller (NVIC) in the PRM.

>  It seems as though the probe pins are never pulled up. 

I'll have to check this. However, you should never rely on the weak internal pullup only - at least add an external pullup that will ensure current flow above the wetting current of the probe switch.

closed 2024-11-29 by paukstelis

terjeio, 2024-11-30:
> (still not clear why none of the limit switch pins can be pulled up)

Have you checked the schematic?
Later I will remove the pullup settings if a board has circuitry in front of the pins that render them meaningless.

paukstelis, 2024-11-30:
> > (still not clear why none of the limit switch pins can be pulled up)
> 
> Have you checked the schematic? Later I will remove the pullup settings if a board has circuitry in front of the pins that render them meaningless.

Great point. Having a look at the schematic, they have the motor driver DIAG pins and limit pins connected together. I suspect the driver is pulling them low.

misan, 2025-03-01:
@paukstelis Did you get the FLY-CDY-V3 board working with GrblHal? Is that effort supporting WiFi connectivity?

paukstelis, 2025-03-01:
> [@paukstelis](https://github.com/paukstelis) Did you get the FLY-CDY-V3 board working with GrblHal? Is that effort supporting WiFi connectivity?

I have the board working with GrblHal, but I have not yet tried to figure out WiFi. I haven't found an easy way to flash the ESP32 chip (though I honestly haven't looked into it very hard). 

I am also hoping to eventually do a plugin to read an incremental encoder using the hardware timer of the STM32. Then it will feel like I am back in the old dcservo days :).

misan, 2025-03-01:
@paukstelis I reckon (but I have not yet tried it myself) an easy way could be for the STM32 to act as a bridge between its USB and ESP32 serial ports. This way, esptool.py could be used to program the ESP32. Alternatively, a firmware upload process might run on the STM32 that would program the ESP32 from the serial port, mimicking the serial programming that esptool.py uses. 

> I am also hoping to eventually do a plugin to read an incremental encoder using the hardware timer of the STM32. Then it will feel like I am back in the old dcservo days :).

That sounds like fun ;-)

Cheers,

Miguel

dresco, 2025-03-01:
> an easy way could be for the STM32 to act as a bridge between its USB and ESP32 serial ports. This way, esptool.py could be used to program the ESP32

Yes, have a look at the experimental [ESP-AT](https://github.com/grblHAL/Plugins_misc/tree/main?tab=readme-ov-file#esp-at) plugin. There is a pasthru command `$PTRGH` to do exactly that...

Although, @terjeio I've just re-tested on the Scylla board, and the passthru command isn't being added. Seems [ioports_enumerate()](https://github.com/grblHAL/core/blob/39977aa3c44aa0f1451e88e2e7ae8c53241015c4/stream_passthru.c#L184-L187) is always returning false?

terjeio, 2025-03-02:
> Seems [ioports_enumerate()](https://github.com/grblHAL/core/blob/39977aa3c44aa0f1451e88e2e7ae8c53241015c4/stream_passthru.c#L184-L187) is always returning false?

Change [this line](https://github.com/dresco/STM32H7xx/blob/f4725cb2221831035930028f2b2b95cbd5001537/Src/driver.c#L36) to `#define AUX_CONTROLS (AUX_CONTROL_SPINDLE|AUX_CONTROL_COOLANT|COPROC_PASSTHRU)`

I'll have to update the F4 and F7 drivers too.

paukstelis, 2025-03-16:
Sorry for the silly question, but perhaps I am missing something here. I am trying to enable ESP_AT plugin, but `-D ESP_AT_ENABLE=1` in  platformio.ini throws an error: `.pio/build/fly_cdy_v3_tmc2209/src/driver.o: In function `driver_init': driver.c:(.text.driver_init+0x51e): undefined reference to `esp_at_init'`

dresco, 2025-03-16:
> I am trying to enable ESP_AT plugin, but `-D ESP_AT_ENABLE=1` in platformio.ini throws an error

Looks like the `plugins` folder is missing from the `lib_deps` section in the platformio.ini file. I think if you add that it should compile..

paukstelis, 2025-03-17:
> > I am trying to enable ESP_AT plugin, but `-D ESP_AT_ENABLE=1` in platformio.ini throws an error
> 
> Looks like the `plugins` folder is missing from the `lib_deps` section in the platformio.ini file. I think if you add that it should compile..

Thanks. That did it.