grblHAL/STM32F4xx #204: `Flood` Signal Fluctuation with Spindle Running

siganberg, 2024-12-27:
#### Description  
The auxiliary port `Flood` signal fluctuates when the spindle is running. It works as expected when the spindle is off. The expected behavior is for the `Flood` signal to remain in the state it was set to, regardless of whether the spindle is running or not.

#### Steps to Reproduce  
1. Set the `Flood` signal to the desired state.  
2. Start the spindle.  
3. Observe the `Flood` signal behavior.  

https://youtu.be/Zw3sDdpLiZE

#### Expected Behavior  
The `Flood` signal should remain in the set state regardless of whether the spindle is running or not.  

#### Actual Behavior  
The `Flood` signal fluctuates when the spindle is running.  

#### Additional Context  
- Firmware/Software version: Latest
- Hardware details: Sienci SLB

terjeio, 2024-12-28:
Not able to replicate.
What are your `$I` and `$pins` outputs.

siganberg, 2024-12-28:
$I

```
[VER:1.1f.20241222:]
[OPT:VNMPZHSL+2,128,1024,4,0]
[NEWOPT:ENUMS,RT+,HOME,ES,EXPR,SED,ETH,FTP,TMC=7,SD,YM]
[FIRMWARE:grblHAL]
[SIGNALS:DEP]
[NVS STORAGE:*EEPROM 16K]
[FREE MEMORY:129K]
[DRIVER:STM32F412]
[DRIVER VERSION:241221]
[BOARD:]
[AUX IO:7,4,1,0]
[PLUGIN:Bootloader Entry v0.02]
[PLUGIN:SLB Probing v0.02]
[WIZCHIP:W5500]
[MAC:02:08:dc:ca:62:d3]
[IP:192.168.5.1]
[PLUGIN:Trinamic v0.21]
[PLUGIN:MODBUS v0.16]
[PLUGIN:HUANYANG VFD v0.12]
[PLUGIN:HUANYANG P2A VFD v0.11]
[PLUGIN:Durapulse VFD GS20 vv0.06]
[PLUGIN:Yalang VFD YL620A v0.03]
[PLUGIN:MODVFD v0.04]
[PLUGIN:H-100 VFD v0.04]
[PLUGIN:Nowforever VFD v0.02]
[PLUGIN:Macro plugin v0.08]
[PLUGIN:RGB LED strips (M150) v0.05]
[PLUGIN:Spindle offset v0.01]
[PLUGIN:Events plugin v0.03]
[PLUGIN:SDCARD v1.16]
[PLUGIN:FS macro plugin v0.12]
[SPINDLE:Huanyang v1]
```

$pins

```
[PIN:PD7,Safety door]
[PIN:PB12,Emergency stop]
[PIN:PC4,Probe]
[PIN:PB5,X limit min]
[PIN:PA15,X home]
[PIN:PB9,Y limit min]
[PIN:PA1,Y home]
[PIN:PC13,Z limit min]
[PIN:PA2,Z home]
[PIN:PE14,A limit min]
[PIN:PD12,Aux in 0,P0]
[PIN:PD13,Aux in 1,P1]
[PIN:PD14,Aux in 2,P2]
[PIN:PC15,Aux in 3,Toolsetter]
[PIN:PE1,Aux in 4,Macro pin]
[PIN:PE0,Aux in 5,Macro pin]
[PIN:PC11,Aux in 6,P3]
[PIN:PD15,Aux in 7,P4]
[PIN:PB8,Aux in 8,P5]
[PIN:PB3,Aux in 10,P6]
[PIN:PA0,Aux analog in 0,E0]
[PIN:PE2,X step]
[PIN:PE3,Y step]
[PIN:PE5,Y2 step]
[PIN:PE4,Z step]
[PIN:PB10,A step]
[PIN:PE7,X dir]
[PIN:PE8,Y dir]
[PIN:PE10,Y2 dir]
[PIN:PE9,Z dir]
[PIN:PE11,A dir]
[PIN:PD11,Motor CSX]
[PIN:PE15,Motor CSY]
[PIN:PD9,Motor CSZ]
[PIN:PE12,Motor CSM4]
[PIN:PD4,X enable]
[PIN:PC7,Y enable]
[PIN:PC7,Y enable]
[PIN:PD3,Z enable]
[PIN:PC10,A enable]
[PIN:PE13,Spindle on]
[PIN:PB1,Spindle direction]
[PIN:PA8,Spindle PWM]
[PIN:PC12,Spindle 2 on]
[PIN:PC6,Spindle 2 PWM]
[PIN:PB4,Mist]
[PIN:PB14,Flood]
[PIN:PC14,Aux out 0,P0 <- Spindle enable]
[PIN:PC0,Aux out 1,P1 <- Flood enable]
[PIN:PC1,Aux out 2,P2 <- Spindle enable]
[PIN:PC8,Aux out 3,P3 <- Flood enable]
[PIN:PB0,Aux out 4,Modbus RX/TX direction]
[PIN:PC9,LED adressable]
[PIN:PA13,LED adressable 1]
[PIN:PA6,MISO]
[PIN:PC2,MISO]
[PIN:PA7,MOSI]
[PIN:PC3,MOSI]
[PIN:PA5,SPI CLK]
[PIN:PB13,SPI CLK]
[PIN:PA4,SPI CS]
[PIN:PB15,SD card CS]
[PIN:PA9,SPI reset]
[PIN:PA10,SPI IRQ]
[PIN:PB6,I2C SCK]
[PIN:PB7,I2C SDA]
[PIN:PC5,RX,UART1]
[PIN:PD8,TX,UART1]
[PIN:PD6,RX,Modbus]
[PIN:PD5,TX,Modbus]
```

siganberg, 2024-12-28:
let me try pulling the latest again and retest.

siganberg, 2024-12-28:
Same. Might be my voltage divider on my micro controller. Gonna try my new PCB once it arrive. 

But new problem is my CNC keep stalling on job.  Any idea?

https://www.youtube.com/shorts/6pe417o9iAE

Here's my firmware settings dump.
[gSender-firmware-settings-12_28_2024-1_49_38 PM.json](https://github.com/user-attachments/files/18267817/gSender-firmware-settings-12_28_2024-1_49_38.PM.json)

siganberg, 2024-12-29:
Still happening,  but might be the  circuit on  PCB for my micro controller, I think my voltage divider is lowering the voltage too much and it's very sensitive to voltage sag for the SLB 5v output. 

I'll update once I get my new version of the PCB circuit.

terjeio, 2024-12-29:
Are the status LEDs inside the controller (FLD, AUX) blinking in sync with the fluctuations?
If not it is likely your microcontroller is at fault?

siganberg, 2024-12-29:
It's not blinking although the changes on my micro-controller is too fast for the LED to blink if that is the case. 

My theory is I put wrong resistors on the voltage divider cause I need to shift the 5v of the SLB to  3.3v for my micro-controller input and it might it's too low that a tiny sag from the 5v when modbus signal is on. Don't have the right equipment to measure it so I'll just wait for my new PCB that has the right voltage divider.

siganberg, 2024-12-29:
Well, not my lucky day. I'm not sure if this issue is somehow related to the recent changes, but my X-stepper driver burned out today while my machine was idle. I heard the stepper disengage and saw some smoke coming from the SLB. Upon inspection, I confirmed that the stepper never gets hot during idle or regular operation.

I have extra SLB but worry if this is cause by firmware and might happen to the 2nd one too.  
[image]

siganberg, 2025-01-09:
After hooking up a cheap oscilloscope, I found the issue. When spindle is ON, it's creating noise that spike the voltage input up to 3v which is enough to trigger HIGH on my micro-controller.  I just added a debounce delay and switch to analogRead just consider high threshold value.

closed 2025-01-09 by siganberg