grblHAL/STM32F4xx #205: Alarm 14 triggering when job stop.

siganberg, 2024-12-31:
### Issue Description

When a job is intentionally stopped, **Alarm 14 (Spindle Timeout)** is triggered. This behavior should not occur as the spindle timeout is unnecessary in such cases.

### Expected Behavior

The firmware should wait for a specific amount of time to check if the spindle has completely stopped before triggering a timeout alarm, especially for spindles that do not have braking mechanisms. 

### Suggested Solution

Introduce a firmware setting to define a delay period after stopping a job, allowing the spindle to naturally come to a halt without triggering Alarm 14. This delay should be adjustable to accommodate different spindle setups.

### Steps to Reproduce
1. Run a job with a spindle that lacks braking.
2. Intentionally stop the job.
3. Observe that Alarm 14 is triggered immediately or prematurely.

Sender: gSender
Board: SLB
FIrmware: Latest

siganberg, 2024-12-31:
https://youtu.be/88jE804Yzaw

dresco, 2025-01-01:
Can't tell from your video, but are you using a modbus controlled spindle? If so, could be [this](https://github.com/grblHAL/Plugins_spindle/issues/32) issue..

terjeio, 2025-01-01:
Try with this code:

[modbus_rtu.zip](https://github.com/user-attachments/files/18285481/modbus_rtu.zip)

I am waiting for feedback from Sienci for the changes in it, mainly that retries has been implemented and reset handling changed. 
Retries will only be for blocking (synced) messages, and is disabled in the code. To enable/test with retries enabled set the number of retries in line 280, it is the last parameter in the function call.

siganberg, 2025-01-01:
> Can't tell from your video, but are you using a modbus controlled spindle? If so, could be [this](https://github.com/grblHAL/Plugins_spindle/issues/32) issue..

Yes I'm using Huanyang VFD.

siganberg, 2025-01-02:
Alarm 14 is fixed but M8 (Flood) keep turning ON after the stop. 

https://youtube.com/shorts/fyn2PcZenqM

terjeio, 2025-01-02:
> Alarm 14 is fixed but M8 (Flood) keep turning ON after the stop.

Finally found it. It is kind of annoying that senders often uses different terms than used in the controller, pause instead of feed hold, stop instead of reset. And I introduced stop as a new term in grblHAL to make matters worse.
Whatever, the issue only shows up when a feed hold (pause) is followed by a reset (stop)...
Will commit a fix soon, but unsure about adding the Modbus fix just yet.

dresco, 2025-01-02:
> unsure about adding the Modbus fix just yet.

Do you want more testing? I have a spare Huanyang on the bench here..

siganberg, 2025-01-02:
I also have a new H100 that will be testing soon for the coming Alt Mill.

siganberg, 2025-01-02:
> Try with this code:
>
> [modbus_rtu.zip](https://github.com/user-attachments/files/18285481/modbus_rtu.zip)
>
> I am waiting for feedback from Sienci for the changes in it, mainly that retries has been [...]

Wonder if this might be also related to the other issue I have that the FLOOD is flunctating value only when spindle is ON.

terjeio, 2025-01-02:
Fix for flood issue committed, I have also committed the updated Modbus driver along with modified VFD drivers now using retry functionality in the Modbus driver. Testing the retry functionality is kind of hard if the communication is solid? I have only tested with just unplugging the RS485 cable, using my CNC simulator.

@dresco FYI I have added retry data to [modbus_callbacks_t](https://github.com/grblHAL/core/blob/5a052010418d5f49ffa42114713ff7de56362ff4/modbus.h#L67-L72), you may want to set it in your MPG plugin?

> Do you want more testing? I have a spare Huanyang on the bench here..

Yes please.

> Wonder if this might be also related to the other issue I have that the FLOOD is flunctating its state only when spindle is ON.

Highly unlikely since a reset when in feed hold state is required to trigger the errant coolant output. For me the flood output is stable (not triggering the scope) whether the spindle is running or not.

dresco, 2025-01-02:
> FYI I have added retry data to [modbus_callbacks_t](https://github.com/grblHAL/core/blob/5a052010418d5f49ffa42114713ff7de56362ff4/modbus.h#L67-L72), you may want to set it in your MPG plugin?

Thanks, I think they are all non-blocking calls, but will double check

> Testing the retry functionality is kind of hard if the communication is solid?

I don't know, these cheap VFDs are pretty effective noise generators! ;) I think I should be able to get enough interference with some bad cabling.. Will do some testing over the next couple of days

siganberg, 2025-01-03:
Verified the alarm 14 is no longer triggering on stop and also the flood is no longer turning back on after the stop.

dresco, 2025-01-05:
> I don't know, these cheap VFDs are pretty effective noise generators! ;) I think I should be able to get enough interference with some bad cabling.. Will do some testing over the next couple of days

Yep, seems to be working well. I added a counter + live expression to track, and was able to generate intermittent retries with some deliberately dodgy RS485 wiring. No errors or alarms raised..

closed 2025-01-06 by terjeio