grblHAL/STM32F4xx #206: BTT SKR Pro v1.2 won't connect to Laserweb 4

UniqueLemming, 2025-01-04:
Hi, I'm trying to connect this board to Laserweb 4, but it keeps saying "no supported firmware detected" and disconnects. Compatibility mode does not fix the issue. IO Sender connects and works fine. I know that Laserweb is pretty old and it must be the fault of the program, but I'm hoping there is a way to get it going. I put a current version of GRBLHAL on an ESP32 and used that driver. Laserweb connects to it just fine, so I'm wondering if there is something different with this STM32F4xx driver.  I had an old version of GRBLHAL installed from around June, 2022 for this SKR board that connected just fine as well. Any idea how I could get this going?

terjeio, 2025-01-04:
The difference is that some boards are not reset, and thus not outputting the welcome message, when connection is established.
And Laserweb relies upon the welcome message arriving within a few seconds of connecting...
Recently I added a linestate changed event to the core that can be used to detect a USB connection, depending the client setting the DTR line on connection. Laserweb does so.
So I have to add code to trap DTR set event for drivers that provide it and output the welcome message - will do so in the next commit.

UniqueLemming, 2025-01-04:
Amazing, Thank you! I attempted to update my Smoothieboard on the LPX176x driver last night as well, and the new build also has this same connection issue. If I understand you correctly, it sounds like you are providing a fix in the core and so I assume that will fix it for both boards. Thank you so much for being willing to provide a fix for this old software!

terjeio, 2025-01-05:
Change commited. Be aware that some of the drivers do not implement the linestate changed event, RP2040 and the ones that uses the Arduino framework does not.

UniqueLemming, 2025-01-05:
Thank you very much! Both my SKR Pro 1.2 board and  my Smoothieboard can connect to Laserweb perfectly now. Very much appreciated!

closed 2025-01-06 by UniqueLemming