grblHAL/STM32F4xx #207: Button 3 Action Not Responding

siganberg, 2025-01-07:
#### **Description**
Button 3 is not working as expected. It does not perform any action when mapped, regardless of the function assigned. However, Button 1 and Button 2 are functioning correctly and perform their assigned actions without issues.

#### **Steps to Reproduce**
1. Attempt to map Button 3 to any action.
2. Test the button functionality.
3. Observe that Button 3 does not perform the assigned action.

#### **Expected Behavior**
Button 3 should execute the assigned action when pressed.

#### **Actual Behavior**
Button 3 does not perform any action, even when mapped correctly.

#### **Environment**
- **Device/Hardware**: Sienci SLB
- **Software/Firmware Version**:  Latest

terjeio, 2025-01-09:
The core and plugins has to be updated for this as well.

siganberg, 2025-01-09:
I haven't see any plugins update? Last changes still date Dec 20.

siganberg, 2025-01-09:
Also I notice that dwelling G4 is not working properly with the new core. 

Before this is working:

GP 0.100 fo 100 milliseconds delay. 

The new core, the delay is 1 second mininum.

terjeio, 2025-01-10:
> I didn't see any plugins update. Last changes still date Dec 20.

Odd, it is [this commit](https://github.com/grblHAL/Plugin_keypad/commit/d77f889f1e9156def86a91ac927582db024729b3) for the key handling. Which version is reported by `$i`? Should be `[PLUGIN:Macro plugin v0.09]`.
And the `$pins` output should contain these lines:
```
[PIN:PE1,Aux in 4,Macro pin]
[PIN:PE0,Aux in 5,Macro pin]
[PIN:PC11,Aux in 6,Macro pin]
```

> The new core, the delay is like 1 second mininum.

It works for me. Is `G4 P0.100` sent while motion is ongoing? If so the delay includes the time it takes to complete the motion.

siganberg, 2025-01-10:
> Odd, it is [this commit](https://github.com/grblHAL/Plugin_keypad/commit/d77f889f1e9156def86a91ac927582db024729b3) for the key handling. Which version is reported by `$i`? Should be `[PLUGIN:Macro plugin v0.09]`.
> And the `$pins` output should contain these lines:

Oh it's  on the keyboard plugin. I was actually looking on the `plugin` folder.  Will try tomorrow to pul the submodule keyboard. 

> It works for me. Is `G4 P0.100` sent while motion is ongoing? If so the delay includes the time it takes to complete the motion.

No. 

I have this on my macro just to make sure I toggle the Flood.

```
M8
G4 P0.100
M9
```

This was working from previous firmware prior update. When I update the same command takes like a second now vs 100 milliseconds.  I confirmed when I rollback  that the behavior change.

siganberg, 2025-01-13:
$673 is not even available. I found $393 instead but even changing that zero, still having 1 second delay instead of 100 ms for G4 P0.100

[image]

siganberg, 2025-01-13:
I created a gSender macro.

[image]

I can hear my relay takes a seconds to flip which is connected to the flood. This was not the case if I revert from the previous firmware.

terjeio, 2025-01-13:
Post your `$I` output please.

siganberg, 2025-01-13:
Here

```
[VER:1.1f.20250109:]
[OPT:VNMPZHSL+2,128,1024,4,0]
[NEWOPT:ENUMS,RT+,HOME,ES,EXPR,ATC,SED,ETH,TMC=7,SD,YM]
[FIRMWARE:grblHAL]
[SIGNALS:DEP]
[NVS STORAGE:*EEPROM 16K]
[FREE MEMORY:126K]
[DRIVER:STM32F412]
[DRIVER VERSION:250107]
[BOARD:SLB_B5.0.12.2]
[AUX IO:6,4,1,0]
[PLUGIN:Bootloader Entry v0.02]
[PLUGIN:SLB Probing v0.02]
[WIZCHIP:W5500]
[MAC:02:08:dc:29:c9:6a]
[IP:]
[PLUGIN:Trinamic v0.21]
[PLUGIN:MODBUS v0.18]
[PLUGIN:HUANYANG VFD v0.14]
[PLUGIN:HUANYANG P2A VFD v0.12]
[PLUGIN:Durapulse VFD GS20 vv0.07]
[PLUGIN:Yalang VFD YL620A v0.04]
[PLUGIN:MODVFD v0.05]
[PLUGIN:H-100 VFD v0.05]
[PLUGIN:Nowforever VFD v0.02]
[PLUGIN:Macro plugin v0.09]
[PLUGIN:RGB LED strips (M150) v0.05]
[PLUGIN:Spindle offset v0.01]
[PLUGIN:Events plugin v0.03]
[PLUGIN:SDCARD v1.16]
[PLUGIN:FS macro plugin v0.12]
[SPINDLE:Huanyang v1]
ok
ok
ok
ok
ok
2 - Huanyang v1, enabled as spindle 0, active, SDRV, current
0 - PWM
1 - PWM2
3 - Huanyang P2A
4 - Durapulse GS20
5 - Yalang YS620
6 - MODVFD
7 - H-100
8 - Nowforever
```

siganberg, 2025-01-13:
Oh, my tool change is affected too on RapidChangeATC macro. The spindle pauses a bit.

terjeio, 2025-01-13:
`$673` is hidden when safety door is enabled, it should not be, change [this line](https://github.com/grblHAL/core/blob/c681661516f3a803af88183e14fb0330918286dd/settings.c#L1846) to `available = hal.coolant_cap.mask;` Or reset settings and reload them as described below.

`$394` (spindle on delay) is set to what?

FYI both the new spindle on and coolant on delays are inherited from the previous "misused" door settings on an update from a pre 20250103 build. A `$RST=*` after updating followed by reloading settings will clear `$673` to 0 (or rather to whatever default is set in [config.h](https://github.com/grblHAL/core/blob/c681661516f3a803af88183e14fb0330918286dd/config.h#L1027-L1033)).

siganberg, 2025-01-14:
I started using the web configurator for SLB and this issue is no longer happening. Even the delay on the G4. 

Closing the ticket now. Thanks again for the support

closed 2025-01-14 by siganberg