grblHAL/STM32F4xx #208: Board Hangs When Network Connection is Used

siganberg, 2025-01-07:
## Description

When the board is connected to my home network (not tested in a peer-to-peer connection), it randomly hangs during a job, requiring a hard reset. Switching to a USB connection does not resolve the issue if the network cable remains connected; random board freezes persist.

However, if the network cable is not connected from the beginning, all jobs finish successfully without any issues.

## Steps to Reproduce

1. Connect the board to a home network using a network cable.
2. Start a job (e.g., via USB or network connection).
3. Observe the board hanging randomly during the job, requiring a hard reset.
4. Switch to a USB connection with the network cable still connected.
5. Note that the random freezes continue.

## Expected Behavior

The board should not hang during a job, regardless of whether the network cable is connected or not.

## Additional Notes

- Jobs run successfully when the network cable is not connected from the beginning.
- The issue has only been observed on my home network. Peer-to-peer connection behavior has not been tested.
- Disconnecting the network cable completely resolves the issue.

## Environment

- **Board Model**:  SIenci SLB
- **Firmware Version**: Latest
- **Network Type**: Home network (wired)
- **Connection Type**: USB and Network

terjeio, 2025-01-08:
I've had a similar report for the RP2040 driver and have not been able to replicate the issue.
Running some long term tests now, currently streaming gcode via the WebUI which should stress the system a fair bit since the SPI bus is shared between the SD card and the Ethernet chip. I have run some shorter tests (a few hours) earlier over the USB connection with no issues.

I am wondering if the Ethernet chip is, for some reason, generating a lot of interrupts in some setups and that could be a reason.
If there is a lot of traffic in the network then that may increase the risk. To find out if that could be the cause testing with either connecting via a dedicated switch or via a peer-to-peer connection might be helpful.

How long does it take, on average, before a freeze?
Do freezes happen even if no job is running?

siganberg, 2025-01-08:
> How long does it take, on average, before a freeze?

Around 1 to 2 minutes. Kinda random. 

> Do freezes happen even if no job is running?

No, only when job is running. I haven't encounter freeze with just controlling or jogging. 

I will try to test the peer-to-peer. I have pi-4 that I can dedicate for the gSender.

andrewmarles, 2025-01-08:
I think that with W5500 it can be necessary to check the pin explicitly after an IRQ is generated since if you get 2 in short succession a 2nd event is not always generated and this can cause instability.

https://github.com/Sienci-Labs/Plugin_networking/commit/d5284db41684ac540875454445cd9009fdc73513#diff-72ee043828422a425a05466a434c25e68ba362801a24e956bad823d0b4480a50

siganberg, 2025-01-09:
Another observed behavior, sometimes network connection is not completed successful unless you initially connect to USB.

terjeio, 2025-01-09:
I have now run two 10 hour jobs in paralell, on a SLB via the WebUI and on a PicoCNC via Telnet/ioSender. Both from the same client machine and both completed successfully.

> I think that with W5500 it can be necessary to check the pin explicitly after an IRQ is generated 

_enet.c_ has been refactored since, so this change is no longer relevant?

> Another observed behavior, sometimes network connection is not completed successful unless you initially connect to USB.

"completed successfully" is a bit ambiguous - is it a client connect that fails?
If so does it fail to connect at all or at some point while exhanging data with the controller?
Or is it the network connection that fails resulting in no response from pings?
Is the USB cable inserted or not when this happens? Or does it not matter?
I have tried power cycling the controller about 10 times without any issues connecting, this with the USB cable both inserted and not inserted. IP aquired via DHCP.

siganberg, 2025-01-09:
In the case I can't connect to the board using network  on startup (this is separte issue with freezing). 

> "completed successfully" is a bit ambiguous - is it a client connect that fails?
> If so does it fail to connect at all or at some point while exhanging data with the controller?
> Or is it the network connection that fails resulting in no response from pings?

In gSender, when everything is connected properly, all the UI button is enabled and you can see on the console all the response from the board. If this fail, I only see connected with baud rate speed and that's it. It never continue. 

> Is the USB cable inserted or not when this happens? Or does it not matter?

USB is not inserted.  I thint this matter. If USB is connected it works.

terjeio, 2025-01-09:
> .. (this is separte issue with freezing).

Perhaps not...

Can you ping the controller when connecting fails?
And how often does it fail? Once every 5 times or what?

siganberg, 2025-01-09:
> Can you ping the controller when connecting fails?

For connecting  incomplete, Yes. But when it freeze No. 

> And how often does it fail? Once every 5 times or what?

For networking running job, 100% failure for me. 

Here's a video for both connection failure and board freeze when running a job on the network.  Sorry it's a bit long. 

https://youtu.be/9yWhnSkbbiM

terjeio, 2025-01-09:
Can you post your settings ($$ output)?
And your firmware version is 20250104 or later?
With gsender 1.4.11 and iosender 2.0.45 I can connect to the SLB even with the USB cable disconnected so hard to tell why you can not.

siganberg, 2025-01-15:
Tried the firmware generated from the web configurator and still having board freezes when using network connection. 

I also tried the WebUI to run the job and did the same. Board freeze. 

Here's my `$I`

```
[VER:1.1f.20250114:]
[OPT:VNMZHSL2,128,1024,3,0]
[NEWOPT:ENUMS,RT+,HOME,ES,EXPR,ATC,SED,RTC,ETH,FTP,TMC=7,SD]
[FIRMWARE:grblHAL]
[SIGNALS:EP]
[NVS STORAGE:*EEPROM 16K]
[FREE MEMORY:140K]
[DRIVER:STM32F412]
[DRIVER VERSION:250107]
[BOARD:SuperLongBoard]
[AUX IO:6,9,1,0]
[PLUGIN:Bootloader Entry v0.02]
[PLUGIN:SLB Probing v0.02]
[WIZCHIP:W5500]
[MAC:02:08:dc:ca:62:d3]
[IP:192.168.68.240]
[PLUGIN:Trinamic v0.22]
[PLUGIN:MODBUS v0.18]
[PLUGIN:HUANYANG VFD v0.14]
[PLUGIN:KEYPAD v1.38]
[PLUGIN:Macro plugin v0.10]
[PLUGIN:WebUI v0.24]
[PLUGIN:RGB LED strips (M150) v0.05]
[PLUGIN:Feed override v0.01]
[PLUGIN:Events plugin v0.05]
[PLUGIN:SDCARD v1.18]
[PLUGIN:FS macro plugin v0.15]
ok
ok
ok
ok
ok
0 - Huanyang v1
ok
```

closed 2025-01-17 by siganberg