grblHAL/STM32F4xx #210: Build with protoneer_3.xx_map.h fails

fitch22, 2025-01-08:
Attempting a build with F446 Nucleo64 and protoneer_3.xx_map.h.  The failure occurs at line 47 with warning about unexpected token "&" and later at line 49 also with unexpected token "&".  I wish I understood more about the intent of this section and could offer help in a resolution.

If I change lines 45 - 50 to look like this, the compile succeeds, but I don't know if it works or not.

#if SPINDLE_ENABLE & (1 << SPINDLE_PWM0)
#undef SPINDLE_ENABLE
#undef DRIVER_SPINDLE_ENABLE // & SPINDLE_DIR
#define SPINDLE_ENABLE SPINDLE_PWM0_NODIR
#define DRIVER_SPINDLE_ENABLE SPINDLE_DIR
#endif

terjeio, 2025-01-09:
That code segment is gibberish, here is the correct one:

```
// Spindle PWM and dir signals cannot be assigned to the same pin
#if SPINDLE_ENABLE & (1 << SPINDLE_PWM0)
#undef SPINDLE_ENABLE
#undef DRIVER_SPINDLE_ENABLE
#define SPINDLE_ENABLE SPINDLE_PWM0_NODIR
#define DRIVER_SPINDLE_ENABLE (SPINDLE_ENA|SPINDLE_PWM)
#endif
```

Will fix in the next commit.

terjeio, 2025-01-23:
Fix [comitted](https://github.com/grblHAL/STM32F4xx/commit/363bb677e5c430cbfd614f16e503c82d2aaf7a76).

closed 2025-01-23 by terjeio