grblHAL/STM32F4xx #213: mks_robin_nano_v3.0 board map wrong AUXINPUT0_PORT selected

calabr, 2025-01-23:
Hi Terge,
 Started to play with MKS robin nano v3.1 board found AUXINPUT0_PORT  are configured to use GPIOG, which are not exists on this CPU. 
As result, created by WebBuilder firmware mapping _Safety door_ and _Motor fault_  inputs to nowhere.
Most probable it's some copy-past error from different board map

[mks_robin_nano_v3.0_map.h](https://github.com/grblHAL/STM32F4xx/blob/07d865df12a1c2ce45c3ef8f9571463c539332fe/boards/mks_robin_nano_v3.0_map.h#L170)

```[VER:1.1f.20250120:]
[OPT:VNMSL,100,1024,5,0]
[AXS:5:XYZAB]
[NEWOPT:ENUMS,RT+,ES,MPG,SED,ODO,SD,YM]
[FIRMWARE:grblHAL]
[SIGNALS:HSEFP]
[NVS STORAGE:*EEPROM 4K]
[FREE MEMORY:81K]
[DRIVER:STM32F407]
[DRIVER VERSION:250118]
[BOARD:MKS ROBIN-NANO 3.0]
[AUX IO:0,1,0,0]
[PLUGIN:Bootloader Entry v0.02]
[PLUGIN:Trinamic v0.24]
[PLUGIN:KEYPAD v1.38]
[PLUGIN:Macro plugin v0.10]
[PLUGIN:Homing pulloff v0.01]
[PLUGIN:ODOMETERS v0.06]
[PLUGIN:SDCARD v1.19]
[PLUGIN:FS macro plugin v0.15]
ok
```

```
GrblHAL 1.1f ['$' or '$HELP' for help]
[PIN:PC1,Feed hold]
[PIN:PD9,Cycle start]
[PIN:PC0,Emergency stop]
[PIN:PG6,Motor fault]
[PIN:PC4,Probe]
```

It's also mapping the same pin for  AUXOUTPUT1 and  AUXOUTPUT4 and some other overlapping definitions

```#define AUXOUTPUT1_PORT         GPIOB // Spindle direction, FAN1
#define AUXOUTPUT1_PIN          1

#define AUXOUTPUT4_PORT         GPIOB // Coolant mist, HEAT1
#define AUXOUTPUT4_PIN          1
```

MOOR_UARTZ and MOTOR_UART3 are mapped to the same pin,  MOTOR_UART5 are configured but board only supports 5 drivers in total
[mks_robin_nano_v3.0_map](https://github.com/grblHAL/STM32F4xx/blob/07d865df12a1c2ce45c3ef8f9571463c539332fe/boards/mks_robin_nano_v3.0_map.h#L210)

calabr, 2025-01-23:
I'm trying to summarize board alignment from MKS documents and board mapping. 
Problematic or not even mapping are highlighted yellow in this table
[MKS Robin nano V3.1 port map](https://docs.google.com/spreadsheets/d/1HD0BWfU2akqEgphzP7JfZ9qKVr0MQPdY7IHvMpuM0XM/edit?usp=sharing)

terjeio, 2025-01-23:
This is better? I do not have a board to verify with.

[mks_robin_nano_v3.0_map.zip](https://github.com/user-attachments/files/18516190/mks_robin_nano_v3.0_map.zip)

calabr, 2025-01-24:
> This is better? I do not have a board to verify with.

[mks_robin_nano_v3.0_map.zip](https://github.com/user-attachments/files/18516190/mks_robin_nano_v3.0_map.zip)

Thank you! it's looks much better.
  MOT3 and MOT4  UART are crossed -  should be:

#ifdef  M3_AVAILABLE
#define MOTOR_UARTM3_PORT       GPIOD
#define **MOTOR_UARTM3_PIN       9**
#endif

#ifdef  M4_AVAILABLE
#define MOTOR_UARTM4_PORT       GPIOD
#define **MOTOR_UARTM4_PIN       8**
#endif

#ifdef M3_AVAILABLE
#define MOTOR_CSM3_PORT         GPIOD
#define **MOTOR_CSM3_PIN          9**
#endif

#ifdef  M4_AVAILABLE
#define MOTOR_CSM4_PORT         GPIOD
#define **MOTOR_CSM4_PIN          8**
#endif

I'll also propose different port and button allocation It will use more protected inputs for buttons, allow more free inputs :  

```
PA13	SAFETY_DOOR			PW_DET
PA4	FEED_HOLD			MT_DET1
PE6	CYCLE_START			MT_DET2
PB2	RESET				PW_OFF
PA8	AUXOUTPUT1 (SPINDLE_DIRECTION)	BLTOUCH	
PC0	PROBE				TB 
PA2	AUXINPUT0 (motor_fault)		TH2	
PC1 	AUXINPUT1 - FREE		TH1	
```
   power outputs:  - move spindle PWM to most powerful output to allow connecting DC motors directly and logical output for the same port, direct connection of coolant DC motor or relay, reserve fan port for fan/vacuum etc. control
```
(PC14)	AUXOUTPUT5 - FREE reserved power out	FAN1 
(PB1)	AUXOUTPUT4 (COOLANT_MIST)		FAN2 
  High power outputs
PB0	AUXOUTPUT3 (COOLANT_FLOOD)		HE2/HE2-	
PE5	AUXOUTPUT2 (SPINDLE_ENABLE)		HE1/HE1-	
PA0	AUXOUTPUT0 (SPINDLE_PWM)		HB/HB-
```
It's also lots of ports still free on the board, but I'm only few days playing with board and cannot explore it yet.

terjeio, 2025-01-24:
I'll commit the version I now have with the CS/UART pins fixed.

Note that limit inputs for M3 and M4 motors has to be interrupt capable, this puts a restraint on how you assign other pins that requires interrupt capability, this is due to how STM32 interrupts are "wired". There are only 16 GPIO IRQs and a given pin number can only be assigned to one IRQ regardless of which port it belongs to. So e.g. the M3_LIMIT_PIN (PC4) and MT_DET1 (PA4) cannot both be assigned to an interrupt.
TH1 and TH2 has to be tested, not sure these inputs can be used to detect digital signals reliably due to the voltage divider at the inputs.

When you are happy with a board map please create a PR with your changes.

terjeio, 2025-01-27:
> pins with the same ID on different ports should be served by the same IRQ - Like PA0 and PB0 will trigger IRQ PinChange0.

Not so, only one of them can be "wired" to the IRQ line. See Figure 32 in the PRM.

> What is requirements of grblHAL for IRQ? 

[Limit](https://github.com/grblHAL/core/blob/0462c97dc919eb040fd96dd91a09d5ef8a8e8033/nuts_bolts.h#L125-L163) and [control](https://github.com/grblHAL/core/blob/0462c97dc919eb040fd96dd91a09d5ef8a8e8033/system.h#L137-L159) inputs (reset, feed hold, cycle start etc). Probe is optional since it is polled, but best to have on an IRQ line since that triggers a SR latch.

Some plugins requires interrupt capable auxiliary inputs as well.
There is support in some drivers for polling the Z limit input (for badly engineered 3D printer boards). I am no fan of polling inputs constantly, if you need to use polling it can be implemented in a board specific plugin. 

Encoder signals can be fed to a encoder peripheral if available (STM32 timers can also be used) - this way it is possible to avoid "using" up" the available GPIO IRQ lines.

calabr, 2025-01-28:
Unfortunately trying to build firmware with WebBuilder causing error 
```Command failed: pio run -etest -cjob.ini
Src/driver.c:83:2: error: #error Interrupt enabled input pins must have unique pin numbers!
#error Interrupt enabled input pins must have unique pin numbers!
^~~~~
*** [.pio/build/test/src/driver.o] Error 1```

Cannot decide what exactly overrides - minimal setup for 3 axis, no probe etc. giving the same results
Not sure if WebBuilder excluding all declarations related to unused inputs.
at-least I see 
```
#define CYCLE_START_PORT        GPIOE
#define CYCLE_START_PIN         6                           // MT_DET2
#define AUXINPUT0_PORT          GPIOA                       // EXP1 PA6
#define AUXINPUT0_PIN           6
...
#define RESET_PORT              GPIOE
#define RESET_PIN               12                          // EXP2 PE12

#define SD_DETECT_PORT          GPIOD
#define SD_DETECT_PIN           12
```
It's will be helpful if WebBuilder allows to download generated config files no matter successful build or not

terjeio, 2025-01-28:
I'll remove support for SD detect - it is not that important and is causing the compilation failure.

> It's will be helpful if WebBuilder allows to download generated config files no matter successful build or not

Sure, I'll try to add it.

calabr, 2025-01-29:
But what about AUXINPUT0 ? it's mapped to SAFETY_DOOR and MOTOR_FAULT
 Proposing the next mapping with regard of PIN interrupt restriction
```
Con 	| port name | PortID|IRQ 	| Column 1
----------------------------------------------------------
J54.3 	| PW_DET 	| PA13 	| 13 	| SAFETY_DOOR
J53.3 	| MT_DET2 	| PE6 	| 6 	| PROBE
EXP1.5 	| LCD_D4 	| PE14 	| 14 	| CYCLE_START
EXP1.7 	| LCD_D6 	| PD11 	| 11 	| FEED_HOLD
EXP1.8 	| LCD_D7 	| PD10 	| 10 	| Reset
EXP2.1 	| SPI1_MISO	| PA6 	|	| ENCODER A reserved
EXP2.2 	| SPI1_SCK 	| PA5 	| 5 	| INDEX reserved
EXP2.6 	| SPI1_MOSI 	| PA7 	|   	| ENCODER B reserved
SD5 	| SD_DET 	| PD12 	| 12 	| SD_DETECT
```
The idea is to move Door and Probe inputs to PW_DET and  MT_DET2 because this connectors are 3-pin and contains +5V power, copacitor - better much for external connectors.
move buttons to EXP1 connector - it's open pins from controller, better much buttons and future control panel connection
PA6,PA7 are inputs of encoder capable counter, so good candidates for encoder connection without PIN interrupts.
 
Regarding analog inputs TB, TH1, TH2 - it's isn't divider - it's filtered - 10u capacitor in parallel, 4k pool-up to 3v3, protective diodes from GND and 3v3, 100 Ohm resistor toward controller PIN. So it's not the keys for digital inputs/outputs until capacitors are removed.

calabr, 2025-01-30:
I'm just realized, analog inputs TB, TH1,TH2  on this boards should fit button connection - 10u capacitor not best, but not the problem for mechanical buttons.
Proposing better inputs  allocation with some room for encoder and AUX inputs

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/mkipnis/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/mkipnis/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#467886" vlink="#96607D">

Con | Board port name | Port ID | PIN   IRQ | grblHAL   input
-- | -- | -- | -- | --
J54.3 | PW_DET | PA13 | 13 | SAFETY_DOOR
J53.3 | MT_DET2 | PE6 | 6 | PROBE
TB.2 | TB | PC0 | 0 | CYCLE_START
TH1.2 | TH1 | PC1 | 1 | RESET
EXP1.5 | LCD_D4 | PE14 | 14 | AUX_in reserved   (reserved)
EXP2.1 | SPI1_MISO | PA6 | TIM3 | ENCODER A (timer)   (reserved)
EXP2.2 | SPI1_SCK | PA5 | 5 | INDEX (reserved)
EXP2.4 | SPI1_CS | PE10 | 10 | ENCODER2 A (reserved)
EXP2.5 | BTN_EN2 | PE11 | 11 | ENCODER2 B (reserved)
EXP2.6 | SPI1_MOSI | PA7 | TIM3 | ENCODER B (timer)   (reserved)
SD5 | SD_DET | PD12 | 12 | SD_DETECT

</body>

</html>

I'm didn't setup my build environment yet, so it's not tested.

Going to prepare map.h @terjeio  is it Ok for WebBuilder to allocate PROBE_PIN and SAFETY_DOOR_PIN directly without reference to AUXINPUT0_PIN  AUXINPUT1_PIN ? I'm want to free AUX for different tasks from plugins. 
Or probable add additional AUXINPUT declaration

terjeio, 2025-01-31:
> I'm just realized, analog inputs TB, TH1,TH2 on this boards should fit button connection

You have to verify that these inputs can actually be used first. There are no values for the resistors in my schematic so impossible to calculate if the input low voltage is below the min. required.

> Ok for WebBuilder to allocate PROBE_PIN and SAFETY_DOOR_PIN directly without reference to AUXINPUT0_PIN AUXINPUT1_PIN ?

No, add additional aux inputs instead.

terjeio, 2025-02-24:
> How to properly select timer for spindle PWM ?

Select from [this list](https://github.com/grblHAL/STM32F4xx/blob/fc92ee2700ef76c28e85486e69da960c2dcf386c/Src/pwm.c#L34-L97) - or add to it if possible. I have not worked through every possible combination, refer to the datasheet for options.

> What is right settings for SPINDLE0_ENABLE ? 

That depends on the map file, and/or which spindles in the spindle plugin in needed. SPINDLE_PWM0 requires pin definitions for [PWM, enable and dir](https://github.com/grblHAL/STM32F4xx/blob/fc92ee2700ef76c28e85486e69da960c2dcf386c/boards/st_morpho_map.h#L139-L151) in the map file.

> Sould I create #ifdef statement to assign alternative aux, or it's just enough to mention AUX as used by function? like this

Define aux pins and bind them to the function like I do in the map files - [like this](https://github.com/grblHAL/STM32F4xx/blob/fc92ee2700ef76c28e85486e69da960c2dcf386c/boards/st_morpho_map.h#L115-L176). The pins will be claimed from the aux pool as needed by the configuration.

> How to control interrupt claiming of some inputs, like PROBE_DISCONNECT, SINGLE_BLOCK etc.

They will be claimed from the aux input pool [if enabled](https://github.com/grblHAL/STM32F4xx/blob/fc92ee2700ef76c28e85486e69da960c2dcf386c/Inc/my_machine.h#L126-L137). Some may be labeled on the board and are [explicitly mapped](https://github.com/grblHAL/STM32F4xx/blob/fc92ee2700ef76c28e85486e69da960c2dcf386c/boards/st_morpho_map.h#L212-L271).

> Is it any chart of AUX in/out allocation and priority?

Not really - [perhaps this](https://github.com/grblHAL/core/blob/3e84d4b96c822568b9a6f45f14726e3c5bbf989b/pin_bits_masks.h#L218-L285). Those not explicitly mapped has `.pin = 0xFF, .port = NULL` set in the list. `.irq_mode` is the IRQ requirement for the input.

> what is the usage of SPINDLE_ENCODER_ENABLE 1 - how is it works without SPINDLE_SYNC_ENABLE ?

To get RPM feedback - can be used for "at speed" detection. I have a driver with experimental closed loop control of RPM...

> What is the usage of QEI_ENABLE and how it's correlated with SPINDLE_ENCODER and SPINDLE_SYNC ?

Currently not related, it is for input to the encoder plugin.

> Is it any way to inspect resulting pin allocation, IRQ and timer usage without re-fleshing board every time?

You can partially check in the IDE if it supports indexing. Like this in the STM32CubeIDE:

[image]

Diasabled code is greyed out.

For pins allocated not explicitly you cannot - those are claimed at startup. For those allocated explicitly I can hover over the pin and port definitions to see the allocation, which is also visible in the map file:

[image]

calabr, 2025-03-01:
I'm trying to remap timer usage to allow PA0 for SPINDLE_PWM. this port can be mapped to TIM2_CH1 or TIM5_CH1. TIM5 are in use as STEPPER_TIMER_N and  TIM2 are mapped to RPM_COUNTER_N or RPM_TIMER_N
When I'm trying to remap RPM_TIMER_N to TIM9, which haі available exposed pin I'm getting error
```../Inc/driver.h:65:21: error: 'TIM9_IRQn' undeclared (first use in this function); did you mean 'TIM7_IRQn'?
   65 | #define timerint(t) TIM ## t ## _IRQn```
Could you help to properly define timer.
Is it possible to move  STEPPER_TIMER_N TIM5 and  PULSE_TIMER_N  TIM4 to higher timer number - TIM10, TIM11?  - this timers are not using in/out pins and probable  better to save TIM4, TIM5 for other functions

calabr, 2025-03-02:
After investigation it's seems TIM8_CC_IRQn should be used instead of TIM8_IRQn, but I cannot resolve how to properly handle it

terjeio, 2025-03-02:
> Is it possible to move STEPPER_TIMER_N TIM5 and PULSE_TIMER_N TIM4 to higher timer number - TIM10, TIM11? - this timers are not using in/out pins and probable better to save TIM4, TIM5 for other functions

STEPPER_TIMER_N should be 32bit, only TIM2 or TIM5 are.

The pulse timer could be mapped to a higher numbered timer but since not all processor variants supported has these available it must be done by adding code guards.

> After investigation it's seems TIM8_CC_IRQn should be used instead of TIM8_IRQn, but I cannot resolve how to properly handle it

It can be done by by defining the IRQ explicitly instead of using the macro.